dresden-elektronik/deconz-rest-plugin #2679: Legrand and LUMI Aqara : Bad clustering

smesguich, 2020-04-11:
Hi !
I've just check somethin strange. I'm mooving from my Legrand GW to Deconz and with an Aqara/LUMI Power Outlet, a "Group 0" is created but it's affecting all Legrand devices.

Don't now if the issue came from Legrand Device zigbee integration or Aqara/LUMI.
As far as I can see in the IHM (Phoscon and old one), the Group 0 contain only the Aqara Device.

[image]

Smanar, 2020-04-12:
Bad luck, there is already an issue for it https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2065

But there isn't answer yet.

You know the API ? To try to delete it ?

smesguich, 2020-04-12:
> 
> 
> Bad luck, there is already an issue for it #2065
> 
> But there isn't answer yet.
> 
> You know the API ? To try to delete it ?

Oh, I didn't saw this issue.
Yeah I know the API but I didn't want to do some mistakes or removing some intresting trace to solve the issue :)

Smanar, 2020-04-12:
Yep, you are right,

smesguich, 2020-04-20:
Hum, the group disapear without reason .... I Will re-open if it happens again with some trace and logs wich I didn't found right now ...

closed 2020-04-20 by smesguich