dresden-elektronik/deconz-rest-plugin #2680: Two instances of Deconz running, one useless.

maxheartrate, 2020-04-12:
Hi all,
I guess my problem is more of a user related problem. Anyway, I hope someone can help me. I seem to have two instances of Deconz running. One is on port 8090. It shows the Conbee icon on the login screen. The second is running on port 8080. It shows the Raspbee icon. I cannot login to this. On Deconz GUI the status is "Not connected". I guess I must have made a mistake somewhere along the way. I probably have the GUI and headless version running at the same time.
I tried to uninstall Deconz and reinstall but the problem persists. Does anyone have an idea how to fix this?
Best, Max

Smanar, 2020-04-13:
So yes deconz can work if you have 2 instance running.
But I m seing only one running on your command ?

And It seem you are using the headless version, not the GUI version ? (-platform minimal)

jensflorian, 2020-04-13:
It is also possible that the non-accessible deCONZ instances arereturned by the appspot server on discovery. Someone else who used your IP might be running a Raspbee on 8080.

maxheartrate, 2020-04-13:
Thank you both for your replies, @Smanar and @jensflorian.

Here are my outputs for 
sudo systemctl status deconz

`● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2016-11-03 18:16:47 CET; 3 years 5 months ago
 Main PID: 319 (deCONZ)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/deconz.service
           └─319 /usr/bin/deCONZ -platform minimal --http-port=8090

Nov 03 18:16:47 magicmirror systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Nov 03 18:16:49 magicmirror deCONZ[319]: libpng warning: iCCP: known incorrect sRGB profile
Nov 03 18:16:50 magicmirror deCONZ[319]: This plugin does not support propagateSizeHints()
Nov 03 18:16:50 magicmirror deCONZ[319]: This plugin does not support raise()
Nov 03 18:16:52 magicmirror deCONZ[319]: This plugin does not support propagateSizeHints()
Nov 03 18:16:53 magicmirror deCONZ[319]: This plugin does not support propagateSizeHints()
Apr 13 22:21:07 magicmirror deCONZ[319]: This plugin does not support propagateSizeHints()
Apr 13 22:21:07 magicmirror deCONZ[319]: This plugin does not support raise()`

and
sudo systemctl status deconz-gui

`● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz-gui.service.d
           └─override.conf
   Active: active (running) since Mon 2020-04-13 22:21:10 CEST; 1min 49s ago
 Main PID: 1760 (deCONZ)
    Tasks: 12 (limit: 4915)
   CGroup: /system.slice/deconz-gui.service
           ├─1760 /usr/bin/deCONZ --http-port=8090
           ├─1789 dbus-launch --autolaunch 2408e64e348a4831a0e2accf375652c9 --binary-syntax --close-stderr
           └─1790 /usr/bin/dbus-daemon --fork --print-pid 5 --print-address 7 --session

Apr 13 22:21:10 magicmirror systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Apr 13 22:21:10 magicmirror deCONZ[1760]: libEGL warning: DRI2: failed to authenticate
Apr 13 22:21:11 magicmirror deCONZ[1760]: libpng warning: iCCP: known incorrect sRGB profile
Warning: deconz-gui.service changed on disk. Run 'systemctl daemon-reload' to reload units.`

Smanar, 2020-04-14:
Ok so IDK why the first command hven't worked but you realy have 2 instances.

You can just disable the one you don't want.

```
$ sudo systemctl disable deconz
$ sudo systemctl stop deconz
```

or
```
$ sudo systemctl disable deconz-gui
$ sudo systemctl stop deconz-gui
```

maxheartrate, 2020-04-14:
Thank you for your advise, @Smanar. Will these commands also prevent deconz or deconz-gui from auto starting on the next reboot? Or do I have to disable autostart somewhere else?

jensflorian, 2020-04-14:
the disable command makes sure it won‘t restart after reboot.

maxheartrate, 2020-04-14:
Thank you, @jensflorian. You just made a newbie happy. :-)
Stay safe.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix