dresden-elektronik/deconz-rest-plugin #2681: Reset network settings to default

rkjnsn, 2020-04-12:
I have a ConBee II. Is there any way to restore the network settings (stored in the NVRAM) to their defaults?

rkjnsn, 2020-04-13:
Alternatively, are the default settings documented anywhere? I don't see them in the manual. I probably should have taken a screenshot before tinkering, but it didn't occur to be there wouldn't be an option to revert to the defaults.

Smanar, 2020-04-13:
In phoscon / gateway there is a reset feature (I have never tried it on my side, IDK if it's the good one)

rkjnsn, 2020-05-05:
Huh. I tried that and it seemed only to clear the lights, scenes, et cetera. I just did it again, though, and it actually did seem to update the network settings this time. :shrug:

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix