dresden-elektronik/deconz-rest-plugin #2683: Conbee II Stick not found after a some time

kmplngj, 2020-04-13:
Hello,

since one of the last Deconz updates my Conbee II does not show a fireware version after some hours after a reboot and stops working.

I did not change anything on the hardware site. Before the system was running months without problems.

Any idea where I could look to find the problem?

Thanks
Jan

kmplngj, 2020-04-27:
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4539668/deconz.log)

I added my log file. It show the process between same restarts. 

Any one have an idea?

Thanks a lot
Jan

Smanar, 2020-04-27:
Hello, how are you powering the device ?
Have you tried the USB extension cable ?

You device don't alternate connexion / deconnexion ? like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493

kmplngj, 2020-04-27:
I run it on my ESXi host.
I have an USB extension cable. 

The setup was running for months. Since I did an update of the docker image I have the problems.

I think I have to check my logs to find if I have the same problem as #2419

Smanar, 2020-04-27:
I know lot of user with the issue #2419, for most of them to reconize the defaut
- Pluged on a Windows machine they hear the "USB sound" endlessly
- The firmware version appear/disapear endlessly

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

kmplngj, 2020-06-05:
@Mimiix no not solved ):

closed 2020-07-11 by stale[bot]