dresden-elektronik/deconz-rest-plugin #2685: HEEEEELP!  houstan we have a "open" issue problem here. Please close issues.

scobby, 2020-04-14:
Hello Community!
Please help and close some of the issues, so we get things cleaned up for better overview!
If an issue is solved or not an issue anymore, just close it. If its still open, ask the author if this problem is still there.

Any other suggestions how to get the messy issues away?

Sjonnie2018, 2020-04-17:
Agree! I have closed my issues as soon as they were resolved. Seems like the issues not closed are related to "hit-and-run" users. Maybe introduce a bot that marks issues closed after 10 days?

Hypfer, 2020-04-19:
The bot already exists and is closing issues. Even the ones which are still relevant so that's definitely not the best solution.

Honestly this meta-issue is just making it worse.

wvuyk, 2020-04-20:
So better close this one...

closed 2020-04-23 by scobby