dresden-elektronik/deconz-rest-plugin #2686: Disable Hue Motion Sensor via REST API

Thorhammerk, 2020-04-14:
Hi, i´m trying to disable one of my motion sensor.
My system:
- Raspi 3
- ConBee II
- Version: 2.05.74
- Sensor Firmware: 6.1.1.27575 

With YARC! i send this PUT to the sensor

```
http://192.168.xxx.xxx/api/A6XXXXXXXXXX/sensors/44/config
{
"on":true
}

```
so it seem to work.
```
{
  "config": {
    "alert": "none",
    "battery": 100,
    "delay": 0,
    "ledindication": false,
    "on": false,
    "pending": [],
    "reachable": true,
    "sensitivity": 2,
    "sensitivitymax": 2,
    "usertest": false
  }
```
But the sensor still detected presence.
Any ideas on that?
best regards

ebaauw, 2020-04-14:
Yes, this is one of the subtle differences with the Hue API.  When setting `config.on` to false, rules are no longer triggered, but, the sensor `state` is still updated.

Thorhammerk, 2020-04-14:
Hi, thx for the quick answer. Is there any workaround to achieve this goal?

ebaauw, 2020-04-14:
No, it needs to be fixed in the REST API plugin.  I raised this years ago, see #90, but low prio apparently.

Thorhammerk, 2020-04-15:
ok, thx. Then i ´ll find a workaround outside the REST API :-)

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix