dresden-elektronik/deconz-rest-plugin #2687: Aqara vs. Xiaomi door sensor - periodic updates difference

bkupidura, 2020-04-15:
I own a few Aqara and Xiaomi door smart sensors.
Recently discovered, that Aqara sends periodic updates every ~60minutes when Xiaomi is sleeping till state change (reedswitch open/close) (at least for last 48h).

This is not neccessary bad thing, depending on needs based on that knowledge user can choose between Aqara and Xiaomi.

Product: Conbee II
Version: 2.05.75 / 3/8/2020
Firmware: 26490700

Aqara: 
Vendor Xiaomi
Product Window / Door Sensor
Version 20161128

Xiaomi:
Vendor Xiaomi
Product Window / Door Sensor
Version unknown

Can anybody else observed that? If yes, maybe we should add some information in supported devices page?

And second question, anybody knows if this behavior can be changed? 
Im new in Zigbee/Deconz world, so not sure if this is somehow configurable if not maybe sensor FW update is possible?

Smanar, 2020-04-16:
Ha, I did not know that, these devices are visually identical?

bkupidura, 2020-04-16:
No, both are different (not sure about internals, didnt open enclosure).

Xiaomi: https://pl.gearbest.com/alarm-systems/pp_3006951909382808.html
Aqara: https://pl.gearbest.com/access-control/pp_626703.html

bkupidura, 2020-04-16:
And probably its more complicated than just aqara vs. xiaomi.

I also own 3 aqara leak sensors, till yesterday 3 of them were reporting every ~60m. I was migrating deconz to different node (GW was offline for ~2h). 
After reconnect i see that 2 of 3 are reporting as always (~60m).
But 1 of 3, is deep sleeper now... All 3 are connected to GW directly (without routers).

I tested if sensor is really alive but when water is discovered, it reported properly. But no intermediate messages...

Willl observe, that further.

If anybody have any idea why sensors behaves this way, please share yours idea.

ebaauw, 2020-04-16:
Battery is running empty.  They tend to stop periodic reporting, then report wrong values, then die.  Don't be fooled by the value of `config.battery`: it's computed from the voltage reported through the hourly report and inherently inaccurate.

bkupidura, 2020-04-16:
Thanks for hints about battery.

Regarding periodic updates aqara vs. xiaomi, its my misunderstanding off "lastUpdated" in API.
lastUpdated != lastSeen.

Looks like Aqara sensors, are sending messages, treated by deconz as different than previous state (this is why lastUpdated is changed).
I dont see any difference in state (battery, temp, lux, ...) but it dosent mean deconz thinks same way ;)

Now i created my `*_last_updated` metric based on lastSeen (from WS), and looks like both aqara and xiaomi are always seen at least 1 per hour.

ebaauw, 2020-04-16:
`lastupdated` is updated when deCONZ receives a message from the corresponding cluster; `lastseen` is updated when deCONZ receives any message from the device.  Did you compile the plugin yourself?  `lastseen` is not yet in 2.05.75.

> Now i created my *_last_updated metric based on lastSeen (from WS)

You lost me here.  Web socket notifications for `lastseen` are not yet implemented.

bkupidura, 2020-04-16:
Im using nodered and attached to deconz WS. Building my own "lastSeen" based on messages from web sockets.

When new message arrives, i save current time for given sensor and treat this as "lastSeen". Its not true lastSeen (as described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2590), but still better than lastUpdated from API.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix