dresden-elektronik/deconz-rest-plugin #2689: Develco Door/window sensor with temperature not added correctly to deConz (conbee2)

FredrikNas, 2020-04-15:
Hi guys! 

I found this thread regarding my issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1667

I have the same problem, the Develco magnet sensor for door / window that i bought cant be added to deconz. 

When i set the sensor in pairing mode (red rapid flashing light) Deconz seems to notice it, and says a device has been found, and lets me pair it to Deconz. But when the pairing is done, the sensor is not added to the list of sensors, nor switches or lights. 

Altought when i move into home assistant, i can see my door/window sensor with 3 entities open/close, temperature and battery status.

Battery and temperature seems to be working, but it does not pickup on the magnet-activity regarding open/close. 

(Sidenote) my Develco smoke sensors added correctly to Deconz, but are registrered as lights.

SwoopX, 2020-04-15:
@FredrikNas The device will not be visible in Phoscon unless specifically added by DE; it's not open source. If you want to check correct integration, you must go via REST API.

Could you please also provide all the screenshots required for a device support request (check the wiki)? Wanna make sure that nothing has changed.

fl0pp, 2020-06-06:
I have the same problem.
The Cluster info tab reports "unknown Cluster".
I have just jumped into ZigBee, so I do not know what to do next.

Here are some screenshots:

[image]

[image]

[image]

SwoopX, 2020-06-06:
All working fine, what's the issue?

fl0pp, 2020-06-06:
Same as @FredrikNas 

Battery and temperature is working, but it does not pickup on the magnet-activity regarding open/close.

The entity ID binary_sensor.openclose_13 is always reporting "open" in Home Assistant.

When I remove the magnet from the sensor, the sensor blinks green, but there are no activity on the LED's inside deconz GUI.
When I place the magnet on the sensor, the sensor blinks green, inside deconz GUI the LED blink the following pattern: blue => gray => blue => gray => green => gray => green => gray => green => gray

SwoopX, 2020-06-06:
Which deconz version are you running?

Can you run deconz with `--dbg-info=2 > dbg`, trigger the sensor a few times and provide the debug log? Also, please trigger the device and directly read the IAS zone attributes. It should be enrolled...