dresden-elektronik/deconz-rest-plugin #2690: Reset / Delete gateway

MickL, 2020-04-15:
I reinstalled a fresh installation of hassio on my Raspberry Pi and added the deCONZ addon. Now when I open Phoscon it shows me my old gateway but it is greyed out. How can it still have the old gateway when the sd card was formatted? Further I cant find my sensors now that I resetted.

How may I reset the ConBee? Or at least delete the old gateway?

Smanar, 2020-04-16:
Hello, on phoscon / gateway, you have a number near "firmware" or "not connected"

MickL, 2020-04-16:
It says not connected

Smanar, 2020-04-16:
Ok, so I think it s a raspbee ?
Have you set the serial port according to the installation guide ?

MickL, 2020-04-16:
Yes it is a RaspBee. What do you mean by setting the serial port? I did set `/dev/ttyAMA0` within hassio deconz addon, otherwise it wouldnt work at all. I assume everything else is done by the addon?

Where would I execute `GCFFlasher_internal -l`? I can SSH to my hassio but there the command is not known. Also I can VNC to the deCONZ app but there is not terminal. When I click there on connect nothing happes:

[image]

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

MickL, 2020-06-06:
How does it seem so? I am still having a useless RaspBee and a useless smart home system :(

I wrote an e-mail to Dresden Electronics with no response, too.

Mimiix, 2020-06-06:
The reason for me to close was the time that has passed without any response or bumps. However, I'll re open and put it on the todo

SwoopX, 2020-06-06:
The issue is not with deconz or the Raspbee, it's the configuration based on your system setup and the correspondingly introduced restrictions, dependencies and additionally introduced complexity. However, that doesn't help us here.

I'd recommend you join the deconz Discord channel (invite link the the repo's readme). We've folks around also running Hass and as I recall, there should have been comparable questions there before. At least you should find someone who can support in troubleshooting.

Maybe, given success in resolution, try to shape some guidance for the FAQ/wiki to share with others.

smartins4, 2020-06-15:
I have exactly the same issue as described by MickL.
The main question is whether a hard reset can be applied to de Raspbee as the button reset gatway fails.

Mimiix, 2020-06-15:
As far as i can tell: did you also start with a clean addon installation? Or did you recover a backup ?

MickL, 2020-06-15:
I formatted everything and did a fresh install with no backups

SwoopX, 2020-06-15:
So what's the issue now here? I'd say the device not being in network, right? in 99% of all cases, that is a configuration issue of the system where people forgot something or doing it incorrectly. Can you please post the outout from the command `ls -la /dev/tty*` here?

smartins4, 2020-06-15:
> As far as i can tell: did you also start with a clean addon installation? Or did you recover a backup ?

in my case I've restored a backup file, and in the meanwhile i have uninstalled/installed/reconfigured the addon.

smartins4, 2020-06-15:
> So what's the issue now here? I'd say the device not being in network, right? in 99% of all cases, that is a configuration issue of the system where people forgot something or doing it incorrectly. Can you please post the outout from the command `ls -la /dev/tty*` here?

ls -la /dev/tty*
crw-rw-rw- 1 root root 5, /dev/tty
crw-rw---- 1 root audio 166, /dev/ttyACM0
crw-rw---- 1 root audio 204, /dev/ttyAMA0

Mimiix, 2020-06-15:
@smartins4 Please open a own case, as you restored a backup file.

SwoopX, 2020-06-15:
Agree. However, permissions are incorrect.

smartins4, 2020-06-16:
Case #2956 is now open.

closed 2020-07-18 by stale[bot]

MickL, 2022-06-18:
2 years later I try to setup my smart home again. Hours and hours later I cant get it to work. Phoscon says "Vendor unknown" and "Firmware not connected". I cant update the firmware manually, always runs into a timeout.

I tried with a Raspberry Pi 3 and Raspberry Pi 4, both have the exact same problems.

This issue has been closed. The follow up issue also has been closed. Why?

Smanar, 2022-06-19:
Hello, have you used the command line with timout parameter ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

MickL, 2022-06-19:
Yes as I described here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2828

The Raspbee is not working anymore. I tried with two Raspberry Pi's. There is no way to reset it. This issue has been closed, they dont seem interested to implement a reset feature. 40€ for the garbage bin :(

Smanar, 2022-06-19:
We can re-open but your issue have nothing to see.
Your issue just concern a Raspbee not reconised by the hardware.

Can you show your command line and the result you have with it ?

MickL, 2022-06-19:
What do you mean exactly by "show the result"?

I tried a Raspberry Pi 3 and 4, both with fresh installs of Homebridge. Phoscon shows "Vendor unknown" and "Firmware not connected". When I try to update the Raspbee i get a timeout:

```
flashing 127650 bytes: |======error: timeout flashing firmware after 3000 ms
```

```sh
pi@homebridge:/tmp $ sudo GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee

pi@homebridge:/tmp $ sudo GCFFlasher_internal -t 60 -f deCONZ_Rpi_0x26400500.bin.GCF -d /dev/ttyS0
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyS0 (RaspBee)
Reboot device /dev/ttyS0 (RaspBee)
Reboot device /dev/ttyS0 (RaspBee)
action: reset device RaspBee
wiringPi 2.52 initialized
Reboot device /dev/ttyS0 (RaspBee)
action: reset device RaspBee
RaspBee Bootloader premium
Vers. 1.02
uild 2013/ (52 ms)
uild 2013/08/01
 (52 ms)
uild 2013/08/01
R (390 ms)
uild 2013/08/01
RE (390 ms)
uild 2013/08/01
REA (391 ms)
uild 2013/08/01
READ (391 ms)
flashing 127650 bytes: |=======error: timeout flashing firmware after 3000 ms
```

Smanar, 2022-06-19:
Ok, so sorry, but I don't see something special on my side.
Can try with the "-x 3" parameter to have more logs.

You don't have other device on GPIO pin ?

And better to ask here https://forum.phoscon.de/c/deconz/general-support/12

Because the github is for developpement, not hardware issue.

Mimiix, 2022-06-19:
Please use the forum for this type of question. We are not using github anymore for user questions.

Https://forum.phoscon.de

MickL, 2022-06-24:
Maybe there is no reset needed because the gateway itself does not have any persistent storage? This is was I was expecting initially.

Anyway, if anyone reads this: I wasnt able to update the Rsapbee with the Homebridge image (that uses Raspberry Pi OS) but I was able to update the RaspBee I with a clean install of Raspberry Pi OS and following the [docs](https://phoscon.de/en/raspbee/install). For the RaspBee II it was actually flashing the firmware but it didnt report success so it flashed 3 times and then reported a timeout. In fact it actually flashed.

For Home Assistant I also made it work. The important thing is to update the `config.txt` and add:

```
enable_uart=1
dtoverlay=pi3-miniuart-bt
```

(This step is fairly the hardest because Home Assistant uses weird partition formats)