dresden-elektronik/deconz-rest-plugin #2691: Lamp does not get added to group

ChrisPrefect, 2020-04-15:
This one lamp cannot be added to a group. The Phoscon app shows that it is added to a group, but when I go to lights, it is not there.
If I go back to the lights, the group is missing again.
I made a screen recording about the issue.

[Phoscon App - Google Chrome 2020-04-15 22-17-34.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4483580/Phoscon.App.-.Google.Chrome.2020-04-15.22-17-34.zip)

ebaauw, 2020-04-15:
What type of lamp?  Does the GUI show a server (blue) _Groups_ cluster for it?

ChrisPrefect, 2020-04-15:
A Tradfri 10W transformator.

I can’t use the GUI as it is a headless raspberry pi.

Can I access the information about the zigbee network from a windows pc?

Sadly, the network works very poorly with just half the lamps (around 60) added. Szenes often only change half the lamps in a group of 10-15 lamps. It was better (but still bad) with the original Tradfri gateway.

ChrisPrefect, 2020-05-05:
I had to remove the lamp from the socket in the ceiling and connect it to an adapter and bring it closer to the conbee II stick, then I was able to set the group one last time and it got saved.

The lamp was about 6 meters away (clear line of sight) and I had to bring it to about 2 meters from the stick till it worked.

Very strange that the stick was not able to reach the light correctly just 6 meters away..

ebaauw, 2020-05-06:
Sounds like radio interference.  The usual hints: Use a USB extension cable for the ConBee, keep some distance from Wifi, Bluetooth, DECT, and other devices using the 2.4GHz band, disconnect any USB3 devices.  Don’t use metal enclosures for the bulb (Faraday’s cage).

ChrisPrefect, 2020-05-07:
I use an USB extension cable (Apple USB extension). I changed the 2.4G wifi channel to channel 1 (zigbee is on channel 25) and moved all devices that support it to 5.8G. There is no DECT in the house and no nearby wifis. No USB3 devices on the RPi3. The bulbs are in a ceiling dry wall with a metal GU10 spot holder.

Also scenes are nearly unusable. The brightness and color settings are never saved or read properly.

ebaauw, 2020-05-07:
These are all Trådfri spots?  Recalling scenes is done using a Zigbee broadcast command, which is immune to routing issues.  Still smells like interference, but you seem to have covered most of that.  Just to be sure: how much of the spot does the holder cover?  Is the holder open in the ceiling (exposing the side/back of the spot, or does the holder only leave an opening for the spot’s lens, where the light comes out.

Are running deCONZ with debug logging enabled?  Are you seeing any error messages in the log?  Interference issues would be visible there, as would issues between deCONZ and the ConBee.

Did you change the Zigbee channel after pairing the spots?  Not all devices might have picked up the new channel, if reception on the old channel was bad (or if they were powered down when the channel was changed).

Is it always the same lights failing to respond, or is it different lights per attempt?

ChrisPrefect, 2020-05-07:
Does the mesh network / routing work properly? I read something about a problem with the routing and Tradfri-devices. 

Yes it looks like a scene broadcast only reaches some of the lights. Are scene-settings read from the lights back to Phoscon when I want to edit them? Or why are the sliders fluctuating every time I switch the scene I want to edit? I made a video here: https://github.com/dresden-elektronik/phoscon-app-beta/issues/65

It's just a horizontal holder like this one:

[image]

Is there a chance to check the signal strength and mesh topology without having an X-server and desktop on the Raspberry pi?

How can I get the logs when I am connected with ssh to the Raspberry Pi?

I did not change the channel after pairing.

It seems to be random lights most of the time. Some times the 10 GU10 spots in the bathroom take over 5 seconds to properly turn on or off.

I also have weird problems with Alexa, like when I say "Bedroom warmer" it works, but Alexa says "something went wrong". Other times only one Lamp from the group reacts to Alexa "on/off" commands but the tradfri-remote that is connected to this group successfully controls both lights.

Thanks for your help!

ebaauw, 2020-05-08:
> Does the mesh network / routing work properly? I read something about a problem with the routing and Tradfri-devices.

There are still issues, see #1261, but broadcasts don’t use routing and are unaffected by these issues.

> Yes it looks like a scene broadcast only reaches some of the lights.
> It seems to be random lights most of the time.
> I did not change the channel after pairing.

Very weird.  As a hail Mary, could you try powering off all of your lights, powering them back on in order of proximity to the ConBee.

> It's just a horizontal holder like this one:

That should be ok.  I have similar ones in my bathroom.

> Is there a chance to check the signal strength and mesh topology without having an X-server and desktop on the Raspberry pi?

No.  You need the full version of Raspbian (incl. the X server), but you don’t need a connected monitor; use VNC to connect to the desktop over the network.

> How can I get the logs when I am connected with ssh to the Raspberry Pi?

If you’re using the standard service to start deCONZ, the log messages are written to syslog.  Use `journalctl` to display them, or simply check `/var/log/syslog`.  If you’re starting deCONZ manually, the messages are written to the standard output (or error output).  Make sure to start deCONZ with `--dbg-error=1`, or it won’t output the error messages.

> I also have weird problems with Alexa

I don’t know Alexa, but typically those integrations don’t use groups.  They’re almost guaranteed to swamp the Zigbee network, controlling each light individually.  Here you would run into any routing issues.

SwoopX, 2020-05-08:
You can use XForwarding, so you don't need to have a Desktop installed as decribed here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

ChrisPrefect, 2020-06-04:
So, thanks a lot for your input some weeks ago. I did try a lot since then.

I saved the config in the Phoscon app, installed the XWindow/desktop version of the SD card image and imported the config again. Which actually worked :-)

So now I have a VNC connection and can see the network graph, but it shows way too much lines to be actually useful. How do I read this? Can I see a list of nodes which have weak connection to the mesh?

When I close the desktop app viewer the whole network does not work anymore? Does everything really hinge on this one desktop app keep running on the raspberry pi? Shouldn't this essential service be implemented as a low level service?

Sometimes everything works as it should. Other times, some lights get stuck in the ON position and I have to override them with a remote or power cycle them (which is quite a lot of work).

The Tradfri remotes work about 70-80% of the time. Some button presses just don't get registered or the lights don't react. 

Some Alexa commands work, some others do nothing or take 5-10 seconds or just affect some lights in a group while others don't react at all.

Scenes barely work. I have one scene which involves 15 spots and I set all the light to OFF and only three to 1%. When I activate the scene, all lights stay on but only the three go to 1%. It's like the "off" position was not saved in the scene for the other lights.

Other lights constantly forget the white temperature. I have to tell Alexa "bedroom warmer" to get them back to warm lights, and Alexa always repeats "something isn't working" (when will this bug be fixed?). The color temperature is not saved in the scenes.

There are no wifi from neighbors around, just my own Unifi-Network with four AP AC Pro or HD and most of the devices on 5.8GHz. The 2.4GHz wifi is on channel 1 while the zigbee net is on channel 25 (so, far away).

Mesh-Routing:
> There are still issues, see #1261, but broadcasts don’t use routing and are unaffected by these issues.

OK so I installed three Tradfri-repeaters throughout the house. They appear in the network overview but sometimes lights 2 meters away from a repeater don't have a connection to a repeater but go to lights on the other side of the house?

For example, the Xiaomi open/close sensor on the very top of the network graph picture has a repeater 1 meter away from it but only has one red connection showing?

[image]

ChrisPrefect, 2020-06-04:
BTW: I still have another 60 devices in the basement on an original Tradfri-Gateway. Should I move them over to the Conbee-Network too? Maybe they interfere and only one network with nearly 200 nodes is better? How many nodes does Conbee/Phoscon handle reliably?

closed 2020-07-11 by stale[bot]