dresden-elektronik/deconz-rest-plugin #2692: Unable to identify / pair via touchlink

white91wolf, 2020-04-15:
hey,

after i am struggeling hard to pair hue lights i now try to pair them via touchlink. So i get the touchlink response like this:
192.168.2.117/api/2970Exxxxx/touchlink/scan

```
{
    "lastscan": "2020-04-15T20:49:03",
    "result": {
        "1": {
            "address": "0x0017880100xxxxx1",
            "channel": 11,
            "factorynew": true,
            "panid": 65534,
            "rssi": -37
        },
        "2": {
            "address": "0x0017880100xxxxx2",
            "channel": 11,
            "factorynew": true,
            "panid": 65534,
            "rssi": -29
        }
    },
    "scanstate": "idle"
}
```
so the two hue lights are getting listed right! but now i want to pair them with my gateway.
I am following the doc https://dresden-elektronik.github.io/deconz-rest-doc/touchlink/#identify
and i get stuck at the identify point. What is the <id> in the POST /api/<apikey>/touchlink/<id>/identify
of my two results?

i have already tried: [1,2][0x0017880100xxxxx1, 0x0017880100xxxxx2], [17880100xxxxx1,17880100xxxxx1]

for example: 192.168.2.117/api/2970Exxxxx/touchlink/17880100xxxxx1/identify

[
    {
        "error": {
            "address": "/touchlink/17880100xxxxx1/identify",
            "description": "resource, /touchlink/17880100xxxxx1/identify, not available",
            "type": 3
        }
    }
]

generally how can i add the lights via the REST api? I mean the identify will only make the light blink but wont pair it?!

thanks a lot

ebaauw, 2020-04-15:
> What is the id in the POST 

RTFM, http://dresden-elektronik.github.io/deconz-rest-doc/touchlink/:
> Note id must be one of the indentifiers which are returned in the scan result.

> generally how can i add the lights via the REST api?

deCONZ doesn’t support touchlink linking, only identify and reset.  After reset, search for new devices, by doing a POST to `/api/key/config` With a body of `{"permitjoin": 60}`, whatever time you want to search for.

white91wolf, 2020-04-15:
> > What is the id in the POST
> 
> RTFM, http://dresden-elektronik.github.io/deconz-rest-doc/touchlink/:
> 
> > Note id must be one of the indentifiers which are returned in the scan result.
> 

so what's the id in my result example i posted above?

i installed the docker od deconz and i am using the api on port 80 but also the web interface - so i tried a lot of variations. and also like in my result above the factorynew value is true - so a resett would not really make a difference.
But thanks so far i will try tomorrow the REST way you mentioned.
And via the web interface i set the searching time to 3 minutes - but the lights just dont get found.

SwoopX, 2020-04-15:
You still don't read.

The manual says:
_id_ must be **one** of the indentifiers which are returned in the scan result.

Since the ID is not the address, use 1 OR 2.

white91wolf, 2020-04-16:
> You still don't read.
> 
> The manual says:
> _id_ must be **one** of the indentifiers which are returned in the scan result.
> 
> Since the ID is not the address, use 1 OR 2.

Okay sorry now i understand.. nur Like i wrote i already tried that the result is simmilar a 404 Not found:

```
[
    {
        "error": {
            "address": "/touchlink/1/identify",
            "description": "resource, /touchlink/1/identify, not available",
            "type": 3
        }
    }
]
```

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix