dresden-elektronik/deconz-rest-plugin #2693: The Tradfri remote does not work as expected.

Kristian8606, 2020-04-16:
The left and right arrows on the remote control do not change the color of the bulb all other buttons function correctly.
The bulb has white and warm light.
Is it possible to repair it?

Smanar, 2020-04-16:
It depend how you have make the "pairing"
If you have put the bulb in the same group than the remote, it's a "direct connexion", in this mode it can work without deconz, but you can't have color.

Kristian8606, 2020-04-16:
I made a group with one bulb in which I added the remote and configured the buttons, but the knobs do not change color.
Could you tell me exactly how to configure the remote and the bulb to work properly?

Smanar, 2020-04-16:
It seem fine.
I m not using Phoscon but can you show me the screen with your button configuration pls ?

Kristian8606, 2020-04-16:
[image]

DrMullings, 2020-04-24:
I have the exact same issue, but after like 3 times new binding and setting up both arrow buttons at least made the light warmer

Kristian8606, 2020-04-24:
Are you saying that you made the setting as in the picture and you can change the color temperature?

DrMullings, 2020-04-24:
Yes, but only to warmer and only after 3 retries by completely setting it up new.
So this was more of a random (halfway) success

jonassam1, 2020-05-10:
I see the same behaviour.... On/off, dim up and down works, but both left/right buttons both makes the light warmer. So the left button does not work as it should.