dresden-elektronik/deconz-rest-plugin #2694: Philips Hue Outdoor Motion Sensor Sensitivy

x2005254, 2020-04-16:
Hi,

I added 2 Outdoor motion sensor, but one of them is too sensitive when trees moved with the wind and I saw that wi can change sensitivy on Deconz Gui, but I don't know the possible Value :

[image]

someone know what is the possible value, on Hue app I saw in videos that there is 3 level :

[image]

But I don't have hue gateway and app

Thanks in advance.

Smanar, 2020-04-16:
Hello, have you tried value 0/1/2.
But take care, the device need to be awake to receive command.
Or you can try with API

`curl -H 'Content-Type: application/json' -X PUT -d '{"sensitivity": 1}' http://IP:PORT/api/KEY/sensors/ID/config`

x2005254, 2020-04-16:
thanks it works :)

jensflorian, 2020-04-22:
Would be nice to have a config option in Phoscon as well.

ebaauw, 2020-04-22:
Sensitivity goes from 0 (least sensitive) to the value of sensitivity max.  I think the outdoor sensor has actually five sensitivity levels.

x2005254, 2020-04-22:
Yes, the outdoor have 0 to 4 level of sensitivity.

Thanks