dresden-elektronik/deconz-rest-plugin #2695: Unable to add Xbee as a Resource

pandalanax, 2020-04-16:
Hello!
With the profile mentioned in #2645 the XBEE is able to join the Network.
However when i scan for new resources it does not get detected.
I am using a Conbee II (which is showing up as a light resource as expected)

I had a look at #2472 and had trouble understanding this part:

> The REST API resources should be created when searching for new devices through the API. If not, read the Basic cluster (on the gateway node) or read the descriptors (on the gateway and XBee). To pair the XBee, configure it as follows (from https://www.dresden-elektronik.de/faq.html#deconz):

Help appreciated!

ebaauw, 2020-04-16:
Did you set `SC` to `FFFF` as mentioned in #2472?  The FAQ has the wrong value, causing not all channels to be scanned.

pandalanax, 2020-04-17:
Yes i did that.
Just to make sure ...

> The REST API resources should be created when searching for new devices through the API.

.. means  scanning via touchlink via `POST /api/<apikey>/touchlink/scan` right?

ebaauw, 2020-04-17:
No, POST `/api/`_apikey_`/config` with a body of `{"permitjoin": 60}`.

deCONZ doesn't support touchlink pairing, only scan, identify and reset.

pandalanax, 2020-04-17:
Response from the POST request you mentioned.
```
{
  "error": {
    "address": "/config",
    "description": "method, POST, not available for resource, /config",
    "type": 4
  }
}
```

ebaauw, 2020-04-17:
Does it appear in the GUI?  Did you request one of the descriptors while the permitjoin was active?
Issue the POST, then select the node for the XBee in the GUI and press F5.  The virtual LED on the node should blink blue.

pandalanax, 2020-04-17:
After the request and pressing F5 on the node the virtual LED has the following behaviour:
1. lights up yellow 
2. conbee node blinks blue
3. xbee switches from yellow to blinking red

[image]

ebaauw, 2020-04-17:
Looks like the XBee left the network, and the coordinator can no longer communicate with it.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix