dresden-elektronik/deconz-rest-plugin #2697: [Device support request] Heiman remote control HS1RC-M

SwoopX, 2020-04-16:
[image]

**Power cluster**
[image]

accelle17, 2020-05-24:
Hi @SwoopX, do you have a confirmation from the web app that you were able to add the remote? I followed your instruction by adding 0x0501 but can only see the device on the VNC/phoscon. Not able to see any new device or events but can see it appears on the logs

SwoopX, 2020-05-24:
Yeah, you get a confirmation, at least I got one. However, that remote is pretty much unusuable. It uses IAS ACE commands which deconz has no logic for. And that's unfortunately nothing to be added like in 5 mins. Thinking about bending to towards a regular remote, if that's feasible.

SwoopX, 2020-05-27:
I've made an update so it can be used as a regular remote to the extend feasible. It should be available as a switch and allows 4 short press button events.

accelle17, 2020-06-01:
great, thanks. I'll test it as soon as it releases and appears on hass add-on since I don't have any dev environment for it.

joschaschultze, 2020-06-07:
I've added the new endpoint and can see the device in deconz-gui but there is no home automation software which shows the remote. I've tried domoticz and iobroker.

SwoopX, 2020-06-18:
Closing this as PR has been merged. Supported with .78

closed 2020-06-18 by SwoopX

rweise, 2020-09-06:
Also here, why it is closed? i can see the remote in Deconz-gui as conected  but not in Phoscon and not in the iobroker