dresden-elektronik/deconz-rest-plugin #2699: IKEA color bulb have wrong color

freekode, 2020-04-17:
Hi, I moved from Tradfri hub to conbee + deCONZ and paired my two bulbs.
All it controller via home-assistant.
Now when I set some color for bulb I see that bulb actually has slightly different color than I choose.
Warm orange now is more red and white is more yellow.

So, how it can be fixed?

SwoopX, 2020-04-18:
Sounds familiar https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2304#issuecomment-573443753

freekode, 2020-04-18:
Yeah, sounds the same.
Tried locally resolve merge conflicts and apply #2402, no luck, too many changes on master 😞

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix