dresden-elektronik/deconz-rest-plugin #2700: Everytime new password logout of Phoscon

netzbus, 2020-04-17:
I have the problem that when I log out of phoscon app and log in again it always wants to define a new password.
Whatever password I choose, phoscon app states "login denied". The only way to came arround is to delete "/home/pi/.local/share/dresden-elektronik/deCONZ/zll.db".
Then defining new password is working but after logout the drama reconvenes

Hardware is:
Raspberry 3 with Raspbian
Conbee 2 Stick

SwoopX, 2020-04-18:
Are you using a reverse proxy or "localhost" to connect to Phoscon?

netzbus, 2020-04-18:
Indeed. I am connect via SSH Portforwarding to Localhost

SwoopX, 2020-04-18:
I've seen funny behaviour when using localhost to connect to Phoscon. In my cases, whatever password I set is totally irrelevant. When I login again, one single random character is accepted.

When I recently went briefly through the code, I saw that loopback interface should have been ignored (which does not always seem to be the case). Can you use the outfacing IP?

ebaauw, 2020-04-18:
I think it might actually be an API "feature" to accept any username (API key) for connections from `localhost`.  I know for sure, you can create an API key without unlocking the gateway over a local connection.

netzbus, 2020-04-18:
I am talking about the app not the API. 
Any Idea how to come around the locahost problem?

ebaauw, 2020-04-18:
Phoscon uses the API to connect to deCONZ.  I don't know Phoscon in detail, but I think the localhost problem is probably caused by the API.