dresden-elektronik/deconz-rest-plugin #2701: Device Support - WAXMAN LeakSMART v2

jtscott, 2020-04-17:
Hello I am opening this issue to request device support for the LeakSmart Smart Home Sensor 2.0. This is a water leak sensor and I hope to provide as much info as I can:

- [User Manual](https://leaksmart.com/storage/2019/10/LeakSmart-Smart-Home-Water-Sensor-User-Manual.pdf)
- [ZHA Device Handler](https://github.com/zigpy/zha-device-handlers/tree/c4fb37523e88f8ae743a9bc6c63f63090c539263/zhaquirks/waxman)

Device Screenshots
[image]
[image]
[image]
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4495601/debug.txt)

Please let me know if I can provide any addition information, and thank you for your support and consideration.

SwoopX, 2020-04-18:
Please also provide a screenshot of the node itself from deconz GUI where you can see the available clusters. Thanks!

jtscott, 2020-04-18:
Sure, is this correct?
[image]

SwoopX, 2020-04-18:
It is, thanks. I'll let you know it I need anything else.

jtscott, 2020-04-18:
Yes, I removed and re-added the node to be sure. When I had it paired to my wink hub a year ago it provided both leak detection and temperature.

SwoopX, 2020-04-18:
Hm, then it moght be the unknown cluster. I'll check it out later.

ebaauw, 2020-04-18:
Or a hidden _IAS Zone_ cluster.  Wouldn't be the first device to do that.

SwoopX, 2020-04-22:
@jtscott Could you please also read the attributes from power configuration cluster and provide the screenshot? Wanna have the battery information right. Thanks!

jtscott, 2020-04-22:
Sure thing; Here is what I am getting from it, it is updating the data from what I can see when I click read but not sure why the rest is grey. Device takes 3x AAA's physically:

[image]

SwoopX, 2020-04-22:
Hm, can you read to try battery percentage once again (by double clicking on it) The 2 values seem not to fit each other.

jtscott, 2020-04-22:
This looks weird for sure! I clicked read a couple times and it comes back 200 and says "reading done". I tried to read config as well and it comes back as not_found. Same with voltage, reading at 46 and read config not_found.
[image]

SwoopX, 2020-04-22:
It's just the "read" button. "read config" is to read the reporting configuration, which the device currently not has unless you did all the binding manually.

jtscott, 2020-04-22:
No definitely outside my area of expertise there, I'm just capable of gui button clicks :) Reading still comes back 200: 
[image]

SwoopX, 2020-04-22:
Hm, that still feels a bit strange 3x1.5V is not 46 (or 4.5) ;) Brand new batteries maybe?

Regardless, I'd rather go for the voltage for the time being. We can change that later on if required.

jtscott, 2020-04-22:
Yes had to add new batteries last week, as this thing chirps loud like a fire alarm when its low.

SwoopX, 2020-04-22:
Ok, we go for the percentage then as it doesn't require for any changes then. Just please keep an exe on if, if it changes over time.

ebaauw, 2020-04-22:
> This looks weird for sure! 

No it doesn’t, check the ZCL spec:
- _Battery Percentage Remaining_ is in 0.5%, so a value of 200 means 100%.
- _Battery Voltage_ is in 0.1V, so a value of 46 means 4.6V.

If the reporting configuration returns _NOT FOUND_, it simply hasn’t been set.  Quite normal for factory new devices.  Try writing some values and reading them back.

jtscott, 2020-05-24:
Hi Gents, I reset and re-added my sensor on the new release today. Looks like the leak part of the sensor is still missing. Is there something else I can provide you to help finish this device integration?

SwoopX, 2020-05-24:
Just taken another looks and that one's gonna be interesting. You have 2 unknown clusters there, where the FC02 is a manufacturer specific cluster. You probably only have a chance to get some info out of it if you can compile an additional plugin and get some reasonable data back, which is not guaranteed.

Now, for 0b02, I recall I checked the specification already but haven't found anything. Just taken another look and this time, I was lucky as the zigbee specification contains a typo for that cluster. It's apparently "EN50523 Appliance Events and Alerts", so presumably responsible for the alerts. This is definitive a first timer here. Getting the cluster data in is the easy job, evaluating how it works in conjunction with the capabilities at hand is a different thing.

So long story short, we need to investigate if and how we can make it play.

SwoopX, 2020-05-30:
Can any of you guys support in compiling the plugin and do some tests once all required parts have been implemented?

jtscott, 2020-05-30:
Sure I can give it a shot, I'm probably savvy enough to follow specific instructions if you have those available?

SwoopX, 2020-05-30:
You find the instructions in the readme of this repo. You just need to make sure you also installed the deconz-dev package. Hopefully, you don't run any virtualization (docker), otherwise it will be more difficult.

Give it a shot with the current state repo to see if it works out for you.

jtscott, 2020-05-31:
switched over to dev today using the readme with no problem. I deleted the node from deconz and re-paired the sensor after factory reset but I don't see a difference. I looked over the clusters vs the old screenshots and it all looks the same.

Let me know what else I can do?

SwoopX, 2020-05-31:
Hey, you were a bit too fast for me here ;) I'll prepare a seperate repo of mine for that particular testing so we stay kinda clean on the official one.

Take this one: https://github.com/SwoopX/st_playground.git
I already implemented code for water sensor creation. Currently working on the pick-up of the corresponsing alert commands...

jtscott, 2020-06-01:
Woops sorry, haha! Ok gave it a shot this afternoon and looking really close! Attached is a debug where I removed the old sensor, re-added, and triggered two wet events on the sensor. The sensor makes a really annoying beep when it is wet so I am sure it fired.

[waxman_dbg.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4713066/waxman_dbg.txt)

Device looks the same in deconz gui, but reflecting more info in webui:
[image]

Also showing up good in HASS, but of course no wet/dry events and sensor always reports dry right now:
[image]

Hope that helps!

SwoopX, 2020-06-02:
Alright, thanks for trying. No relevant output there, besides sensor creation.

Take 2 is out. Some more debug output and extended by one additional command. Please run it this time with `--dbg-info=2 --dbg-aps=1> dbg`, which hopefully gives me more insights in the command used.

jtscott, 2020-06-02:
Sure thing! Here is a new debug output; I removed the device prior to updating and then re-added the device after compiling and triggered 2 water events again.

[dbg_waxman_20200602.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4719850/dbg_waxman_20200602.txt)

SwoopX, 2020-06-02:
Thanks. Sorry, but gotta ask: Have you pulled the repo once again and compiled? I don't see any debug output I'd expect...

jtscott, 2020-06-03:
Hmm that's weird, yeah I removed prior cloned repos and re-cloned and re-compiled again. I can re-flash my SD back to before dev today and try again.

SwoopX, 2020-06-03:
Just puzzled that I don't see anything from that particular cluster. Not sure anything else might be required upfront. However, I see at z2m that they seem to have added binding for the cluster in question.

Maybe give it a try when the sensor is paired:
**Manually bind a cluster**
In deconz GUI, select Panels and then Bind dropbox.
For your device as well as the coordinator (blue node), press the most right bullet to expand the available clusters.
For the coordinator: Drag&drop 01 Home Automation Endpoint as destination.
For the device: Select the windows covering cluster as source, also drag&drop it. Take note that this must be a server cluster (colored blue).
Press `bind`.

jtscott, 2020-06-03:
ok did a couple things and it looks like the binding worked!

New debug after re-flashing to pre-dev, installing dev, and then recompiling from your repo here: 
[dbg_waxman_20200603.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4726094/dbg_waxman_20200603.txt)

Binding 0B02
[image]

Binding FC02
[image]

Now... after the fact I thought I should have just did one cluster binding first to see what exactly happens but let me know if that's needed and I can repeat the process above :)

Thanks!

SwoopX, 2020-06-03:
Better now 👍 Will check it later in detail.

SwoopX, 2020-06-03:
Looks like we got a winner here:

```
03:31:07:839 APS-DATA.indication srcAddr: 0xfc14, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B02, lqi: 255, rssi: -46
03:31:07:840 [APPL] Command is cluster command
03:31:07:840 [APPL] Alert count: 1
03:31:07:840 [APPL] Alert Structure: 4481
03:31:07:840 [APPL] Device currently alerting...
03:31:07:840 [APPL] Settings sensor state alert to 'true'...
03:31:07:842 Websocket 192.168.1.15:58846 send message: {"e":"changed","id":"12","r":"sensors","state":{"lastupdated":"2020-06-03T20:31:07.840","water":true},"t":"event","uniqueid":"00:22:a3:00:00:0f:e6:4a-01-0b02"} (ret = 159)
```

```
03:31:17:832 APS-DATA.indication srcAddr: 0xfc14, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B02, lqi: 255, rssi: -46
03:31:17:832 [APPL] Command is cluster command
03:31:17:832 [APPL] Alert count: 1
03:31:17:832 [APPL] Alert Structure: 385
03:31:17:832 [APPL] Device ceased alert...
03:31:17:832 [APPL] Settings sensor state alert to 'false'...
03:31:17:834 Websocket 192.168.1.15:58846 send message: {"e":"changed","id":"12","r":"sensors","state":{"lastupdated":"2020-06-03T20:31:17.832","water":false},"t":"event","uniqueid":"00:22:a3:00:00:0f:e6:4a-01-0b02"} (ret = 160)
```

jtscott, 2020-06-04:
Yep looks like 0x0b02. I unbound the cluster and made the sensor wet and it did not trigger a state change. Rebinding 0x0b02 worked again and it triggers a wet state change.

SwoopX, 2020-06-05:
Alright. Updated the code so the binding should take place automatically. If you could test once more and it succeeds, I guess we're done here.

jtscott, 2020-06-05:
Ok all done and can confirm works good. I reflashed to pre-dev, recompiled your repo, then re-added the sensor. Works out of box now!

Thanks so much, really appreciated!

arjewell23, 2020-06-05:
Hi Guys,  I've been following this because I have a few leaksmart sensors.  How can I get this to work?  Do I need to upload some files, etc?

arjewell23, 2020-06-05:
I have my HA instance installed on a VM,  Will this still work?

SwoopX, 2020-06-05:
It should, if you don't use docker.

Added the code to the PR. Enjoy!

SwoopX, 2020-06-18:
Closing this as PR has been merged. Supported with .78

closed 2020-06-18 by SwoopX

wkearney99, 2022-10-29:
Anyone had one of these sensors trip a false detection of moisture?  I've had one running for about a year.  The battery level was down to about 10%.  It detected water (and there is none).  I changed out the batteries, and the gateway now sees the new AAAs as 100%, but it's still reading 'Wet'.

Is there anything else I have to do to clear the error?

wkearney99, 2022-10-29:
And nevermind, I wet the unit and it alarmed and when I dried it the condition went back to Dry.

So apparently batteries alone will not clear it?