dresden-elektronik/deconz-rest-plugin #2702: can't connect to ConBee II after firmware downgrade

Adminiuga, 2020-04-18:
Hi All,

Is there a way to reset NVRAM settings on Conbee II ? 
I was running 0x26530700 alpha firmware on Conbee II in a test system and it was running fine. Then decided to change the network key, PAN ID, Ext PAN ID through `deCONZ` linux app, in GUI it was showing correct network key, but Zigbee sniffing was showing transport key of `0xcccccccccccccccc` so I've decided to downgrade firmware back to stable `0x264a0700` and it flashed successfuly, but that was the last time I could connect to Conbee through deCONZ or Homeassistant ZHA integration.

I can flash any firmware just fine and have tried `0x26490700` and `0x26420700` and everytime if flashes successfuly (tried both on LInux or Win10) but I cannot connect to it.
Have tried:
 - deCONZ on Win10 64bit
 - Debian virtual box VM
 - Pi3 Hassio deCONZ plugin

any ideas how to fix it?

Adminiuga, 2020-04-18:
Here's last flashing attempt on Win
[image]

Adminiuga, 2020-04-18:
and on the same machines (win/linux/hassio) Conbee I is connecting just fine.

Smanar, 2020-04-18:
Have you take a look on the configuration in deconz ?
Those setting are written on the key. You can co,mpare with this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/768

Adminiuga, 2020-04-18:
I can access phoscon but can't reset anything because gateway is not connected.

Adminiuga, 2020-04-18:
And  I can't take a look at the settings (otherwise i'd just restore the backup) because I can't connect with deCONZ to ConBee II. The ConBee II is shown in the gateway drop-down list, but when I press connect it seems to try but never succeeds

Smanar, 2020-04-18:
You can't access to the conbee trought deconz (you don't need to press connect) ?
What is the result of the command "GCFFlasher_internal -l" ?

Adminiuga, 2020-04-18:
Correct.
Gccflasher -l shows the serial port of ConBee. I can flash firmwares just fine, but can't connect to gateway in deCONZ

SwoopX, 2020-04-18:
You may start deconz with `--dbg-info=2 --dbg-error=1` to see if there's anything suspicious at the beginning.

Adminiuga, 2020-04-19:
debug log
https://gist.github.com/Adminiuga/18b07c10a9c4dee7c51eb11623bfdca7

Adminiuga, 2020-04-19:
added strace log. Essentially I see the `DEVICE_STATE` command issued, but no reply whatsoever
```
3205  openat(AT_FDCWD, "/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK|O_CLOEXEC) = 18
....
3205  write(18, "\300\7\0\0\10\0\0\0\0\361\377\300", 12) = 12
3205  ppoll([{fd=18, events=POLLIN}], 1, {tv_sec=0, tv_nsec=3000000}, NULL, 8) = 0 (Timeout)
3205  ppoll([{fd=18, events=POLLIN}], 1, {tv_sec=0, tv_nsec=3000000}, NULL, 8) = 0 (Timeout)
3205  ppoll([{fd=18, events=POLLIN}], 1, {tv_sec=0, tv_nsec=3000000}, NULL, 8) = 0 (Timeout)
3205  ppoll([{fd=18, events=POLLIN}], 1, {tv_sec=0, tv_nsec=3000000}, NULL, 8) = 0 (Timeout)
```

Adminiuga, 2020-04-19:
I've tried it on a different laptop with Debian Buster as host OS, different extension cables but got exactly the same behavior. Can't connect to Conbee II, but Conbee I works just fine

manup, 2020-05-26:
Hi can you share a screenshot of deCONZ Network Settings .

If possible also flash the latest firmware version 26580700 included in 2.05.77. it has improved startup after update.

Adminiuga, 2020-05-26:
> Hi can you share a screenshot of deCONZ Network Settings
I can't get deCONZ connected to ConBee so can't get to the network settings. 

I did try `deCONZ_ConBeeII_0x26580700.bin.GCF` when it was posted in "ikea" issue two weeks ago, but not with 2.05.77 deCONZ 
let me try with newer deCOZN

Adminiuga, 2020-05-26:
Ok, installed 2.05.77 and flashed 2658 again:

[image]

It is detected by `deCONZ` but won't connect to it
[image]

Pretty much same output in deCONZ log as before
```
10:51:08:836 found node plugin: libde_rest_plugin.so - REST API Plugin
10:51:08:837 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:51:09:800 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:51:09:808 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:51:09:808 COM: /dev/ttyS0 :  (0x0000/0x0000)
10:51:09:808 COM: /dev/ttyS1 :  (0x0000/0x0000)
10:51:09:851 UPNP socket not bound, state: 0
10:51:09:853 discard sensor state push for 1: state/dark (already pushed)
10:51:09:854 discard sensor state push for 1: state/status (already pushed)
10:51:09:861 discard sensor state push for 1: state/lastupdated (already pushed)
10:51:12:361 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
10:51:12:388 Serial com connected
10:51:12:915 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:12:916 device state timeout ignored in state 2
10:51:13:968 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:13:968 device state timeout ignored in state 2
10:51:14:114 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:51:14:115 COM: /dev/ttyS0 :  (0x0000/0x0000)
10:51:14:116 COM: /dev/ttyS1 :  (0x0000/0x0000)
10:51:14:116 GW update firmware not found: 
10:51:14:818 Announced to internet http://dresden-light.appspot.com/discover
10:51:15:019 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:15:019 device state timeout ignored in state 2
10:51:16:069 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:16:069 device state timeout ignored in state 2
10:51:17:119 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:17:120 device state timeout ignored in state 2
10:51:18:132 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:18:132 device state timeout ignored in state 2
10:51:19:132 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
10:51:19:132 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:19:133 device state timeout (handled)
10:51:19:196 Serial com disconnected, reason: 1
10:51:19:197 device disconnected reason: 1, index: 0
10:51:19:631 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
10:51:19:631 device state timeout ignored in state 1
10:51:21:131 wait reconnect 15 seconds
10:51:22:132 wait reconnect 14 seconds
10:51:23:132 wait reconnect 13 seconds
10:51:24:132 wait reconnect 12 seconds
10:51:24:167 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:51:24:168 COM: /dev/ttyS0 :  (0x0000/0x0000)
10:51:24:168 COM: /dev/ttyS1 :  (0x0000/0x0000)
10:51:24:168 GW firmware update select USB device
10:51:24:169 GW update firmware not found: deCONZ_ConBeeII_0x26490700.bin.GCF
10:51:25:132 wait reconnect 11 seconds
10:51:26:132 wait reconnect 10 seconds
10:51:27:131 wait reconnect 9 seconds
10:51:28:131 wait reconnect 8 seconds
10:51:28:631 networkState: CC_ReconnectNetwork
10:51:28:632 start reconnect to network
10:51:29:132 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
10:51:29:133 wait reconnect 7 seconds
10:51:30:131 wait reconnect 6 seconds
10:51:31:133 wait reconnect 5 seconds
10:51:32:132 wait reconnect 4 seconds
10:51:33:131 wait reconnect 3 seconds
10:51:33:883 failed to reconnect to network try=1
10:51:34:132 wait reconnect 2 seconds
10:51:34:166 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:51:34:166 COM: /dev/ttyS0 :  (0x0000/0x0000)
[... 21 lines cut ...]
10:51:41:631 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:51:41:631 device state timeout (handled)
10:51:41:631 MASTER kill cmd 0x08 (ERROR)
10:51:41:695 Serial com disconnected, reason: 1
10:51:41:695 device disconnected reason: 1, index: 0
10:51:42:133 wait reconnect 15 seconds
10:51:42:134 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
10:51:42:135 device state timeout ignored in state 1
10:51:43:131 wait reconnect 14 seconds
10:51:44:131 failed to reconnect to network try=3
10:51:44:131 wait reconnect 13 seconds
10:51:44:163 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:51:44:165 COM: /dev/ttyS0 :  (0x0000/0x0000)
10:51:44:165 COM: /dev/ttyS1 :  (0x0000/0x0000)
10:51:44:165 GW firmware update select USB device
10:51:44:165 GW update firmware not found: deCONZ_ConBeeII_0x26490700.bin.GCF
10:51:45:132 wait reconnect 12 seconds
10:51:46:133 wait reconnect 11 seconds
10:51:47:131 wait reconnect 10 seconds
^C10:51:47:756 shutdown after signal(2)
10:51:47:811 saved node state in 0 ms
10:51:47:827 sync() in 15 ms
10:51:47:827 REST API plugin shutting down
10:51:47:827 DB save zll database items 0x000000C1
```

Adminiuga, 2020-05-26:
same with `--dbg-prot=2`
```
10:59:55:844 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
10:59:55:873 Serial com connected
10:59:55:873 zmMaster::setState() state: O -> S
10:59:56:350 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:59:56:350 zmMaster::setState() state: S -> I
10:59:56:350 device state timeout ignored in state 2
10:59:56:825 zmMaster::setState() state: I -> S
10:59:57:174 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
10:59:57:174 COM: /dev/ttyS0 :  (0x0000/0x0000)
10:59:57:174 COM: /dev/ttyS1 :  (0x0000/0x0000)
10:59:57:175 GW update firmware not found: 
10:59:57:302 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:59:57:302 zmMaster::setState() state: S -> I
10:59:57:302 device state timeout ignored in state 2
10:59:57:777 zmMaster::setState() state: I -> S
10:59:57:901 Announced to internet http://dresden-light.appspot.com/discover
10:59:58:255 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:59:58:255 zmMaster::setState() state: S -> I
10:59:58:255 device state timeout ignored in state 2
10:59:58:730 zmMaster::setState() state: I -> S
10:59:59:207 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:59:59:207 zmMaster::setState() state: S -> I
10:59:59:207 device state timeout ignored in state 2
10:59:59:682 zmMaster::setState() state: I -> S
11:00:00:158 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:00:00:158 zmMaster::setState() state: S -> I
11:00:00:159 device state timeout ignored in state 2
11:00:00:635 zmMaster::setState() state: I -> S
11:00:01:133 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:00:01:133 zmMaster::setState() state: S -> I
11:00:01:133 device state timeout ignored in state 2
11:00:01:632 zmMaster::setState() state: I -> S
11:00:02:133 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
11:00:02:133 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:00:02:133 zmMaster::setState() state: S -> I
11:00:02:133 device state timeout (handled)
11:00:02:212 Serial com disconnected, reason: 1
11:00:02:213 device disconnected reason: 1, index: 0
11:00:02:632 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
11:00:02:632 zmMaster::setState() state: I -> O
11:00:02:632 device state timeout ignored in state 1
11:00:04:132 wait reconnect 15 seconds
11:00:05:132 wait reconnect 14 seconds
11:00:06:132 wait reconnect 13 seconds
```

Adminiuga, 2020-05-26:
Unfortunately I don't have screenshots of the original parameters, but what I've noticed:
1. the network key was displayed correctly in GUI (i was trying to replace Silabs coordinator with ConBee, so updated pan-ids, nwk keys etc in gui)
2. Existing devices had trouble communicating and upon re-joining I've noticed that network ket packet had a key of all `0xcccccc...`
3. it was running alpha firmware `0x26530700` 
4. after downgrade to `0x264a0700` I was no longer able to connect to ConBee. But I know others were able to downgrade successfuly from same alpha firmware.

SwoopX, 2020-05-26:
I always felt pitty for the poor souls experiencing this but interestingly, I had the very same experience with my Raspbee I last week. I activated OTAU plugin and then, deconz crashed like 2 minutes afterwards and didn't came back to life. I'm running a bare metal Pi without any virtualization or containerized software and also didn't make any configuration changes.

Everything seemed to be fine, except for deconz wasn't able to start. Permissions were good, reboots haven't brought any change.

[image]

Also tried device pinning, still no change. While trying `GCFFlasher -l`, I surprisingly ran into the error described here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1087

In the end, re-flashing the exact same firmware I was running made it come back to life again...

Adminiuga, 2020-05-26:
Well, i've tried reflashing different versions and same version, but still no signs of life :(

manup, 2020-05-29:
Hmmm really strange, the connection between ConBee and deCONZ should work regardless of the actual network parameters.

In any case please stick to deCONZ_ConBeeII_0x26580700.bin.GCF for now since it is is more stable on startup compared to earlier versions.

From the logs it seems that the serial connection can't be established. But I wonder why GCFFlasher works, looks like it is executed as root.

Can you provide the output of: 

`ls -l /dev/ttyACM0`

Since deCONZ runs as non-root it needs to have access to that device.

Adminiuga, 2020-05-29:
On a phone right now. But it t is not the device permission issue, I'll get screen shots later. ConBee I ( yes a different device, but being assigned same permission by udev) works fine. 
Also, in the strace log provided earlier in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2702#issuecomment-615996020 you can see that port was opened successfuly ( got file descriptor 18) and it wrote something which I assume is device state command tonport, but haven't got a reply. exactly same behavior with zha. And same behavior on another machine.

Adminiuga, 2020-05-29:
> From the logs it seems that the serial connection can't be established. But I wonder why GCFFlasher works, looks like it is executed as root.
> 

My guess, the nvram content is corrupted breaking firmware start -- before it went down, network key was displayed correctly in deCONZ, but the wrong one was used to operate the network) 
Bootloader is independent of the main firmware and doesn't really care about nvram, so it operated successfuly.

Adminiuga, 2020-05-29:
> added strace log. Essentially I see the `DEVICE_STATE` command issued, but no reply whatsoever
> ```
> 3205  openat(AT_FDCWD, "/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK|O_CLOEXEC) = 18
> ....
> 3205 [...]
\300 is 0xc0 frame delimeter
\7 is 0x07 is device state command

Adminiuga, 2020-05-29:
screenshot of device permissions for ConBee I and ConBee II
[image]

deCONZ can connect to ConBee I, but not to ConBee II

Adminiuga, 2020-05-29:
And running deCONZ as root, produces the same result:
```
Script started on 2020-05-29 09:02:03-04:00 [TERM="xterm" TTY="/dev/pts/0" COLUMNS="104" LINES="47"]
root@debba:/home/lex# deCONZ --dbg-prot=2 --dbg-info=2
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libpng warning: iCCP: known incorrect sRGB profile
09:02:56:594 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
09:02:56:638 CTRL. 3.27.209:02:56:712 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
09:02:56:712 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
09:02:56:712 COM: /dev/ttyS0 :  (0x0000/0x0000)
09:02:56:712 COM: /dev/ttyS1 :  (0x0000/0x0000)
09:02:56:712 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
09:02:56:755 parent process bash
09:02:56:755 gw run mode: normal
09:02:56:755 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
09:02:56:756 DB sqlite version 3.27.2
09:02:56:756 DB PRAGMA page_count: 36
09:02:56:756 DB PRAGMA page_size: 1024
09:02:56:756 DB PRAGMA freelist_count: 0
09:02:56:756 DB file size 36864 bytes, free pages 0
09:02:56:756 DB PRAGMA user_version: 6
09:02:56:756 DB cleanup
09:02:56:756 DB create temporary views
...
09:02:56:761 found node plugin: libde_rest_plugin.so - REST API Plugin
09:02:56:762 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
09:02:56:766 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
09:02:56:775 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
09:02:56:775 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
09:02:56:775 COM: /dev/ttyS0 :  (0x0000/0x0000)
09:02:56:775 COM: /dev/ttyS1 :  (0x0000/0x0000)
09:02:56:839 discard sensor state push for 1: state/dark (already pushed)
09:02:56:840 discard sensor state push for 1: state/status (already pushed)
09:02:56:841 discard sensor state push for 1: state/lastupdated (already pushed)
09:02:58:784 void MainWindow::devConnectClicked() choose com /dev/ttyACM0
09:02:58:820 Serial com connected
09:02:58:820 zmMaster::setState() state: O -> S
09:02:59:296 zmMaster::setState() state: S -> I
09:02:59:297 device state timeout ignored in state 2
09:02:59:772 zmMaster::setState() state: I -> S
09:03:00:249 zmMaster::setState() state: S -> I
09:03:00:249 device state timeout ignored in state 2
09:03:00:727 zmMaster::setState() state: I -> S
09:03:01:204 zmMaster::setState() state: S -> I
09:03:01:204 device state timeout ignored in state 2
09:03:01:660 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
09:03:01:660 COM: /dev/ttyUSB0 : FT230X Basic UART (0x0403/0x6015)
09:03:01:660 COM: /dev/ttyS0 :  (0x0000/0x0000)
09:03:01:660 COM: /dev/ttyS1 :  (0x0000/0x0000)
09:03:01:679 zmMaster::setState() state: I -> S
09:03:02:158 zmMaster::setState() state: S -> I
09:03:02:158 device state timeout ignored in state 2
09:03:02:342 Announced to internet http://dresden-light.appspot.com/discover
09:03:02:633 zmMaster::setState() state: I -> S
09:03:03:134 zmMaster::setState() state: S -> I
09:03:03:134 device state timeout ignored in state 2
09:03:03:632 zmMaster::setState() state: I -> S
09:03:04:131 zmMaster::setState() state: S -> I
09:03:04:131 device state timeout ignored in state 2
09:03:04:632 zmMaster::setState() state: I -> S
09:03:05:131 zmMaster::setState() state: S -> I
09:03:05:131 device state timeout (handled)
09:03:05:204 Serial com disconnected, reason: 1
09:03:05:204 device disconnected reason: 1, index: 0
09:03:05:633 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
09:03:05:633 zmMaster::setState() state: I -> O
09:03:05:633 device state timeout ignored in state 1
09:03:06:632 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
09:03:07:636 wait reconnect 15 seconds
09:03:08:633 wait reconnect 14 seconds
09:03:09:632 wait reconnect 13 seconds
09:03:10:634 wait reconnect 12 seconds
^C09:03:11:532 shutdown after signal(2)
```

manup, 2020-05-31:
Hmm really strange, can you please contact me at mpi@dresden-elektronik.de.
We'd like to send you a replacement would be good if you can send us this one back so we can take a deeper look.

The NVRAM might be scrambled but it should still connect, maybe it's a hardware issue with the connectors or something similar.

Mimiix, 2020-06-12:
Is this solved?

Adminiuga, 2020-06-12:
Sent my stick to de for analysis. Maybe should wait for the post mortem

Mimiix, 2020-06-12:
Sounds good to me!

ICTSMART, 2020-06-27:
Hi All,

Today I've upgraded my Raspbee2 to the latest firmware. After rebooting, network lost, devices lost....etc.etc.
Yup, just read the page on how to solve......which doesn't solve the issue.
Can I downgrade my Raspbee2 to the older version 0x26580700? 
On the firmware download page I can't find that version for the Raspbee2.

Thanks

SwoopX, 2020-06-27:
Please take a debug log first with `--dbg-info=2 > debug` for like 5 mins. We never geg any chance for investigation as we need debug data and people get hecktic.

You can downgrade with the Conbee II FW.

Adminiuga, 2020-06-27:
Also, the opening post is for a different issue, where deCONZ was not able to connect to ConBee. 
In your case it seems like it can connect, but just doesn't see any traffic from the devices

ICTSMART, 2020-06-27:
> Please take a debug log first with `--dbg-info=2 > debug` for like 5 mins. We never get any chance for investigation as we need debug data and people get hecktic and change things.
> 
> You can downgrade with the Conbee II FW.

I am happy to provide you with a 5 min. What should I change in the Home Assistant config for the deCONZ integration to parse the --dbg-info=2 parameter?

Mimiix, 2020-06-27:
@ICTSMART instead of hijacking another issue, could you please open your own bug report?

closed 2020-07-18 by Adminiuga