dresden-elektronik/deconz-rest-plugin #2704: Xiaomi (Aqara / Lumi) sensors unreachable after few hours / days

smesguich, 2020-04-20:
Hello,
I have some issues with my Xiaomi devices sinces I've switch them on conbee II GW. I'm explain :
I have some routers Legrand (Wall Switchs, Shutter SW and Wall Plug).
Each days, one of my Xiaomi at least end devices leave the Zigbee Network (Unreachable on GUI, and grey on deCONZ app). Battery are for all >60%. I've notice that majorly Aqara T° devices.

If any one have an idea ?

smesguich, 2020-04-20:
Screenshot : 
[image]

SwoopX, 2020-04-20:
It may be of further help you you could let us know which deconz version and firmware you're running...

smesguich, 2020-04-20:
> 
> 
> It may be of further help you you could let us know which deconz version and firmware you're running...

Oups. Sorry.
[image]
I precise that I'm running deCONZ via DOCKER but I was facing the same issue with dedicated Virtual Machine

yves19, 2020-04-20:
Xiaomi Aqara sensors are devices power by battery. So they hibernate while not transmitting and wake up only at regular time to transmit their information.
That's why they are considered as unreacheable by deCONZ but with no damage for the rest of operation.
The only way to force a wake up of these sensors is to quikely press their including button and you should see them again ... for a while till they go in sleep mode again

SwoopX, 2020-04-20:
@yves19 In general, you're absolutely right. However, from the REST API data perspective, some devices were not seen for more than one day. And that's the rather unusual part.

smesguich, 2020-04-20:
> 
> 
> Xiaomi Aqara sensors are devices powered on battery. So they hibernate while not transmitting and wake up only at regular time to transmit their information.
> That's why they are considered as unreacheable by deCONZ but with no damage for the rest of operation.

Yup I know this but .... Usually, I have at least few update per days. Here, I have some sensors without any update for more than 1 day. And I had re-included some because it wasn' updated sinces 3 days !

> The only way to force a wake up of these sensors is to quickely press their including button and you should see them again ... for a while till they go back in sleep mode.
> By the way I am interested in the use of your CT WC on one hand and of your CT Congélateur as well

Well, WC is just to check a if this room is very cold sometimes or just a strange feeling related to floor tile. For Congélateur (aka Freezer for English readers), just about monitor temperature inside and be triger an alert if t° grows up (related to electricity stop or just basic breakdown)

> 
> 
> @yves19 In general, you're absolutely right. However, from the REST API data perspective, some devices were not seen for more than one day. And that's the rather unusual part.

Yes ! Thats why I've open an issue

smesguich, 2020-04-20:
An other case from a T° sensors (CT Entrée cf api logs) which I've trigger manually an update with the physical button wich didn't updated in GUI, in API and instead in log I didn't found it ...
I had re-include it yesterday night !
I'm running the docker with those options, IDK if I get enought logs or not
- DEBUG_INFO=1
- DEBUG_APS=0
- DEBUG_ZCL=0
- DEBUG_ZDP=0
- DEBUG_OTAU=0

Smanar, 2020-04-22:
Hello, 60% for a sensor, mean empty battery.
BTW the "ct cuisine" and "ct entree" are at 100%, do theses device use a legrand device as router, or use direct connexion ? I had same problem too, I seem Xiaomi can have same problem with not updated device than Osram (problem is from xiaomi)

Have you updated theses devices ? If you can't, have you a way for I send you something to try ?

smesguich, 2020-04-22:
> 
> 
> Hello, 60% for a Xiaomi sensor, mean empty battery.
> BTW the "ct cuisine" and "ct entree" are at 100%, do theses device use a legrand device as router, or use direct connexion ? I had same problem too, I seem Xiaomi can have same problem with not updated legrand device than Osram (problem is from xiaomi)
> 
> Have you updated theses devices ? If you can't, have you a way for I send you something to try ?

I had re-include everything yesterday after deleting all.
Yesterday graph :
[image]

Today graph : 
[image]

They are using for some Legrand router as you can see.

smesguich, 2020-04-22:
Also, CT Cusine as you can see didn't show any version of firmware :
[image]

I can try to include them to Xiaomi GW before re-switch them on Conbee also ?

smesguich, 2020-04-22:
Hum... After analysing screens, that looks like you hypothésis : Legrand Routeurs are the issue.
Because most of All direct connection was still alive and all undirect connection (with 1 hop by Legrand routers) looks like disconected.

I had some device without any one message since yesterday 7pm ...

Smanar, 2020-04-22:
Yep, I know this issue ^^ I have commented too, I had same problem with a switch.

But pipiche have made lot of test since this date, I can try to ask him, but with updated device this defaut disappears.

It's for that I m asking if you already have updated them, I know you have the legrand gateway, but if you don't want to dismount your installation I can give you something for try.

smesguich, 2020-04-22:
> 
> 
> Yep, I know this issue ^^ I have commented too, I had same problem with a switch.
> 
> But pipiche have made lot of test since this date, I can try to ask him, but from my memory with updated device this defaut disappears.
> 
> It's for that I m asking if you already have updated them, I know you have the legrand gateway, but if you don't want to dismount your installation I can give you something for try.

Well, you can explain me, if will see !

<french>j'utilise encore la gateway legrand pour certaines zone ou si une instabillité du réseau apparait, Madame risque d'être très faché ... donc j'attends d'être full complient sur deCONZ pour tout migrer :) ceci dit, je peux faire quelques test sans trop faire raler madame ! </french>

Smanar, 2020-04-22:
Too late ^^.
I have just edited last post.

It seem you have more recent firmware version than me , strange you still have this problem.
I will try to ask pipiche if he have some information about that.

smesguich, 2020-04-22:
Well I see. Based on https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Legrand-Netatmo-corner.md, I've noticed something very strange. In pairing mode, only 1 wired switch led is green. Other are purple.

Smanar, 2020-04-22:
Haaaa ok, nice ^^ it's perhaps a good discovery.

Ok so if you have paired your device before the version 73 with updated device the led is purple for sure.
Purple mean device paired but not all feature, for exemple you can miss the dimmer or the power consumption, and I hope better mesh.
You just need to include them again, without deleting them, like that, you wil not loose your setting.
Make a try with 1 router first, don't need to dismount all your installation for that.

But fimware version 2b, it's strange you have used old deconz version ... If you have green LED after pairing it will be ok.

They are lot of way to exchange ^^, you can try to contact pipiche on discord if you have legrand zigbee question, but he is realy busy atm, they are making a new zigate model.

smesguich, 2020-04-22:
Sorry but ... Version 73 of what ?
Just Re-include them on Conbee ?

Smanar, 2020-04-25:
Oups, sorry, haven't see the update

So if I make a resume.
- CT sbd and CT H2 use "int plafonnier sdb" 
- CT balcon / CO CH2  use "volet roulant CH2"
- CT salon / CO2 fenetre SDB use direct connexion <<< Realy strange for theses one and with batterie at 100%
- CO CH1 / have double connexion on graph so IDK what is the router.
- CT entree use "int plafonier entrée".

"Int plafonnier entree" is green or purple" ?

BTW pipiche give me this link, nothing for help, but some information in it https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

smesguich, 2020-04-26:
Hello !
Sorry for the long time !
After few try and re-inclusion all my AC Legrand Devices are Green when deCONZ is in paring mode 😄 
Now, I can try to deploy some Xiaomi battery modules and make a new test

Smanar, 2020-04-26:
But I m not optimistic :(

The device that are directly connected and leave too is not a good thing, I realy don't have explanation for that.

Smanar, 2020-04-26:
Ok need to check, but same issue, all Xiaomi device have leaved in less than 2h.

smesguich, 2020-04-27:
Ouch .... Okay so problem is between Legrand and Xiaomi relaying broadcast information about alive.

From my side, all Legrand stuff (which is connected to Conbee) is green now. I'm just quit frustrated now ... 

I will try to re-include on or two module and see what's happens

Smanar, 2020-04-27:
Legrand battery device can be displayed as unconnected, but still working, remember legrand don't make periodic return, so a device can stay somes days without saying it's alive.

I m collecting information ATM, it seem there is the same problem on z2m > https://github.com/Koenkk/zigbee2mqtt/issues/2878

Waiting for zigate return.

But yes, so much frustrated, on my side I have bulbs and switches, and as switch are "inside the wall" sensor prefer use bulb, so I have never see this problem, except the first time.

That's definitively not a good thing, I have so much aquara too.

Smanar, 2020-04-28:
Ok so bad news.

**I m almost sure Legrand is not able to make router for Xiaomi devic**e, I have confirmation from pipiche, he have make try on his side, same result.
Just for information, I haven't checked, so I can't be sure, but like for other manufacture, I think the problem is from Xiaomi too.

akenad, 2020-04-29:
Xiaomi sensors are supposed to work directly with Gateways Xiaomi.
They are not supposed to be routed.
So they can work good on ConBee/RaspBee but directly.
As you can have several GW Xiaomi, you can have several RaspBee for instance.

Smanar, 2020-04-29:
Yep, but it's the only one manufacture with this problem.
It's just because they don't respect zigbee protocol (they are not the only one, but I think the worst), and I m not sure they are not supposed to be routed, there is Xiaomi router too.

And for me it's not usefull having several gateway, as only can be used as coordinator.

akenad, 2020-05-01:
I have 2 GW (first and second floor) on the same Jeedom and it works fine.

smesguich, 2020-05-01:
I don't want to use the Xiaomi GW that's why I've buy a Conbee. Because Xiaomi GW was too closed sources and not open to other manufacturer ...

Right now, I've tried to move on all my Xiaomi sensors on their original GW in order to check if there is any FW update to correct the issue.

Also, @Smanar something strange that I was thinking : you told me that somes routers are working why Xiaomi sensors, but your also saying that is du to non respect of zigbee protocol Xiaomi side. How we could know that isn't from Legrand implementation ?

smesguich, 2020-05-04:
After research and as far as I know, the issue came from the Xiaomi sensors because after sleeping, they doesn't send the right "permit join" request after exit from sleeping.

So I have few options :

- Changing my whole (12Temp sensors and 7 Windows sensors ...) Xiaomi devices to a Zigbee complient brand ([Konke](https://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/#Principe_de_fonctionnement_et_cluster) by exemple, or testing Tuya / HEIMAN / BlitzWolf)
- Create a CC2530 router (as sayed here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-622307275 or here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1488) and make sur that Aqara devices use only this router which is tolerent for Xiaomi zigbee implementation
- Find a way to extract, correct and OTA firmware update (Harcode ...)

SwoopX, 2020-05-04:
@smesguich What makes you think the devices send any permit join request? And what about the OTA, for Xiaomi? Is there a newer firmware?

smesguich, 2020-05-04:
@SwoopX just here : https://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/#Le_capteur_drsquoouverture_de_porte and agreagation nigh reading GH issue and forums about Xiaomi Aqara and the way that they implement Zigbee

Smanar, 2020-05-04:
@akenad yep, but you are using 1 or 2 network ?
If you are using 1 network, only 1 GW is used as coordinator, the second one is just a router, so theres is better router than a complete GW.
If you are using 2 network, it s not the better solution, because you will have a better mesh with 1 big network than 2 smaller.

About the mesh bug, I haven't find technicak information but I have find that

> With latest versions Xiaomi devices should stay online at least when connected directly to the gateway and most routers with a few exceptions (see below).
> 
> A bug was recently discovered where some Zigbee 3.0 routers ask end-devices to rejoin if they time out in their neighbour table. The Xiaomi end-devices don't implement that correctly and will just leave the network and repairing them to the network is required.
> 
> Related issues: #1488 and Koenkk/zigbee2mqtt#1474
> 
> Currently known routers which have this problem are:
> 
>     OSRAM US Sylvania plug
>     Müller Licht tint lights
> 
> Note the bug is in the Xiaomi devices not the routers. As a workaround I suggest powering off all problematic devices during pairing so that the Xiaomi devices select a "working" parent device.

ATM there is no better firmware for Xiaomi.

Unfortunately you are using full connected switch installation, so no zigbee bulb, but if you have unused plug, you can test the ikea repeater (cheaper), but I have good return for the "Master router"

SwoopX, 2020-05-04:
> 
> 
> @SwoopX just here : https://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/#Le_capteur_drsquoouverture_de_porte and agreagation nigh reading GH issue and forums about Xiaomi Aqara and the way that they implement Zigbee

@smesguich permit join != rejoin. A permit join is issued from the coordinator while a rejoin is issued from the device. If you read thoroughly, you see that it's about loosing connectifity and not about waking up from deep sleep state. That's a small but major difference.

However, the rejoin matter is true and so that "issue" stays.

smesguich, 2020-05-04:
> @smesguich permit join != rejoin. A permit join is issued from the coordinator while a rejoin is issued from the device. If you read thoroughly, you see that it's about loosing connectifity and not [...]

Yeah, your right.

smesguich, 2020-05-05:
Just thinking : Instead of building a CC2530 based router, May I use my second conbee as router only ?

Smanar, 2020-05-05:
There is some tries, for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1934
But you need a raspberry or another machine.

The conbee have a better range than usual device, but for me it's something "overkilled". The "master router" will be cheaper and more stable (autonomous, no setting needed, no OS, ... )

The "master router" will be easier to use too, you just have to plug or unplug it, to enable/disable it.

But on your situation (and I think it will be soon for me), I will first make try with turning off all legrand device, and pairing all Xiaomi. So the maximum devices will use direct connection or others routers (will be easier for me, because of my bulbs). And only after I will search for adding routers in "white zone".

smesguich, 2020-05-05:
Hm .... okay so maybe overkilled. Also, it's better to have a 2nd conbee as DEV version or in case of hw failure. I've order all I need to build a router CC2530 so 😄

closed 2020-07-04 by stale[bot]