dresden-elektronik/deconz-rest-plugin #2705: Error when setting Aqara motion duration value

Voodoo2man, 2020-04-21:
I have a Szene which sets a duration of an Aqara Motion sensor to another value.
If that szene is triggered I get this errors from the deconz adapter.
Same if I set the value manually.
Maybe it is since the update to CS Controller 3

Iobroker with all updates (latest)
Cs controller 3 (latest)
Node js 12 (latest)

13172) This value will not be set in future versions. Please report this to the developer.
--
(13172) State value to set is invalid for deconz.0.Sensors.9.duration: The state is missing the required property val!

Smanar, 2020-04-21:
I m not sure the problem is from deconz.
Have you the complete command (seen in log if you can).
The error message seem to say you miss the value in your command.

Voodoo2man, 2020-04-21:
I haven‘t changed anything but updating.
Before updating CS-Controller and NodeJS I haven‘t had this Warning.
There is a value (duration) that I set manually in the Aqara Motion Sensor, and when setting it, the warning appears.

Voodoo2man, 2020-04-21:
Just found this topic on iobroker.net
https://forum.iobroker.net/topic/32513/js-3-0-controler-deconz-adapter-fehler-neustarts

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix