dresden-elektronik/deconz-rest-plugin #2707: Device support Bticino Living Now

zinoalex, 2020-04-22:
Hi guys, would be possible to expand the support to the Bticino Living Now devices?
I was able to include (only on channel 11 and after device's firmware upgrade) and use:

K4027C shutter controller (light in Phoscon)
F20T60A: DIN energy monitor mistakenly called Legrand DIN power consumption module (not visible in phoscon)
K4531C: power consumption module (light in Phoscon)
K4027CW: wireless shutter command (not visible in phoscon)
In all those devices the manufacturer is wrong reporting Legrand.

Unable to use:
K4003C: connected deviator
This one is recognized and included in the network after FW upgrade to v22 and channel 11 but it doesn't report the actual status. Toggle mode in deconz works. Any ideas to do that? I read a post about zigbee2mqtt and they were able to control this device.

Smanar, 2020-04-22:
Hello, xas my question here ^^ https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2674#issuecomment-614096237

How are you using deconz for this device ?
The api haven't toogle fonction, it have only "on" and "off" ?

And it doesn't report status because this device is not in whitelist for binding. I can add it in the code (just one line to add).

To activate the reporting, waiting for code modification, you can use deconz

First make the bind
- Display the bind dropbox
- drag and drop the cluster 0006 from your device to the "source box"
- drag and drop the endpoint 01 (the title of the node") from the conbee to the "destination box"
- press "bind" button

Then enable attribute reporting
- display the cluster 0006 on/off in deconz
- on the table on the bottom, select the attribute 0x0000
- double click, press "read" and check values, 1 /300 is not bad value.

Or if you can compile the code, I can make modification now, but I need to understand how it work, because there is no real "on" and "off", if you use the output 2 it will be "off" but if you use the output 1 it will be "on".

And if you are using a not connected deviator on the same circuit, it can be on and off without action on the device.

zinoalex, 2020-04-26:
Sorry Smanar I've missed the post.
I'm using the GUI accessible via VNC latest release (.75) in docker with Conbee II.
[image]
I have these functions: Off, On and toggle. With off nothing happens, with On and Toggle the device switches between pin 1 and 2.

The result after you instructions:
[image]
the value stays "false" either with power on pin 1 and 2

Smanar, 2020-04-26:
Ok, so if the value stay false, it will be a problem, I will check z2m to see If I found more information

But can you use the device ATM with the API ? There is no toogle command on it, how are you using the device ? If you seend multiple "on" command it toogle the output ?

Smanar, 2020-04-26:
Ok so I have found something
>The device has a single command button that commutes the phase(L) between the 2 outputs (two-way-switch/va-et-vient) so one of the 2 output is always 'ON', and a little reset/paring button (not accessible when the cover is mounted/during normal operation) + LED. As you can see the device always reports back the state of OFF (but the light is ON)

But 
>Both my K4003C are responding correctly to ON OFF commands and reports back correct state **even when switching ON/OFF from non smart switches** (two way wiring)

So I think the device can check the consumption or current presence, are you making try with a bulb or other thing that consumme current ?

zinoalex, 2020-04-28:
Update: I connected the device with a bulb and another deviator and actually works correctly. Now the ON/OFF status changes between true and false, so it must have a kind of consumption meter inside.

Smanar, 2020-04-28:
Realy nice hardware, seriously, was a good idea to use something hardware to guess if it's on or of.
IDK why we haven't it in france, perhaps because we don't use neutral for switch.

So, nothing to change ? I just need to enable the binding/report automaticly ?

zinoalex, 2020-04-28:
I think so, anyway I connected all the 4 deviators that were previously connected to the Bticino gateway and everything is working great with Home Assistant, the trick was to connect a load in order to change the on/off flag.
yes is actually a very smart piece of hardware, the only limitations are that you need the Bticino Living Now series which is quite new and a neutral wire.
Merci anyway Smanar

Smanar, 2020-04-28:
All is already working ?
You have done the binding/report manually, or they don't need it for state return ?

zinoalex, 2020-04-30:
No need for manual binding

Smanar, 2020-04-30:
Ok thx, so I let the code as it.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

lambu76, 2020-10-25:
Hi all, I have a strange problem with a BTicino connected deviator (N4003C) and Conbee II.

I have a fresh installation of deCONZ  (last version V2_05_86).

Once configured to use the channel 11 and reset the deviator to default factory I can see inot the zigbee network, connected t the Conbee II,  but after a few minutes the binding is lost and it's not anymore possible to add it.

let me underline between the automatic adding into the network and the lost of bidning I didn't perform any action...just waiting for the lost of bidnding.

Any ideas ?

thx

Smanar, 2020-10-26:
Yep, probably a bad inclusion.
Legrand device join easily a network but if the procedure is not respected they leave immediatly after.

What if your inclusion procedure ? remember you need to use the hidden button not the deviator itself.

lambu76, 2020-10-26:
mmmmm.....maybe I missed something...I'm very rookie on this topic.

Could you please remember which is the proper procedure ?

Personally I just pressed the reset button in order to restore default factory settings into the deviator, but nothing else.

It has been joined to the Zigbee network.

thx

Smanar, 2020-10-26:
I have only Legrand device on my side, not Bticino but for me it's 

Long press to reset, ignoring the first blink, , release, then short press.
You have a led blinking, you need to have green color.

lambu76, 2020-10-26:
ok thx I will try.

Just a quick question.

Do you have the original gateway in place ?

Currently I'm trying such kind of test without the Legrand/Bticino gateway.

thx

Smanar, 2020-10-26:
I have it, but in a box, not connected.
But it's not the same procedure, it s easier with the original gateway.

lambu76, 2020-10-26:
unfortunately with BTicino if I take pressed button for reset I have the first blink, but after that if I release and repress the button nothing else happen.

I need someone else who have already integrated a BTicino component.

lambu76, 2020-10-26:
> No need for manual binding

Hi, can I kindly ask you your help ?

thx

Smanar, 2020-10-26:
You need to ignore the first blink, and keep pressed.

If @zinoalex have another procedure ?

zinoalex, 2020-10-27:
Hi lambu,
first of all you need to pair your deviator with the original Bticino gateway in order to update the firmware (at least v22, it takes quite a few time), than reset to factory pressing the button (keep it pressed). Switch off the bticino gateway and all the other zigbee gateways (Hue, ecc) except the conbee II. Turn on the discovery as a light and start to press the deviator reset button. Remember that is not immediate, vnc page is very useful to check if it is included or not. Most of my network problems were from the double zigbee network (hue and conbee) + wifi channel overlapping. Now i use 3 deviator, 3 blind controller and a couple of scenery button since months with no issues at all. Anyway check the channels https://community-assets.home-assistant.io/original/3X/2/f/2fa03d15dd0a963475c38cb03d776b8e8750c9b5.png

Smanar, 2020-10-27:
But it works too without the gateway ?
For Legrand not updated device are easier to include. I can send too the firwmare to update it with deconz but need to check if I have it, and the actual deconz version have some problem with OTA.

Smanar, 2020-11-01:
Yeah ^^, I m already on it > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532

By hazard can you compile the code to test it ? The other user  don't have unix machine.

lambu76, 2020-11-01:
unfortunately I installed deconz on Windows and Hass.IO as virtual appliance by downloading a VHDX virtual disk.

I took a look to the post mentioend by you, but I'm not getting if you are mentioning my same identical problem or something else.

thx

P.S.
if you are italian you can contact me on my mail:

user: lambu76
domain: gmail.com

Smanar, 2020-11-01:
Ha yep sorry, the discution have started on discord, but it s a similar issue, if you want to make it working you need to use the "open" command, HA use "bri" and this command is buggy witht his device.

I will make the PR in blind mode, but I think it will be nought, I have mimic the other shutter device.

lambu76, 2020-11-01:
I'm a little bit lost but happy to know that can work...could you please support me in order to instruct HA to send the proper command?
Which conf file I have to update?

Thx a lot

Smanar, 2020-11-01:
I don't know at all HA.
But you can't update your files on Hassio to test the new version.

You need to change the command done by the app to make try with the not corrected version, or use direclty the API with Curl for exemple.

lambu76, 2020-11-01:
Where did you find that the reason for this bug is a wrong command send by HA ?

Maybe I could find someone else in some community to help me with HA, since unfortuantely you are not using it.

thx

Smanar, 2020-11-01:
no it s not HA fault.
HA send 2 commands in one, "bri" and "on" or "lift" and "open", and this device have one of them reversed.

I don't remember, but for one HA use only 1, but both for the other side, so the device faild.

But it s not HA fault, the command need to be reversed in deconz code, or as workaround, you can edit the command send by HA, waiting for the correction.

lambu76, 2020-11-01:
thx for the explanation....I'm trying to find a way to do that....I'm pretty rookie on such topics

Smanar, 2020-11-01:
I m sorry but I don"t know at all HA, but if you find the command used by it, just remove the "bri" or "lift" part.

But the modification will be in the next deconz version I think, not a critic one

testorp, 2020-12-08:
hello i also have a L4003C and I'm trying to pair with deconz but no way am I able to, it never comes up on the ZigBee network, even though I'm able to use it with it's own gateway L4500C

any suggestions how to do so?

this is what is running on Phoscon
[image]

Any help aprreciated

PAOLO

Smanar, 2020-12-08:
Ha, I think I miss this one.
Do you have the model id of the device visible in deconz (on the GUI) ?

testorp, 2020-12-08:
No deCONZ is not able to pick it up

Smanar, 2020-12-08:
The L4003C is the same device than the K4003C ?
Wich one inclusion procedure are you using ?

- Long press (around 10s)
- keep pressed on first blink
- Led go red fixe
- short press
- if LED don't go green, retry.

testorp, 2020-12-08:
@Smanar L4003C is Bticino Living Light series
[image]

K4003C is Bticino Living Now series
[image]

testorp, 2020-12-09:
@Smanar don't know what I've done but in the end I've been able to get the switch in deCONZ, more or less what you have described.
what you are explaining is that red led fixed means the switch must be in factory reset, then pressing the reset switch again it goes green led and it's able to pair

Bticino has 2 lines of connected devices Living Light and Living Now see picture below

[image]

here is a document that explains it from Bticino
[91735.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5666274/91735.pdf)

Smanar, 2020-12-09:
Yeah the deviator is not used in my country, but rely nice device too.
But I haven't all model id for bticino, for light it s not a problem because they are working natively, but you can miss some feature on other like power mesurement, but the good news is legrand repect zigbee rule, so every time it s just a "1 line edition" modification.

testorp, 2020-12-09:
@Smanar Any chance of having the switches identified as switches and not light?

Thx
Paolo

Smanar, 2020-12-09:
Nope this is not a bug, it s normal.
Battery switches are sensor (sub section switch).
Wired switches are lights.

You send order to the deviator to turn on the light, you don't use this position to make something else.

testorp, 2020-12-09:
Is it possible for me to change this behavior in code?

If I use the gateway from Bticino they appear as switches in HomeKit

Smanar, 2020-12-09:
It depend of your third app, but this device realy need to be managed as a light, not a switch.
So I don't think you can change that on "core" perhaps on the third app that display it ?

testorp, 2020-12-09:
Can you please explain why?

Please, trying to understand.

I'm using Phoscon and Homebridge-Hue to go directly to Apple HomeKit so I can't change anything

Smanar, 2020-12-09:
For exemple, you can use a group light, and turn all the light in it in same time, you put in it the deviator, but not a battery switch.
Wired switches are working like light. You can use a battery switch to turn on the deviator, but not the reverse.

testorp, 2020-12-09:
Ok the point is I want to use both connected switches and connected lights

What you are saying makes more sense if lights are not connected

lambu76, 2021-01-03:
Hi guys, I'm back on this topic after a long time.

After last updates from Deconz BTicino shutter works fine in opening and closing  direction...very good.

Now I have two questions.

1) How can I force the opening or closing to a specific position ? currently I can only drive the full opening (value 100) or full closing (value 0). Any other values are not set and in case I open only paritally a shutter manually the values grabbed is always 50.

2) is it possibile to have the light embedded into the device swithced on ? Currently is swtiched on (red color) only when it's not connected to zigbee network, then it becomes for  short while on green when I'm joinning the network and then it's definitely switched off.

thx to all

Smanar, 2021-01-03:
@lambu76 I m interested by your version ^^ https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3876#issuecomment-747613679
There is soon a new version for this device, don"t want to broke something.

1- this device don't support set level command
2 - https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3876#issuecomment-747447209

zinoalex, 2021-01-03:
@Smanar  yes it's true but then I don't know how to interface it with home assistant. I use the dockerized version of deconz.

@lambu76 
1: works fine with bticino app after "calibration" but there is no way to do it in deconz, I used timers with home assistant (not a very elegant way but it works)
2: option 2: true

lambu76, 2021-01-03:
> @Smanar yes it's true but then I don't know how to interface it with home assistant. I use the dockerized version of deconz.
>
> @lambu76
> 1: works fine with bticino app after "calibration" but there is [...]

1: damn.....can you share me some tips about how to use timers ? Not elegant solution but working is better than NO solution

2: I'm getting "WRITING FAILED" on some devices (shutter or normal lights always bticino) and some other not...any ideas ?

lambu76, 2021-01-03:
Let me rephrase.

I'm getting "WRITING FAILED" on ALL my shutters L4027C, but Option 2 is working fine on on light device L4003C.

Firmware on both devices is 00b...as per Phoscon webapp indication.

Ideas ?

zinoalex, 2021-01-03:
In my case It changed after firmware upgrade with the original gateway.
About timers:
take the time for the shutter to completly close than make an automation
with the half of the seconds for example for the 50% position and so on.
you can use home assistant automations or either node red. you can also
make an input boolean switch to actuate a script

lambu76, 2021-01-03:
@zinoalex 
got it for the timers....I will try it.

For the upgrade, my problem is that I don't have an original gateway....is there any way to upgrade to latest firmware without it ?

Which is the last firmware version available ? Currently I have 00b on all my BTicino devices.

If I have an original gateway is there any way to upgrade to latest firmware without deconnecting devices from deconz, connecting to original gateway and the reconnecting to deconz ?

Smanar, 2021-01-03:
Someone with the same version than you had exactly same problem.
Led not working and impossible to use deconz for update, the upgrade server was locked to 0x0000000

[image]

lambu76, 2021-01-03:
So the original gateway is neeed....damn, but it would be not a big deal if I can use it without removing the evice from zigbee network managed by deconz...but I don't know if it's possible.

So my question is....supposing to have a BTcino device connected to zigbee network managed by deconz device, whihc are the steps to upgrading to last firmware by using a the original gateway ?

Smanar, 2021-01-03:
Sorry I m seing only 2 solutions
- find someone with the bticino firmware and use deconz, but if you are in same situation than my previous post, it will not work
- disconnect your device from deconz network to connect it to bticino one, and just wait 24 h.

zinoalex, 2021-01-03:
I had to remove the device from deconz and then repair It with original
gateway, It takes more or less one hour to do the upgrade. I was unable to
update the firmware with deconz even if my server was not 0x000

zinoalex, 2021-01-03:
As far as i remember the switch has 4 pins, meaby with some reverse
engineering and an Arduino programmer we are able to download the firmware

Smanar, 2021-01-03:
You can just get them using this method https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Retreiving-Legrand-Firmware.md

But it have a little changed ATM, for the file name.

muchasuerte, 2021-02-11:
hello @lambu76 , 

did you set any specific value on network for channel mask or APS ack or Security mode?

I have a Relè 3584C but I cannot include it, I have not found any way to get a green led.

[image]

muchasuerte, 2021-02-11:
I did some test by setting the channel mask to 11 and 15 and also checking APS Acks and now I can see the device, the problem now it that I lost all other devices :(

[image]

[image]

[34 more replies]

closed 2022-07-26 by github-actions[bot]