dresden-elektronik/deconz-rest-plugin #2708: Disconnected devices - better visability

PrimaryInc, 2020-04-22:
Hi @manup,

Just a thought, when a device is disconnected it show up as just a bit "grayed" out in DeConz app.
Would it be possible to get this a bit more visible? 
Example, connected devices has green "header" and disconnected has a red "header"?

Similar to the where the device has the color yellow on screenshot?

[image]

PrimaryInc, 2020-05-11:
No Comments?

PrimaryInc, 2020-05-28:
Okay, so guess it was a bad idea :)

Smanar, 2020-05-28:
lol.
No, you are right for me, but I think they have more important in the "TODO list'

PrimaryInc, 2020-05-28:
I guess that it would take @manup approx 10min to change it. But as you say, probably not a priority :)

closed 2020-07-04 by stale[bot]