dresden-elektronik/deconz-rest-plugin #2709: Im lost - please HELP!

X015, 2020-04-22:
Hello Guys and Girls,

i just got my ConBee II yesterday and tried to set it up all.
Im running it in a Docker Container which runs on my Debian (Open Media Vault) Bare Metal Server.

I tried to pair Xiaomi Door Sensor but that didnt work tried another one that also didnt work.
Then i tried a Tradfri Remote Control which also didnt work.

I got one Door Sensor to appear in the deconz GUI (VNC) but not in the Phoscon GUI

If you need anything else please let me know.

Over & Out

Smanar, 2020-04-22:
You have a door sensor in the GUI but not in phoscon ? Have you a screenshoot of the deconz GUI pls ?

How are you making inclusion ? By phoscon ?

Are you sure deconz is running ? Have you a firmware number in Phoscon/Gateway ?

X015, 2020-04-22:
Hello, yes i get a firmware number and it says its the latest
Here is the log from the Docker Container:

`
17:01:58:149 GW firmware version is up to date: 0x264a0700
17:02:43:145 Current channel 15
17:02:43:153 Device TTL 5099 s flags: 0x7
17:03:43:144 Current channel 15
17:03:43:151 Device TTL 5039 s flags: 0x7
17:04:43:144 Current channel 15
17:04:43:152 Device TTL 4979 s flags: 0x7
17:05:43:146 Current channel 15
17:05:43:155 Device TTL 4919 s flags: 0x7
17:05:58:155 GW firmware version: 0x264a0700
17:05:58:155 GW firmware version is up to date: 0x264a0700
17:06:43:144 Current channel 15
17:06:43:152 Device TTL 4859 s flags: 0x7
17:07:43:144 Current channel 15
17:07:43:152 Device TTL 4799 s flags: 0x7
17:08:43:145 Current channel 15
17:08:43:153 Device TTL 4739 s flags: 0x7
17:09:43:145 Current channel 15

`

This is going on all the time..
I have no Devices in the Phoscon GUI but here is a Image of my deconz GUI (VNC)

[image]
[image]

SwoopX, 2020-04-22:
Is the ConbeeII connected to a USB3.0 port? Do you use an extension cable?

Hav you brought the devices close to the ConbeeI while pairing?

X015, 2020-04-22:
I tried booth USB 3.0 and USB 2.0 no one did work yes im right next to the ConBee with the Sensor.
I'll try it with an extension cable tomorow.

Thanks for any feedback guys!

SwoopX, 2020-04-22:
You may check https://phoscon.de/en/support#faq and go through the steps under "Troubleshooting for ConBee II with Linux"

Smanar, 2020-04-23:
I see nothing strange on your setup ....

I think too the USB extension cable is a good thing to try, you can too check your deconz configuration, you can compare with this one for exemple 
https://user-images.githubusercontent.com/383386/59607539-cf4f5600-9113-11e9-975d-758d1846e9c3.png

X015, 2020-04-23:
So i tried it now with a extantion cable still no luck i tried to pair 2 Xiaomi Door Sensor and 1 Tradfri Remote Contorl. They appear in the Deconz app:

[image]

But the Phoscon website says it cant find anything.

SwoopX, 2020-04-23:
The are not paired correctly! How did you do it? Please always pair via Phoscon.

X015, 2020-04-23:
> 
> 
> The are not paired correctly! How did you do it? Please always pair via Phoscon.

I did pair them over Phoscon but like i said they appear only in the deconz app Phoscon always says it couldn't find any Switch/Sensor

SwoopX, 2020-04-23:
Hmm. Have you tried that already? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2709#issuecomment-618076967

X015, 2020-04-23:
Yes i did everything was as described.

Smanar, 2020-04-24:
And on several devices, on 2 manufactures ...
You can try directly trought deconz, without using phoscon, but for me it's not good.

If you click on the first circle, you can relaunch some command (while awaking the device), if they succed they can finish the inclusion.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix