dresden-elektronik/deconz-rest-plugin #2710: devices keep dropping

hosting-io, 2020-04-22:
hi, using dconz on a ubuntu server box.  Issue is, wired devices (lights) stay connected. Any battery devices, after a day or so, lose all connectivity, and the only way to get them back is to search and re-add them.  from reading around on here, it seems its because they go to sleep, is there any way to force a check/ping or something that they stay connected.

when connecting them to the xiamo aqara hub, they don't drop offline, but I cant connect the hub to Home Assistant, so am stuck in a loop.

SwoopX, 2020-04-22:
Can you post a screenshot from deconz GUI?

hosting-io, 2020-04-22:
[image]

SwoopX, 2020-04-22:
That's Phoscon.

hosting-io, 2020-04-23:
ok, i deleted everything and have re-added a few items this morning -- here is the result, I will add a new image once it updates and everything drops

[image]

hosting-io, 2020-04-23:
this is connected directly to an ubuntu box, USB 2 and a 2M USB ext lead. its currently sitting on the floor in a big open room. family room door that disappeared is about 2 metres across and 2 meters up from the USB dongle

just over an hour later, only dimmer switch and family temp left.

[image]

but in phoscon, we still have  everything 

[image]

SwoopX, 2020-04-23:
Not sure if I got that right, are 2 different systems involved? If yes, have you imported a backup?

I've never seen devices just vanish without doing anything.

hosting-io, 2020-04-23:
same system, first image, was a couple days back, I deleted everything this morning and started fresh, re-added those devices you saw in first image, then just over an hour later, stuff disappeared.  TO be fair. it normally did not disappear, it just did have any connections in or out

hosting-io, 2020-04-23:
here is a screenshot from right now - everything is back 
[image]

SwoopX, 2020-04-23:
Not sure if you're aware of this but Xiaomi devices are deep sleepers and it takes up to 1h for them to report in again. In that sense, an hour was considered too short. That might change, however, if you wake them by pushing the pairing button repeatedly.

hosting-io, 2020-04-24:
here is this mornings view - as you can see, Pool sensor has dropped off

[image]

and here it is from home assistant, last connection 11 hours ago

[image]

hosting-io, 2020-04-25:
@SwoopX please see updated screenshots showing dropped device and not sending updates

Smanar, 2020-04-25:
We haven't see the "pool sensor" connected on your screen (To see mesh and color line)

Perhpas the pool is just too far away ? Have you tried the USB cable extension to improve the connexion ?

hosting-io, 2020-04-25:
sorry @Smanar i missed that one, here it is, this was just after I added it to the system

[image]

hosting-io, 2020-04-25:
for context, here is the position and layout of the dconz to the sensor. that family room door is the one next to the dconz

[image]

hosting-io, 2020-04-25:
here is view from phoscon now, -- its dropped here also now

[image]

SwoopX, 2020-04-25:
For my taste, that's range-related. What happens if you bring the pool sensor next to your Conbee and leave it there for a day? And is it the only sensor which passed out then?

Btw, I envy you for the pool ;)

Smanar, 2020-04-26:
He is right, the test is easy to do, just put the sensor closest for some days.

But IDK, for range.
9 m, only 1 glass obstacle, the conbee use an USB cable extension, and the connexion is realy green on deconz.

hosting-io, 2020-04-26:
@SwoopX thank you, its been a godsend for this lockdown with the kids. 

regarding range, I did not think it was an issue, it should have 200M line of site or 30M indoor, we are less than 10M away and its almost line of site, the sensor was on inside wall.  I have added a lot more sensors and some more lights now as well

I have tried to arrange this in the general location

everything under the deconz is in the same room, within 6 metres. the front door is about 25M direct, but has 3 walls between it. Patio light is directly opposite the sensor, one piece of glass in between and it is in a metal casing
Gym door is just behind that light 

back door is just off that as well, around 8M from the patio light, window in the way. also almost line of site from deconz, and around 20M away

[image]

Smanar, 2020-04-26:
I m seing not connected router on your capture ?
You have loose connexion with all your router ?

hosting-io, 2020-04-26:
yes, they drop off constantly like this, especially the lights when they are not physically on, maybe that's part of the problem, let me turn the lights on and leave them on, but turn off in app (they are all on a cat5 based system, so its not quite possible to isolate individual lights and leave switches on, , but I can try this manually, outside lights are all hard coded on a timer, again, I can try and re-do this and control from app. ill update screenshot in a bit

Smanar, 2020-04-26:
Ok so you realy have a big problem. Loosing end device and loosing router without power off is another problem.

It's something new ?

Have you try to change or block wifi channel ? https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

When you are saying "turn off light" it mean off but still powered ?

hosting-io, 2020-04-26:
latest attached, as you can now see, its dropped front door, family room door (which is about 3M from sensor) and fam temp which is about 4M, both in the same room
even though family room door is not connected, if i open and close it, it shows correct status in home assistant and then it connects again with a green line in second.  this is my experience over around 3 weeks of removing and re-adding stuff.

weird thing is, if use Philips Hue, i don't have any of these issues, and if i use the xiaomi gateway, everything stays connected as it should, only problem is, xiaomi wont integrate with home assistant, and i cant use the hue lights as part of the mesh, so that's why i wanted to get this working

Ive put zigbee on channel 25 and wifi on 6 to try and keep them seperate as well, but based on both hue and xiaomi working, I don't believe is a channel issue either

[image]

Smanar, 2020-04-26:
You are saying not having problme with philips hr, but on previous capture "right island" and "left island" was not connected ?

>When you are saying "turn off light" it mean off but still powered ?

BTW have you check battery level ?

hosting-io, 2020-04-26:
i mean, if i use hue app or mi home app, there are no issues, that right and left not connected is because someone turned mains off,   these are all new devices, battery all 100% still

yes, turn off was at mains, I have told everyone not to touch mains for now

SwoopX, 2020-04-26:
Are you using Homeassistant and maybe homebridge-hue?

hosting-io, 2020-04-26:
homeassitant only

philips hue was direct through Hue app and Home Assistant integration

don't have apple products

SwoopX, 2020-04-26:
What happens if you reboot the addon in HA?

hosting-io, 2020-04-26:
done many times including removing completely and re-installing, same end result, works for a day or so, then starts to drop non-lights, but what I will say is, even though stuff "appears" to drop off, since moving everything into the one room, its all actually working, see new screenshot of second - all connected again accept lights which are powered off at mains, so maybe it is a range issue, which makes no sense, cause the front door is much further and through walls.

[image]

meremortals70, 2020-04-27:
What zigbee channel are you using?
What Wifi channel are you using?
How far is your conbee from other possible RF sources? I have had an issue with a UPS for example.

hosting-io, 2020-04-27:
zigbee 25
wifi 6

conbee is in the living room which is open space, only thing close to it  is a small PC, TV and a soundbar. their is a ,mitsubishi aircon about 3M above it, but its not in use and hasnt been since last summer

hosting-io, 2020-04-28:
here is the situation from a couple hours ago

almost everything dropped or red now

[image]

some stuff has not had Updated in nearly 20 hours (this is a couple meters away and above the sensor, front door is the furthest and that dropped 5 hours ago

[image]

Smanar, 2020-04-28:
For me you realy have connexion problem, but I can see the USB extension cable, and I don't know enought zigbee to speak about radio interference.

Have you check power ?

IDK if it can be an hardware problem from the key itself.

kmplngj, 2020-04-28:
I think I have the same problem since one of the last updates.

hosting-io, 2020-04-28:
@kmplngj thank God, i have sometimes thought I am going crazy when everything keeps dropping for no reason

Smanar, 2020-04-28:
As you loose all device in less than 1 day, an easy thing to try isusing it on an other machine.
If you have a windows machine (I think it's the OS with the less problem with deconz), just install deconz and phoscon, include 2 / 3 device and let the conbee for 24 h.

If you still have problem, except re-install the firmware, I don't see other thing to test.

Smanar, 2020-04-28:
As you loose all device in less than 1 day, an easy thing to try is using it on an other machine.
If you have a windows machine (I think it's the OS with the less problem with deconz), just install deconz and phoscon, include 2 / 3 devices and let the conbee for 24 h.

If you still have problem, except re-install the firmware, I don't see other thing to test.

hosting-io, 2020-04-28:
Moved everything to a Windows Machine now which is a little bit further from the problem sensors, but leaving them all inside for now, everything is now more or less line of site within 15M max, all open spaces, with a few walls, but not closing the empty spaces

latest view

[image]

hosting-io, 2020-05-01:
day 3 on windows

[image]

SwoopX, 2020-05-01:
So, on Windows it's stable now?

grEvenX, 2020-05-02:
I'm seeing the same problem, but I'm new to both Deconz and Zigbee, so might just be something I don't understand. I run my deconz on the latest Docker image (marthoc/deconz) and the only device I have so far is an Aqara Wireless Remote Switch (Double Rocker) model WXKG02LM. 
I can add the switch fine, and it will be working for ~1-2 hours, then suddenly no message appears in the log when I press the buttons on the switch.

I need to test it more thoroughly, but the only thing I see that might indicate something in the logs is that after a while this message is repeated 100 times or something over a period of 4 minutes, before it goes completely silent:

`skip binding for attribute reporting of ep: 0x00 cluster 0x0012 (end-device might sleep)`

The only way I have gotten it to start working again is to remove the device, reset it and then add it back again.

hosting-io, 2020-05-02:
> So, on Windows it's stable now?

@SwoopX  so far yes, few issues still like unable to find lights that are less than 5M away from other lights, but hoping when I eventually get the plugs from Ikea they will fix that if it is a range issue

Smanar, 2020-05-02:
Ok, the PC machine is at same place than the ubuntu server box ?
You are using too the cable extension ? perhaps without cable, less tension lost ?

hosting-io, 2020-05-02:
its about 12M away from the ubuntu box, still using the same extension cable which is now higher up

Smanar, 2020-05-03:
So it can be the emplacement ?

Smanar, 2020-05-07:
I see only 1 way to explore, the power, the USB port from the PC is perhpas better than this one on the server

After it's same cable, same gateway, same software version, conbee at same place, I don't think the application not work same on windows than Unix.

Have you powered HUB to try ?

hosting-io, 2020-05-09:
i still have massive problems with range, anything more than 20M line of site and it does not connect, indoors, around 10M is maximum

On windows, its in the hallway under the stairs, so not quite the same position.  I have also got 5 x Ikea plugs
i really wish there was a way to extend the range, eg; my home office with the door open, i can connect a light, but as soon as i close the door, that last little bit of glass is enough to kill the signal and i lose it. even one light or plug that was stable here would enable me to add sensors in the office as well

[image]

SwoopX, 2020-05-09:
From what I see I'd say one of the issues is that everything's routing through your coordinator. Your network is no real mesh here. Do your routing devices (yellow) get cut off from power from time to time?

hosting-io, 2020-05-09:
initially, it was just the controller and a few devices, and they kept dropping. i have slowly introduced new items.  we did have issues with the lights being manually turned off, but not any more, the plugs themselves are on always.  i have spread them around a bit, should i remove everything and start again, connecting items one by one in situ ? maybe ill try this with the ones closest to the office first

hosting-io, 2020-05-13:
still stable , so will write this down ti range, which appears to be pathetic at best in the open, internally its OK, now that i have more plugs around to help boost.

for something that works on the same frequency and principle as wifi, it beggars belief that 15 or 20M line of site is too far to reach

[image]

hosting-io, 2020-05-14:
i am at the point where i give up. So, i thought everything was stable, so i started adding stuff back into home assistant, and many devices are actually missing from the deconz app, which i know were there, and they are still in the phoscon app, just not showing in the mesh at al now even though i have extended the mesh with 5 plugs. here is an example, this was working fine, but now its gone from deconz

[image]

Smanar, 2020-05-15:
And this device don't use router ? like all other device "Family room ..." ?
Wich brand are your plug ? Osram ?

hosting-io, 2020-05-15:
I'm standing by the sensor (xiaomi aqara door/window), light you can see is a Philips Hue, green one is a Chinese zigbee, under the tree you can see a Ikea plug, this sensor is less than 3M from the light and 5M from the plug, complete open line of site. This device was connected fine, if you look through above images, 3 or 4 days ago, it just disappeared from the mesh, but stayed in phoscon as unreachable.

[image]

Smanar, 2020-05-15:
Hue light are realy good router for Xiaomi, and Ikea plug too.
Have you check if the sensor have used direct connexion or one of your router ?

It can't be distance or obstacle.

hosting-io, 2020-05-15:
ill delete everything again, and add only routers first, but i tihnk one of my very first screenshots was similar, stuff just disappears

hosting-io, 2020-05-15:
see TV Wall 1 & 2 (2 is red off island left & 1 is green)

image here is of this light, its 2 bulbs next to each other

[image]

[image]

Smanar, 2020-05-15:
I m trying to remember if on my side not connected light keep their green connexion so long time.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix