dresden-elektronik/deconz-rest-plugin #2711: deconz completely crashed

cosmo84, 2020-04-22:
Hi all,

I bought a new RPi 4b and installed OS ( raspbian buster) and deconz from scratch. Then I imported the backup from my previous Pi, all lights and sensors are in the GUI, but greyed out and nothing works. In the deconz GUI all routers are blinking red on the left side. 
I use latest software and firmware with a raspbee module (I moved this from the old to the new Pi).

Any idea is welcome.

Thanks,

Stephan

ebaauw, 2020-04-22:
Are you using a ConBee?  Make sure it's connected to a USB2 port, preferably through a USB extension cable.  Don't connect anything to the USB3 ports.

cosmo84, 2020-04-22:
No, it‘s a raspbee module

ebaauw, 2020-04-22:
Did you set the serial port config?

cosmo84, 2020-04-22:
Yes, I followed exactly the online manual.

SwoopX, 2020-04-22:
Did you install with desktop? are you using docker?

cosmo84, 2020-04-22:
Desktop: yes, Docker:No

Thanks!!!

cosmo84, 2020-04-22:
Okay, I did another fresh install, same procedure, and now It‘s working again, but only after resetting all lights. I don‘t understand it, but it‘s working again. 

Thanks to all!!!

closed 2020-04-22 by cosmo84

SwoopX, 2020-04-22:
Have you tried to restore configuration via that special page in Phoscon? Check the wiki for guidance.