dresden-elektronik/deconz-rest-plugin #2712: Unable to get API Key

jonassam1, 2020-04-22:
Hi!
I run deCONZ using the Docker container on Ubuntu. I use this docker-compose file to setup the container. 
```
  deconz:
    image: marthoc/deconz
    container_name: deconz
    network_mode: host
    restart: always
    volumes:
      - ${USERDIR}/docker/deconz:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/ttyACM0
    environment:
      - DECONZ_WEB_PORT=7080
      - DECONZ_WS_PORT=7443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PORT=5901
    ports:
      - "7080:80"
      - "7443:443"
      - "5900:5900
```

Following the Getting Started part of the deCONZ API REST API documentation, I do get the IP address and port from the request to "https://dresden-light.appspot.com/discover" which results in the following:
```
[
    {
        "id": "00212EFFFF04835C",
        "internalipaddress": "192.168.1.69",
        "macaddress": "00212EFFFF04835C",
        "internalport": 7080,
        "name": "Phoscon-GW",
        "publicipaddress": "xxx.xxx.xxx.xxx"
    }
]
```

I use the information above to call http://192.168.1.69:7080/api with the body of 
**`{ "devicetype": "my application" }`**
and get the expected "403 Forbidden" with the follwing body:
```
[
    {
        "error": {
            "address": "/",
            "description": "unauthorized user",
            "type": 1
        }
    }
]
```
I then go into the Phoscon app, and navigate to where the "Authenticate App" button is and press the button. But I still get the same "403 Forbidden" with the exact same payload as above.

If there is any relation, I also have the following issue: When going to the deCONZ app using VNC, it gets stuck in the "Joining" state and is never able to actuallly joining. I have no idea if those two issues are related.

Best regsrds,
Jonas

SwoopX, 2020-04-22:
Did you use the POST method instead of GET?

jonassam1, 2020-04-23:
Yes, same result.

jonassam1, 2020-04-23:
Sorry, actually I forgot to open the Gateway, once I had done that it worked. Most embarrasing to miss something this easy.. Thanks for the help!

closed 2020-04-23 by jonassam1