dresden-elektronik/deconz-rest-plugin #2715: Unable to update to V2_05_75

KLucky13, 2020-04-24:
I currently have some issues with connecting my "no neutral" Aqara switches to deCONZ, when checking my software version, it is still on V2_05_74 (installed on raspberry pi). When doing apt update & upgrade, the software remains at the old version. I have no clue what else to try to get the software to update to the newest version (I'm pretty new to this).

Thank you,

Kenny

SwoopX, 2020-04-24:
How did you determine that? Have you restarted deconz to be sure?

KLucky13, 2020-04-24:
By the fact that in phoscon I can see the current version and it reads "version is up to date" while I see there is a new version available. I restarted everything multiple times by now. Also in homebridge I get warnings reading something like "not on recommended version of deCONZ, recommended version is 2_05_75" (not exact quote of the message)

jan666, 2020-04-24:
I was told 2.05.75 is still beta, so only available via beta repository. Latest stable is 2.05.74

KLucky13, 2020-04-24:
Ah ok, I did not know that.. Since I was getting those warnings not running the "recommended version" in homebridge, I assumed it was a stable version. I'm also not sure if this version will solve the problems with the "no neutral wire" switches, but I figured I wanted to update to the latest version to be sure. Thanks for the feedback

SwoopX, 2020-04-24:
> I was told 2.05.75 is still beta, so only available via beta repository.

Good point. Since I use the beta repository only, there's a high tendency to forget about that, sorry.

Smanar, 2020-04-24:
I m not sure but the message "version is up to date" concern the firmware version ? no the deconz version ?

KLucky13, 2020-04-24:
Thanks, apparently the problems I have, have nothing to do with the aqara switches but with the conbee. After a reboot of the Rpi, I keep losing all my devices in HomeKit and all linked automations. I have lost so much time on this already, I’m probably just going back to the aqara hub and send the deCONZ back to get a refund. I had hoped that the bigger bugs and stability issues would have been long gone since it is no longer a new tech by now

SwoopX, 2020-04-24:
Wasn't your issue about the deconz version update? I mean, has that been sorted out now?

Apparently you run into a new issue which might be discussed seperately while providing more information around your setup.

KLucky13, 2020-04-24:
Hi, well the update part has been cleared out, since it’s a beta version and I have the “stable” package, it’s normal that it does not update to the newest version. I first thought the problems I was having was due to the “older” version I was running until I came across this topic: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2273

SwoopX, 2020-04-24:
Understood, and that's unfortunate. However, to be fair, the description "I'm having issues with adding my switch" is not exactly a precise description of your current situation and environment you're running. No offense, but since it appears you just started in using deconz, there's quite a number of traps to tap into.

There's also quite a number of possible solutions which might apply, be it due something being overlooked or technical misconfigurations (which might be introduced by additional software or virtualization).

I'd suggest you either chime into the discussion you previously mentioned or maybe join the discord channel for some more interactiviy.

KLucky13, 2020-04-24:
SwoopX, you are absolutely right and I'm sorry for my shortsighted respons. I was just frustrated with having all these issues I was encountering. 

So to give you more details:

- I'm running a Conbee II stick on a Raspberry pi. 
- I'm running domoticz and I'm using homebridge in order to get the devices to homekit, all my automations are created in homekit. If I need more advanced programations or logics, I was planning to do this in domoticz (also one of the reasons why I wanted my devices to link in domoticz and not on the aqara hub)
- I'm also running an MQTT broker in order to get reflashed sonoff basic switches to talk to homekit
- Conbee is started on reboot (and this seems to be a problem), Whenever the raspberry is booted, homebridge tries to make contact with the conbee II through the "hue-hombridge" plugin. Where it runs into an error (it can't connect the first time), after a few seconds it retries and at this point the connection is succesfull, but issues start to occur: switches show up either "not responding" or are removed from homekit. After relaunching homebridge (and triggering switches manually, eventually the switches start to show up again). However, all programations in homekit linked to the problematic switches are broken and need to be reprogrammed again.
- Motion sensors do not seem to have the issue (so far)
- I use all wired aqara swithes (with neutral wire), apart from 2 which are "no neutral" switches. From what i read, these "no neutral" switches do not act as "bridges" or "repeaters", don't know the correct term. The wired switches with neutral wire DO act as bridges. 

Any extra information that can be helpful to get to the root of the issue can be provided offcource and I do appreciate the help. Forgive me for my frustrated outburst from before :-) Long day of work combined with these "toys" not working was a bit of a frustration ;-)

ebaauw, 2020-04-25:
> However, all programations in homekit linked to the problematic switches are broken and need to be reprogrammed again.

This is an old problem with the REST API plugin: it doesn’t return any `/lights` resources after startup, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2473.  For a workaround, see https://github.com/ebaauw/homebridge-hue/issues/471.

> I use all wired aqara swithes (with neutral wire), apart from 2 which are "no neutral" switches. From what i read, these "no neutral" switches do not act as "bridges" or "repeaters", don't know the correct term. The wired switches with neutral wire DO act as bridges.

The correct Zigbee terminology is “routers” vs “end devices”.  The difference is caused by the respective device firmware for these switches, not by the need for a neutral wire.

KLucky13, 2020-04-25:
> > However, all programations in homekit linked to the problematic switches are broken and need to be reprogrammed again.
> 
> This is an old problem with the REST API plugin: it doesn’t return any `/lights` resources after startup, see #2473. For a workaround, see [ebaauw/homebridge-hue#471](https://github.com/ebaauw/homebridge-hue/issues/471).
> 
> > I use all wired aqara swithes (with neutral wire), apart from 2 which are "no neutral" switches. From what i read, these "no neutral" switches do not act as "bridges" or "repeaters", don't know the correct term. The wired switches with neutral wire DO act as bridges.
> 
> The correct Zigbee terminology is “routers” vs “end devices”. The difference is caused by the respective device firmware for these switches, not by the need for a neutral wire.

I appreciate the feedback, however since I'm pretty new to this, I have no idea how to do the thing you described here: **Create a “lightlist” resourcelink to make homebridge-hue delay the homebridge startup, until all lights in the list are available. Easiest way to create the resourcelink is through ph lightlist. Make sure to run this when all lights are available.**

Care to explain a bit more in detail?

SwoopX, 2020-04-25:
@KLucky13 Your frustation is understandable and I don't wanna know how many hours I've dumped im my life due to stupid errors... And if it's been not such a good day...

Bright side is that ebauuw can help you out with homebridge-hue related questions and Smanar on Domiticz.

Could you provide some further information on you setup as follows?
Hardware running the ConbeeII: e.g. NUC, rpi, etc
Connected to USB3: Yes/no
Using docker: Yes/no
deconz version: e.g. 2.05.75
Firmware version: e.g. 264A0700

> Whenever the raspberry is booted, homebridge tries to make contact with the conbee II through the "hue-hombridge" plugin. Where it runs into an error (it can't connect the first time), after a few [...]

So, your issues are solely related to the Aqara switches?
Which model are those?
Do you know how to access deconz GUI (not Phoscon) and can you provide a screenshot of you mesh?

KLucky13, 2020-04-25:
> @KLucky13 Your frustation is understandable and I don't wanna know how many hours I've dumped im my life due to stupid errors... And if it's been not such a good day...
>
> Bright side is that ebauuw [...]

Hi, I'm experimenting with the "lightlist" as described by ebauuw, and it seems to be solving some of the problems regarding losing my switches in homekit. However when I reboot now, it seems to hang on one lightswitch in an endless loop. I have no idea which one it is since it only references it as "light 3". The only way to get through the reboot of homebridge is by updating the lightlist, making it boot and than update it again. So I need to figure that out. I've seen that my "no neutral" aqara switch makes 4 lights in total (combined in 1 tile), only 1 is triggering the light. Maybe it's one of those 3 dummy switches which is causing problems? No idea how to prevent the no neutral switch to generate 4 switches for only one light...

To answer your questions:

- Harware running the Conbee2: RPI 3B+
- USB 3: I think RPI 3B+ only has 2.0 ports. I connected it via an extention USB cord
- Using docker: Sorry but I don't know what you mean by this
- Deconz version: I'm running the "stable" version, not "beta", so I'm running 2.05.75
- Firmware version: 264A0700 indeed

ebaauw, 2020-04-25:
> it seems to hang on one lightswitch in an endless loop.

By design, homebridge-hue waits for all `/lights` resources from the lightlist to re-appear, before allowing homebridge to start its server.  If a `/lights` resource is deleted, the lightlist needs to be updated, or homebridge won't start.

> I've seen that my "no neutral" aqara switch makes 4 lights in total (combined in 1 tile), only 1 is triggering the light

That's not good, see #2048.  Somehow, the REST API plugin exposes 4 `/lights` resources instead of two.  I don't know why.  Could you please attach _to that issue_ the debug dump file created by homebridge-hue, see https://github.com/ebaauw/homebridge-hue#debug-dump-file.  If you run the deCONZ GUI, please also attach _to that issue_ the screenshots I requested in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2048#issuecomment-617695113.

KLucky13, 2020-04-25:
ebaauw, in my case it is a single rocker instead of a double rocker, but also a no neutral switch (2 wire connection). I will post the screenshots as requested tomorrow and try to figure out how I can deliver the debug dump you requested

Mimiix, 2020-06-05:
@KLucky13 is this issue resolved?

KLucky13, 2020-06-05:
> @KLucky13 is this issue resolved?

No, I did not bother, I’m going to route extra cables to the 2 locations in my house where I don’t have a neutral wire so I can use the “normal” switches with 3 connections

SwoopX, 2020-06-05:
@KLucky13 if you could follow up on the information requested, we may be able to follow up and eventually progress. Maybe it can get sorted out.

KLucky13, 2020-06-05:
> @KLucky13 if you could follow up on the information requested, we may be able to follow up and eventually progress. Maybe it can get sorted out.

I already deleted it from my system, I have no easy way of checking the GUI for debugging, I have to take my raspberry pi off it's wallmount, take it up to my desk, hook it up to a second screen, pull a 10meter cable from my router to my pi, I just could not bother with this. I decided to just buy a "neutral" wired switch instead, I just still need to hook it up. Sorry, I just don't have the time to spend on troubleshooting, I rather just buy things that work..

KLucky13, 2020-06-05:
> @KLucky13 is this issue resolved?

Im also not sure which problem you are refering to, I had 2 problems in this topic, 1 with the multiple tiles for 1 switch (aqara no neutral switch), and losing my devices after a reboot. The lost devices after reboot is solved with creating the lightlist described above. 

However since a few days some of my wired lightswitches started acting up as well, they now show "low battery" and have a build in motion sensor for some reason.. When I click the tile in homekit it no longer triggers the light directly but opens it up and shows different non existing devices: 

https://ibb.co/QHzk9dc
https://ibb.co/zXWcWb2

But again I need to unplug the pi and log into the GUI to see what the problem is here and I'm burried in all kinds of home improvement projects so I currently don't have the time to figure out what is happening here.. At least these are still working and my scenes still trigger the lights and everything so it does not have high priority for me. It's just anoying that homekit is showing 3 low battery triggers for every lightswitch with this problem

closed 2020-07-11 by stale[bot]