dresden-elektronik/deconz-rest-plugin #2716: Connect to Rest API problem

Liese2007, 2020-04-24:
Hi,
I am having trouble to connect to REST API via
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

I launch the GET command and receive:
`[
    {
        "id": "00212EFFFF04FF93",
        "internalipaddress": "172.30.33.0",
        "macaddress": "00212EFFFF04FF93",
        "internalport": 40850,
        "name": "Aqara Gateway",
        "publicipaddress": "31.17.161.51"
    }
]`
So the next step would be to connect APP/Gateway - which I released in the DeConz frontend.
Within the 60 sec I launched the POST command (or whatever this is called):

`172.30.33.0:40850/api`
incl. the text
`{ "devicetype":  "my application" }`

I am receiving the message
> Could not get any response .This seems to be like an error connecting to 

> http://172.30.33.0:40850/api. The response status was 0.

> Check out the W3C XMLHttpRequest Level 2 spec for more details about when this 

> happens.

Do I have to forward any port ( I tried via Fritzbox, but the IP address is not in a valid subnet of my Fritzbox...) ?

Best regards,
Jörg

SwoopX, 2020-04-24:
Have you unkocked the gateway prior to issuing the POST request?

Liese2007, 2020-04-24:
Hi,
yeah, thats what I ment by
```
So the next step would be to connect APP/Gateway - which I released in the DeConz frontend.
Within the 60 sec I launched the POST command (or whatever this is called):
```
In German it relates to "Activate Connectivity App/Gateway".

Greets
Jörg

SwoopX, 2020-04-24:
Ah, sorry.

> Do I have to forward any port ( I tried via Fritzbox, but the IP address is not in a valid subnet of my Fritzbox...) ?
What does that mean? Is it not the IP your system is internally reachable?

Smanar, 2020-04-24:
For me it's a network problem, can you connect you to phoscon ?
Can you with local IP, what is the url if it work ?

Liese2007, 2020-04-24:
My Fritzbox Router is running at
192.168.1.1
My Hassio 
192.168.1.70:8080

The Phoscon/Conbee II is running (plugged via USB) on the same Raspi 3B+.

- If I launch the GET command as in my initial post, I receive
172.30.33.0 and the port 40850.

And the POST command shows the error above.

I just tried to 'Forward' the port 40850 for the device 172.30.33.0 in my Fritzbox...just a blind try to see if that changes anything (i am no network expert, but familiar with the basics).
Then I receive the error that this IP doesnt belong to Fritzbox network.

#Hope that helps...

Regards
Jörg

Smanar, 2020-04-24:
Nope I m speaking about the phoscon web app.
Try to acced to it on the local machine then on a distant machine.

You can find it here http://phoscon.de/app 
But better to use the local IP for test.

Liese2007, 2020-04-24:
Ok, accesssing 
Phoscon.de/app 
Directly from within my network in another browser opens the Phoscon (Aqara Gateway) login page.
BUT ...I can't login with the Password I used to access the phoscon/aqara gateway via Home Assistant Side bar.

The login via HA is still possible...

What di you mean by 'Distant machine'?

Does the info above helps?
Now I am at the edge with my network basics.. 

Regards
Jörg

SwoopX, 2020-04-24:
That smells like docker and some misconfiguration. Not my turf. You might want to join deconz' Discord channel and raise that question there once again. There are some folks which should be able to help you out.

Smanar, 2020-04-24:
I think you confure api key and password.
There is an option to change the password too.

In my mid the local machine is the machine with deconz (with VNC) so you can't have firewall/router problem, all is local, and you can use localhost as ip.
Distant machine for all other machines in your network.

But I have just make a try, even on the logging page , if you enable the debugger in your browser, you see the url nd the port used by the application.

Liese2007, 2020-04-25:
OK,
thanks for your effort.
I have now tried the browser debugger.
Here is the complete screenshot - IP and port are the same as listed in my 1st post .
The same as received via GERT command.

IP: 172.30.33.0
Port: 40850

Next step is reset PW in direct access via Phoscon.de/app.

I will come back....may take some time :-) family is getting up....

...Can I upload a screenshot here?

Regards,
Joerg

Liese2007, 2020-04-25:
Anyway...here is a link to the screenshot:

http://www.trave-forum.de/Phoscon-IP-Port.jpg

Liese2007, 2020-04-25:
And here is another update...
I just realized, that inside Home Assistant, the IP is different...
http://www.trave-forum.de/Phoscon-Different-IP-Port.jpg

Liese2007, 2020-04-25:
OK, by checking the integrations page, I found, that I have a deCONZ Integration with 

**deCONZ-000000000000
and one with
**deCONZ-*String of 16 digits - non zero***

I assume the installation went wrong at some time...?
Should I delete the Gateway "deCONZ-000000000000" ?

Smanar, 2020-04-25:
Ok so I don't know your network but for me

172.30.33.0 is an external IP
192.168.1.70 is the local IP

So you need to use the second one, not the first one.

Seriously you have nothing on your os to print your tcp network ? Your router config page ?

Liese2007, 2020-04-25:
A) Network setting
My local network - as you say - is available via 192.168.1.x
My Router / Standard Gateway is 192.168.1.1
My Home Assistant / Hassio is at 192.168.1.70:8123
     - The Conbee II Stick is directly connected to the Rasperry Pi with Home Assistant

B) Reinstallation
Meanwhile, I made a backup for the deCONZ setup and deinstalled Integartion & Add-on.
I re-installed everything (after reboot etc) and loaded the Back Up.

**C) deCONZ Behaviour**
**C1) OPEN VIA HA WebUI**
When I enter the deCONZ WebUI I see the Gateway Name and IP: 192.168.1.70.
I can enter with my given Password. 
Here I can activate Gateway Connect for 60sec.
The Welcome Screen  with the groups opens and in upper right corner is written Gateway Name and IP 172.30.33.0.
**C2) OPEN VIA PHOSCON APP DIRECTLY**
When I directly open 'http://phoscon.de/pwa' , I get same Gateway Name but IP:172.30.33.0.
Here, I can't login. Wrong Password.

**D) Next try to get RestAPI contact acc. the Howto**
Acc. Getting Started 
with 
**GET https://dresden-light.appspot.com/discover**
I receive 
172.30.33.0 as IP
40850 as internal port (see above)

172.30.33.0:40850/api --> Could not get any response
192.168.1.70:8123/api --> Error 404 (Just a try from my side)

**E) Home Assistant Log**
My Home Assistant log shows:
> 20-04-25 16:40:47 INFO (SyncWorker_14) [supervisor.docker.audio] Start Audio homeassistant/armv7-hassio-audio with version 14 - 172.30.32.4
> 20-04-25 16:40:47 INFO (SyncWorker_12) [supervisor.docker.dns] Start DNS homeassistant/armv7-hassio-dns with version 9 - 172.30.32.3
> 20-04-25 16:40:47 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
> 20-04-25 16:40:47 INFO (SyncWorker_6) [supervisor.docker.cli] Start CLI homeassistant/armv7-hassio-cli with version 25 - 172.30.32.5
> 20-04-25 16:40:47 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
> 20-04-25 16:40:48 INFO (SyncWorker_5) [supervisor.docker.interface] Attach to homeassistant/raspberrypi3-homeassistant with version 0.108.9
> 20-04-25 16:40:48 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /
> 20-04-25 16:40:48 INFO (MainThread) [supervisor.hassos] Detect HassOS 3.13 / BootSlot B
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/core repository
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/a0d7b954 repository
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.store] Load add-ons from store: 65 all - 65 new - 0 remove
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.addons] Found 4 installed add-ons
> 20-04-25 16:40:50 INFO (SyncWorker_11) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-ssh with version 8.5.4
> 20-04-25 16:40:50 INFO (SyncWorker_4) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-samba with version 9.2.0
> 20-04-25 16:40:50 INFO (SyncWorker_8) [supervisor.docker.interface] Attach to hassioaddons/ftp-armv7 with version 3.3.1
> 20-04-25 16:40:50 INFO (SyncWorker_10) [supervisor.docker.interface] Attach to homeassistant/armhf-addon-deconz with version 5.3.2
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.discovery] Load 1 messages
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.ingress] Load 27 ingress session
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.secrets] Load Home Assistant secrets: 1
> 20-04-25 16:40:50 INFO (MainThread) [__main__] Run Supervisor
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.api] Start API on 172.30.32.2
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.utils.gdbus] Call de.pengutronix.rauc.Installer.Mark on /
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.hassos] Rauc: B - marked slot kernel.1 as good
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.addons] Phase 'initialize' start 0 add-ons
> 20-04-25 16:40:50 INFO (MainThread) [supervisor.addons] Phase 'system' start 1 add-ons
> 20-04-25 16:40:54 INFO (SyncWorker_19) [supervisor.docker.addon] Start Docker add-on homeassistant/armhf-addon-deconz with version 5.3.2
> 20-04-25 16:40:59 INFO (MainThread) [supervisor.addons] Phase 'services' start 3 add-ons
> 20-04-25 16:41:02 INFO (SyncWorker_1) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-samba with version 9.2.0
> 20-04-25 16:41:02 INFO (SyncWorker_7) [supervisor.docker.addon] Start Docker add-on hassioaddons/ftp-armv7 with version 3.3.1
> 20-04-25 16:41:04 INFO (SyncWorker_3) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-ssh with version 8.5.4
> 20-04-25 16:41:09 INFO (SyncWorker_14) [supervisor.docker.interface] Start homeassistant/raspberrypi3-homeassistant
> 20-04-25 16:41:12 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_ftp
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.homeassistant] Detect a running Home Assistant instance
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.addons] Phase 'application' start 0 add-ons
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.tasks] All core tasks are scheduled
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.hwmon] Started Supervisor hardware monitor
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.core] Supervisor is up and running
> 20-04-25 16:42:10 INFO (MainThread) [supervisor.host.info] Update local host information
[... cut ...]
> 20-04-25 16:42:11 INFO (MainThread) [supervisor.host.network] Update local network DNS information
> 20-04-25 16:42:11 INFO (MainThread) [supervisor.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/NetworkManager/DnsManager
> 20-04-25 16:42:11 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information

++++++++++++++++++++++++++++++++++++++++++++++

If there is still an open point regarding my network or if you require any more logs etc, I can post them. Does my info from today is of any help? Took quite some time :-)

Regards
Jörg

Smanar, 2020-04-25:
And 192.168.1.70:40850 ?

Liese2007, 2020-04-25:
The same:
192.168.1.70:40850/api --> Could not get any response

What do you mean by 'old web app' ? How can I update that?
Would that be of any help?

SwoopX, 2020-04-25:
> 
> 
> Ok so I don't know your network but for me
> 
> 172.30.33.0 is an external IP (to use outside your network)
> 192.168.1.70 is the internal IP (to use on your own network)
> And for the same machine
> 
> So you need to use the second one, not the first one (I presume you are using a machine connected to your own network ? Not a smartphone connected with 4G).
> 
> Seriously you have nothing on your os to print your tcp network ? Your router config page ?

@Smanar sorry to say that, but that's total rubbish. Both mentioned IP addresses are part of the private IP blocks, which are for internal network communication only and NOT towards internet! Since the system comes up with 2 different private IP addresses, there's a virtualization layer involved or you have 2 communication ongoing on 2 different network interfaces with totally different IP addresses (the first applies if I read the log excerpt above correctly).
We don't discuss any network basics here any further.

@Liese2007 You have a NAT issue on your system, NOT in your network. That's why playing on the Fritzbox doesn't bring you any further. That said, it is not a deconz issue.

The 10.XYZ address is not translated over to 192.XYZ and vice versa correctly as it seems, at least not for certain ports. I'd recommend to double check ip your IP and port configuration is set appropriately at the corresponding configuration points.

Liese2007, 2020-04-25:
1. Which address do you mean with **10.xyz** to be translated to 192.xyz?

2. What do you mean by 'recommend to double check ip your IP....etc??

3. If you assume, that it is my system and not my network...should I give another PC a try? Imo the NAT is done in my router, so if the problem is related to NAT, isn't the router to be checked?

I still doubt, that deCONZ installlation process with deConz-000000000 error etcpp has nothing to do with it....
So what can I ask at the Discord channel?

If I move over to a network forum they will most likely handle it like "Not a network issue...." or like "check the databases of the Conbee Stick".

Any chance, that I can follow that issue with a Home Assistant/deCONZ AND network expert?

Thank for your effort, to both of you.
Still hope to find the solution.

Regards
Joerg

SwoopX, 2020-04-25:
> Which address do you mean with 10.xyz to be translated to 192.xyz?

172.30.33.0

> What do you mean by 'recommend to double check ip your IP....etc??

I've read HA, HassIO and docker. Although I don't use any of these, I know they have places or configuration files where you need to specify your network setup. E.g. https://docs.docker.com/network/bridge/

> If you assume, that it is my system and not my network...should I give another PC a try? Imo the NAT is done in my router, so if the problem is related to NAT, isn't the router to be checked?

Based on the information at hand, let's simplify it and assume your system is the only one available (besides your laptop) and having an outfacing IP of 192.168.1.70 and you tell your router: Hey, everything to 192.168.1.70 should now go to 172.30.33.0, what will happen? Nothing, since that address is unknown/not reachable.

> I still doubt, that deCONZ installlation process with deConz-000000000 error etcpp has nothing to do with it....
So what can I ask at the Discord channel?

Never said that it's not of any relevance. You may ask exactly that. And also, if anyone knows why Phoscon comes up with different IPs.

Smanar, 2020-04-26:
For me 172.30.33.0 is not a private IP.

Liese2007, 2020-04-27:
Ok, it seems that the IP is realted to some Docker container.
https://community.home-assistant.io/t/login-attempt-or-request-with-invalid-172-30-33-0/93681/8
Home Assistant Community

Does that give some idea for further steps?

Liese2007, 2020-04-27:
[image]

Hi all,
simple as that.
If I fill in the 40850 in the Add-on configuration, the Phoscon.de/app is available via
192.168.x.xx:40850 
the Postman POST command: 192.68.1.70:40850/api after Gateway release in the Backend establishes contact to API.

So for me this issue can be closed....I think.
A lot of further Help I received by Developers and Admins of deCONZ' Discord Channel!
Thanks guys!

closed 2020-04-27 by Liese2007

jscheurwater, 2021-03-08:
I have the same issue. If I do the GET command in Postman, I'm getting an internal IP like: 172.30.xx.x
As i read in the discussion started by Liese2007, I need another kind of IP (like 192.168.x.x).

Does somebody know how to get this IP address. I can't find troubleshooting on this issue in the REST API documentation.
Thanks in advance!