dresden-elektronik/deconz-rest-plugin #2717: [BUG REST API] ConBee itself is listed as light in deCONZ 2.5.75

nodefeet, 2020-04-24:
I updated to 2.5.75 and noticed that after searching for new lights Conbee itself is now shown as a light.

In Postman I receive for: GET: xxx/api/yyy/lights
```json
{
    "1": {
        "etag": "c3121c5d8afcec507512eaf1361eafd4",
        "hascolor": false,
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Configuration tool 3",
        "state": {
            "reachable": true
        },
        "swversion": "0x264a0700",
        "type": "Configuration tool",
        "uniqueid": "00:21:2e:ff:ff:05:98:a7-01"
    }
}
```

At least I do not see any reason to do that. 
Hardware: RPi3B+, Conbee 2, deconz headless

jensflorian, 2020-04-24:
See #2536

Mimiix, 2020-06-05:
Please read: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2472#issuecomment-590541006

Closed for now :)!

closed 2020-06-05 by Mimiix