dresden-elektronik/deconz-rest-plugin #2718: Is Philips Go really compatible with Deconz?

SnaKKo, 2020-04-25:
Hi,
Does Philips Go works correctly with Deconz?
I'm reading a lot of issues here on Github about it.

Anyone tried succesfully? Do you have problems with Philips Go/Deconz? Thanks

zladukas, 2020-04-25:
I have first version of Philips HUE Go and have no problems (like any other Hue product) at all.

SnaKKo, 2020-04-25:
Thansk @zladukas , are you able to manage the Hue Go (with Deconz) running on battery?
Have you "paired" the Hue Go and Deconz without installing the Hue app?

zladukas, 2020-04-25:
Yes. Paired, installed newer FW once, when able to download. You don't need Hue app. But I have no expirience with newer one, whitch has bluetooth or smth...

Kane610, 2020-04-27:
On battery I don't think the device is listening to Zigbee any more, but I haven't verified that

Gizmo-Ger, 2020-04-27:
I have one of the newer versions and I can´t get it to work with Deconz. With or without on battery.

ebaauw, 2020-04-27:
Does it work with the Hue bridge?

Gizmo-Ger, 2020-04-27:
> Does it work with the Hue bridge?

Since I don´t have a Hue bridge I´m using Hue Essentials on the phone. Pairing was no problem at all. I´ve tried several times to pair it with deconz without any luck.

ebaauw, 2020-04-27:
That connects to the Go over Bluetooth, doesn’t it?  Isn’t their a setting to prep the Go for pairing with the bridge?

Gizmo-Ger, 2020-04-28:
Yes, it´s connected via Bluetooth in the Philips Hue BT app on IOS (Model LCT026, SW 1.69.9_hB3217DF4) but trying to add Hue Bridge in the app it doesn´t find the Deconz Bridge. I´ve tried unlock Gateway and/or open network. Both approaches don´t work.

SnaKKo, 2020-04-28:
So it seems that just the old one works correctly, the newer one have some issues with Deconz.

Gizmo-Ger, 2020-04-28:
maybe it´s the same like https://github.com/Koenkk/zigbee-herdsman-converters/pull/801#issuecomment-566182010 ?

Kane610, 2020-05-02:
You probably need to reset it first. It can't be paired over Bluetooth and zigbee at the same time

Gizmo-Ger, 2020-05-02:
I did reset the Go and made sure no Bluetooth connection was established. Still no luck with adding the Go to Deconz.

wvuyk, 2020-05-02:
Trying to understand this. Where do you want to add the Deconz gateway to? 
Is it the Philips Hue app? As far I know you cannot add the current Deconz gateway to the Philips Hue app anymore.  Certainly not the BT version?

Gizmo-Ger, 2020-05-03:
No, I am not using the Philips Hue app at all. Most of my light bulbs are Ikea Tradfri and I swapped the Ikea gateay for the Conbee II. I control the lights via Alexa or Phoscon, except for the Philips Go which seems to be stubborn.

wvuyk, 2020-05-03:
> but trying to add Hue Bridge in the app it doesn´t find the Deconz Bridge

But what is the app you are referring here to then? Is that Phoscon? Does it not find its own bridge?

Gizmo-Ger, 2020-05-03:
Sorry for the mix up. All I want to do is using the Philips Hue Go with Phoscon. So far I could use the Go with the Hue BT app and Hue Essentials. Switching from bluetooth connection to Zigbee doesn´t work. In order to switch to Zigbee I need to add a bridge in the app but the app doesn´t find the Phoscon gateway. Like I said I don´t have a hue bridge, just the ConbeeII stick.