dresden-elektronik/deconz-rest-plugin #2719: VPIR instantly turns back on (true) after turning off (false)

vandalon, 2020-04-25:
When using a vpir sensor, consistent of 2 motion sensors in deconz the api returns a true right after a false when both motion sensors go to false. 

```
< {"e":"changed","id":"107","r":"sensors","state":{"lastupdated":"2020-04-25T11:22:43","presence":false},"t":"event","uniqueid":"vpir-1587812818604"}
< {"e":"changed","id":"107","r":"sensors","state":{"lastupdated":"2020-04-25T11:22:43","presence":true},"t":"event","uniqueid":"vpir-1587812818604"}
< {"e":"changed","id":"74","r":"sensors","state":{"lastupdated":"2020-04-25T11:22:45","presence":false},"t":"event","uniqueid":"00:15:8d:00:03:02:fc:9f-01-0406"}
```

74 was the sensor turning from true to false, the other one was not triggered. 

The sensors used are one philips hue motion sensor (not triggered) and one xiamomi motion sensor. (74)

Mimiix, 2020-06-05:
Is this fixed?

vandalon, 2020-06-05:
> Is this fixed?

Not to my knowledge :)

vandalon, 2020-06-26:
Would be nice to see this addressed :)

vandalon, 2020-07-18:
Bumb

Mimiix, 2020-07-20:
added backlog so the bot leaves you alone ;)

ebaauw, 2020-07-20:
Where did you create this vpir sensor?  Is it a CLIPPresence sensor with rules to update it?  Looks like an issue with the app that created the vpir sensor, not with the REST API.

vandalon, 2020-07-20:
Hi Eric,

I've created them within the phoscon app via sensor control, adding 2 sensors for 1 automation that does nothing, but it does expose a vpir which i then use in node-red.

ebaauw, 2020-07-20:
What rules did Phoscon create to update the vpir?

vandalon, 2020-07-21:
what do you mean exactly with rules?

ebaauw, 2020-07-22:
http://dresden-elektronik.github.io/deconz-rest-doc/rules/