dresden-elektronik/deconz-rest-plugin #2720: Unable to pair Legrand Netatmo

yves19, 2020-04-25:
I have unsuccessfully tried to pair LEGRAND 067694 Netamo emitter with ConBee2 .
[image]
Pressing on reset button did not have any result.
A long press on reset makes the Led blink red  than quicly blue . Once reset relesased the Led blinks green for few seconds and then turns OFF.

Have you any idea how to pair such device ?

Thanks

Smanar, 2020-04-26:
Hello, this device is not yet included in deconz, but will be fast.
Just 2 things.

Can you show the same screenshoot from deconz than this page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support
(the 2 firsts)

Have you an Unix machine to cmpile the code to test code changes ?

yves19, 2020-04-26:
Helle Smanar.

Unfortunately I can't show you any screeshot as I can't include the equipment. Nothing appear in deCONZ when I press the reset  button.
That's the tricky point.
Have you any advice how to create the node of this equipment in deCONZ (meaning how to include it in the Zigbee network) ?

Smanar, 2020-04-26:
Ok so I think it's the same procedure than for battery device (switches)
- long press, ignore the first blink
- short press
- wait 5 s
- retry if not green led

You are looking in deconz (the GUI) not Phoscon, the web app ?

yves19, 2020-04-26:
OK now I could include the Legrand Toggle Switch (thnks to your advice) here are the screenshots
0) The Node
[image]

1) Node Info
[image]

2) Basic Cluster
[image]

3) Legrand Specific Cluster
[image]

Hope it will be of any help.

By the way I mainly work with deCONZ. 
I successfully included the Toggle Switch in a group. I binded it with a Legrand Wall Plug 16A as well.

Regards

Smanar, 2020-04-27:
Ok so first try.

Do you know how to compile the code to try ? You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"

Just replace the step 1 by
`git clone https://github.com/Smanar/deconz-rest-plugin.git`

Not sure It will work but you will see it in the API.

Smanar, 2020-04-30:
Hello, do you need more informations ?

yves19, 2020-04-30:
Thank you for the time you spent to help to solve my problem Smanar.
Unfortunately I couldn't succeed in compiling the latest beta relase you have prepared.
Do not worry I will wait for the next stable one.
In the meantime I will continue to work with deCONZ even if it is not the easiest way to manage my network.

Smanar, 2020-05-01:
As you want, but I can try to help you for compilation ?

yves19, 2020-05-25:
Included in 2.05.76 release

closed 2020-05-25 by yves19

Smanar, 2020-05-25:
So working mode confirmed ?

yves19, 2020-05-25:
In fact everything works well in decONZ itself but the switch is not seen through the REST API (nor in lights , nor in sensors)

Smanar, 2020-05-25:
So it's not working ^^.
Have you tried to include it using phoscon ? add new sensor ?

yves19, 2020-05-25:
I have planned to try thisnext week with new devices i have ordered . The current ones are no more available for tests (they are integrated in my home system and can't be easyly removed)

Smanar, 2020-05-25:
Ok, because you need to re-include the device to test the new version and using Phoscon.

yves19, 2020-05-26:
OK it rocks.
After new inclusion the Toggle Switch ii reachable through REST API.
When pairing new sensors , Phoscon says it's OK (ready) even the switch can't be seen in Phoscon afterwards. (Phoscon/deCONZ  2.05.77)

Smanar, 2020-05-26:
Yep, this is something normal, sensor are invisible by defaut on phoscon.

jd1900, 2020-06-11:
How is the current state for this device?
Is binding supported for this device?

I find it very appealing.

And..., do you know of a similar device from other brand?

Smanar, 2020-06-11:
I have forget the battery filed, but it will work like other Legand wired switch.

And no, never see this kind of device, except using Xiomi hack on battery switch.

And this switch can support monostable and bistable switch.

yves19, 2020-06-11:
@jd1900 this device supports binding on On/Off clusters . I have tested it on Legrand wall plugs as well as Legrand micro module switch and on Philips Hue bulbs ans Hue ceiling lights.

This remote command acts as any integrated remote command. For instance you can use a Philips smart button on which you connect a monostable physical switch (two small weldings in parallel of the small smart button switch and voila)

cocapic, 2021-01-25:
is someone using this device ?, i correctly paired it, but nothing is coming from it :(

Smanar, 2021-01-25:
Hello, what are you using as third app ?
Do have the device JSON ? or nothing at all ?

cocapic, 2021-01-26:
Hi,

 

Jeedom/phoscon/conbee 2

 

Nothing at all yet …

 

ty

Smanar, 2021-01-26:
Ok so on jeedom, you need to see something in the "sensors" part, can be because the device is badly included. The device can be invisible in phoscon.

BTW on jeedom you havent the GUI, but if you realy have too much problem, will be usefull to make the test on a PC to have the GUI and check for model id.

Retry to include it, if you are it s not working, install deconz on a PC, switch the key 10 mn for test on it, and try to include it again, you need to see something like that on the GUI https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2720#issuecomment-619609578 , if you can check, you have the same values

cocapic, 2021-01-26:
Hi,

 

Removed and re included with no change, below capture from win deconz

 

regards

 

Retry to include it, if you are it s not working, install deconz on a PC, switch the key 10 mn for test on it, and try to include it again, you need to see something like that on the GUI #2720 (comment) <https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2720#issuecomment-619609578>  , if you can check, you have the same values

Smanar, 2021-01-26:
Capture haven't worked (I think, or I can see them on the post)

cocapic, 2021-01-27:
Updated post

Smanar, 2021-01-27:
All your nodes are without name, I think it s because you have not moved your database, but it s strange for the device.
It seem include in the zigbee network, but not in the api (you will have generally a name instead of the 0xCAA6.

So it can be a different device, but unfortunately you miss the principal capture ^^, this one https://user-images.githubusercontent.com/64013119/80317227-cbe02d00-8802-11ea-95cb-738f2583bfb7.jpg

You can find it with selecting the cluster 0x0000 "basic" then in tab "Cluster info", you will see an empty table, just press the button "read" (with making the device awake if needed)

To include the device, deconz use : 
- The mac adress > ok for you
- The manufacture number > ok for you
- The model id, visible in the missing capture, need to be "Remote Toggle switch".

If you have the good model id.
- Set phoscon in permit join (add new sensor)
- Read again the basic attribute (same procedure than to have the "model id".

cocapic, 2021-01-30:
Hi,

 

New capture attached

 

Yes i ddnt bck/restore between jeedom/deconz and deconz/win just unplug/plug

 

Is a way to read on deconz wich value is changing when pressing a button or other ?

 

regards

Smanar, 2021-01-31:
>Yes i ddnt bck/restore between jeedom/deconz and deconz/win just unplug/plug

Not a problem, less complicated

Is a way to read on deconz wich value is changing when pressing a button or other ?
Yes in cluster 0x000F, but we don't use it for the code, we are looking for command send by the device, so visible in log (deconz/ heelp / log with "info" and "info_l2")

>New capture attached

Still not visible ^^, but you just need to compare the "Model indentifier", you need to have exactly the same than in the previous capture > "Remote tooggle switch"

And

> If you have the good model id.
> 
> -    Set phoscon in permit join (add new sensor)
> -    Return on deconz, Read again the basic attribute (same procedure than to have the "model id".

cocapic, 2021-01-31:
^^ correctly updated in post (attch not working by mail)

ive try to reinclude it from phoscon without success, and also try with a second new 
[image]
067694 same issue
Remote tooggle switch is ok

Also your talking about add new sensor, but it is a switch, correct ?

thanks for patience

Smanar, 2021-01-31:
Yep, switch, sorry, ^^, but there is no diffrence for the code.

But the wireless switch "Remote switch" is working and not the module "Remote toggle switch" ?
Take care the device can be invisible in phoscon, better to search it in your third app.

Because all seem fine for your device, You are not using a 2 month old deconz version ?

Have you tried the trick ? with putting phosocn in permit join (add new switch), and read again the basic attribute in deconz. This trick provoke an inclusion, and you can read in same time the log in deconz/help/log (info and info_l2)
But for me the device is somewhere in the api.

cocapic, 2021-02-02:
Version 2.9.1

0x26670700

API version 1.16.0

 

Seem correct in version !?

 

The module is correctly created in third app (jeedom), with 1 option « 01-0006.state::buttonevent » but nothing is comming

 

Will try the trick and try to have a look on log, but difficult as i have to remove the key from prod (waf alarm)

 

Many thanks !

cocapic, 2021-02-02:
some logs from third app

```
[2021-02-02 12:59:58][DEBUG] : {"UTC":"2021-02-02T11:59:58","apiversion":"1.16.0","backup":{"errorcode":0,"status":"idle"},"bridgeid":"00212EFFFF04290A","datastoreversion":"93","devicename":"ConBee II","dhcp":true,"disablePermitJoinAutoOff":false,"factorynew":false,"fwversion":"0x26670700","gateway":"192.168.1.1","internetservices":{"remoteaccess":"disconnected"},"ipaddress":"192.168.1.6","linkbutton":false,"localtime":"2021-02-02T12:59:58","mac":"dc:a6:32:52:dc:06","modelid":"deCONZ","name":"Phoscon-GW","netmask":"255.255.255.0","networkopenduration":180,"ntp":"synced","panid":47874,"portalconnection":"disconnected","portalservices":false,"portalstate":{"communication":"disconnected","incoming":false,"outgoing":false,"signedon":false},"proxyaddress":"none","proxyport":0,"replacesbridgeid":null,"rfconnected":true,"starterkitid":"","swupdate":{"checkforupdate":false,"devicetypes":{"bridge":false,"lights":[],"sensors":[]},"notify":false,"text":"","updatestate":0,"url":""},"swupdate2":{"autoinstall":{"on":false,"updatetime":""},"bridge":{"lastinstall":"2021-01-17T22:10:04","state":"noupdates"},"checkforupdate":false,"install":false,"lastchange":"","lastinstall":"","state":"noupdates"},"swversion":"2.9.1","timeformat":"24h","timezone":"Europe\/Brussels","uuid":"e61e0d43-6884-4962-aff8-471e6ac2005a","websocketnotifyall":true,"websocketport":20877,"whitelist":{"0E2F9A99AC":{"create date":"2020-12-12T16:50:08","last use date":"2020-12-16T04:00:04","name":"Phoscon#B3840x910"},"1349C80683":{"create date":"2021-01-28T18:02:59","last use date":"2021-02-02T11:54:18","name":"Phoscon#B2048x1010"},"467DF12351":{"create date":"2021-01-28T12:30:06","last use date":"2021-01-28T12:34:19","name":"Phoscon#B1504x860"},"4B7EE27383":{"create date":"2021-01-17T20:52:48","last use date":"2021-01-17T21:34:37","name":"Phoscon#B375x640"},"9ECF31D395":{"create date":"2020-11-25T17:12:26","last use date":"2020-12-02T15:39:44","name":"Phoscon#B1903x919"},"A293C230E2":{"create date":"2020-11-24T21:35:53","last use date":"2021-02-02T11:59:58","name":"jeedom"},"CE674A24E2":{"create date":"2020-11-24T23:16:20","last use date":"2020-11-30T17:49:24","name":"Phoscon#B1436x632"}},"zigbeechannel":11}
[2021-02-02 12:59:59][DEBUG] : {"00212EFFFF04290A":{"config":{"group":"2","on":true,"reachable":true},"e":"changed","id":"39","r":"sensors","t":"event","uniqueid":"00:04:74:00:00:0f:3c:26-01-0006"}}
[2021-02-02 12:59:59][DEBUG] : {"00212EFFFF04290A":{"config":{"group":"18","on":true,"reachable":true},"e":"changed","id":"39","r":"sensors","t":"event","uniqueid":"00:04:74:00:00:0f:3c:26-01-0006"}}
[2021-02-02 12:59:59][DEBUG] : {"00212EFFFF04290A":{"e":"changed","id":"7","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"00:04:74:00:00:0b:ef:4f-01"}}
[2021-02-02 12:59:59][DEBUG] : {"00212EFFFF04290A":{"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}}
[2021-02-02 12:59:59][DEBUG] : {"00212EFFFF04290A":{"attr":{"id":"8","lastannounced":"2021-01-28T14:19:12Z","lastseen":"2021-02-02T11:59Z","manufacturername":"Legrand","modelid":"Dimmer switch w\/o neutral","name":"Dimmable light 8","swversion":"0026","type":"Dimmable light","uniqueid":"00:04:74:00:00:9c:d3:1b-01"},"e":"changed","id":"8","r":"lights","t":"event","uniqueid":"00:04:74:00:00:9c:d3:1b-01"}}
[2021-02-02 12:59:59][DEBUG] : Something received and no value updated, performing sync to create commands if needed
```

Smanar, 2021-02-02:
> The module is correctly created in third app (jeedom), with 1 option « 01-0006.state::buttonevent » but nothing is comming

So all seem fine, the device is correclty reconized. But if i m right on jeedom you can read the device JSON, on "Noeud" if I remember, else I can search on forum.
Just to be sure you have "null" too on "buttonevent".

cocapic, 2021-02-02:
`Information noeud
{
  "config": {
    "battery": null,
    "group": "2",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "314e9d627ee18b646a15d38d318dc170",
  "lastseen": "2021-02-02T12:09Z",
  "manufacturername": "Legrand",
  "mode": 1,
  "modelid": "Remote toggle switch",
  "name": "micro module 1",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "swversion": "0020",
  "type": "ZHASwitch",
  "uniqueid": "00:04:74:00:00:0f:3c:26-01-0006"
}`

Also in previous log is _Something received and no value updated, performing sync to create commands if needed_ ok ?

Smanar, 2021-02-02:
Ha yes I remember now, legrand have modified their firmware, switch now use toogle instead on/off.

Can be the explanation, because yes you have problem.

You are sure for the wiring ?
Have you the GUI ? If I remember jeedom use headless installation but have log visible inside the app.

If I m right when you will use the device you will have in logs, a line starting by
> [INFO] - No button map for

If you have the GUI, deconz/help/log with "info" and "info_l2".
I don't know jeedom, but you can change a debug level in the configuration page and have deconz log (not jeedom one)

Smanar, 2021-02-04:
@cocapic nothing in logs ?