dresden-elektronik/deconz-rest-plugin #2721: New Device Support Request - Visonic NEXT PLUS K9-85 SMA

meremortals70, 2020-04-26:
Would you please kindly add support for the above?

[image]

[image]

Thank you

SwoopX, 2020-04-26:
Please provide all relevant screenshots. You may take #2583 as an example. It is also important that the attributes have been read before taking them (its not the case above). Thanks!

meremortals70, 2020-04-26:
I apologise for not providing all which was required.

Then I read the basic cluster this is what is provided.

[image]
[image]

SwoopX, 2020-04-27:
The device unfortunately does not seem to have paired correctly, as I cannot imagine the shown clusters (just basic) and the device profile (ffff) are correct.

Try physically resetting the device (e.g. there should be a button on the device which must be pressed or certain amount of time) and do a new sensor search again in Phoscon.

Also, when making the screenshots, please ensure the devices are "awake". You can wake them by pushing any physical button on the device repeatedly, wave you hand infront of a motion sensor, move the magnet for a dor sensor, etc.. All the attributes are not populated with usable values right now.

meremortals70, 2020-04-27:
No difference, I deleted and re-added.

Is it possible that although this is a zigbee device it is not ZHA compliant?

SwoopX, 2020-04-27:
@ebaauw does that somehow ring a bell how the node is presented? This look like far from being complete. I've seen something similar so far only twice: once with the Xiaomi ZB3.0 smart plug (surprisingly) and accidentially yesterday with one of the Develco smoke sensors.

@meremortals70 Try reading the node descriptor and the simple descriptors once again (can be done via the black bullet right next to the one which is selected in your screenshot).

ebaauw, 2020-04-27:
> Is it possible that although this is a zigbee device it is not ZHA compliant?

The screenshot shows that the device advertises the ZHA profile on endpoint 0x01, so that's not the case here.

There are plenty of Zigbee devices that use a different profile than ZHA (or ZLL) and consequently aren't supported by deCONZ.  I doubt you'll find many of these in the consumer market though.

>  does that somehow ring a bell how the node is presented? 

Yes, many versions ago, before Manuel improved the discovery.  @meremortals70, what version of deCONZ and RaspBee/ConBee firmware are you running.

It could actually mean two things:
- The device indeed only advertises this endpoint and cluster in the descriptors.  I have no idea what kind of device this is supposed to be, but we've seen some _IAS Zone_ devices that don't advertise all clusters;
- deCONZ hasn't read the descriptors.  Try to read them (select the node, pressing `1` and `8` several times, while keeping the device awake).  You might need to close the network - I've seen some intermittent issue with deCONZ not sending the _Descriptor Request_ message while the network is open.  Where you able to read the _Basic_ cluster attributes?

meremortals70, 2020-04-28:
deconz v2.05.75
Conbee Firmwave v26490700

The unit is a residential/commercial PIR used by Telstra's Smart Home in Australia.

It is no longer listed on the Visonic website so obviously deprecated.

The replacement zigbee PIR's on the website says zigbee alliance compliant ZHA1.2 and OTA upgradeable.

I will delete/add and try reading again.

meremortals70, 2020-05-02:
I have retried setting up this sensor with the same result.

I think we can assume and from some googling that it is not ZHA compatible.

The vendor's website lists a new model which is ZHA compatible and I have ordered.

closed 2020-05-02 by meremortals70

SwoopX, 2020-05-04:
@meremortals70 Since I'm just playing around, you could try the following with the sensor:

- Physically reset the device
- Start sensor seach in Phoscon
- After like 10 secs, go to deconz GUI and set permit join to 0 (EDIT -> Permit Join)
- Now read the simple descriptors

The result should be that you have to black bullets on the right side of the device (like with the others).