dresden-elektronik/deconz-rest-plugin #2722: New Device Support - Sercomm SZ-DWS04 Micro Door Window Sensor

meremortals70, 2020-04-26:
Please add support for the above Device

[image]

[image]

Thank you

SwoopX, 2020-04-26:
Please provide all relevant screenshots. You may take #2583 as an example. Thanks!

meremortals70, 2020-04-26:
I apologise for not providing all the information.  Your info on providing info only mentions the basic cluster.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

SwoopX, 2020-04-27:
Unfortunately, the screenshots are vastly not usable. When making the screenshots, please ensure the devices are "awake". You can wake them by pushing any physical button on the device repeatedly, wave you hand infront of a motion sensor, move the magnet for a dor sensor, etc.. All the attributes are not populated with usable values right now.

meremortals70, 2020-04-27:
Lets try this again. I deleted the device and re-added

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

SwoopX, 2020-04-27:
Much better now, thanks! I'll let you know if I need anything else.

SwoopX, 2020-05-10:
@meremortals70 Could you please double check the value for battery voltage of the power configuration cluster? Please double click on the value and the press read in the opened dialog (not read config). 255 does not feel right but eventually, only low battery is reported through battery alarm mask...

meremortals70, 2020-05-11:
Thanks for working on this.  Does this look better?

[image]

SwoopX, 2020-05-11:
Indeed, thanks!

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix