dresden-elektronik/deconz-rest-plugin #2724: Issue Tracker for Phoscon

pattyland, 2020-04-26:
Hi,

I asked this question already [here](https://github.com/dresden-elektronik/phoscon-app-beta/issues/66) but maybe that repository is not longer monitored.

What is the right place to report bugs in Phoscon?

Best Regards
Sören

SwoopX, 2020-04-26:
You may reach out to Dresden Elektronik support.

pattyland, 2020-04-28:
Hm ok, is Phoscon not on GitHub?
I would also do a PR if the developers agree to my proposal

SwoopX, 2020-04-28:
Well, the repository says it's no longer maintained so from my perspective, their support is the only chance you have.

SwoopX, 2020-05-28:
Phoscon repo has been reopened. You may want to raise it there and close the issue here. Thanks!

StephanStS, 2020-05-30:
You find the Phoscon repo [there](https://github.com/dresden-elektronik/phoscon-app-beta/issues).
@pattyland: Would you then please close this issue?

closed 2020-06-06 by Mimiix