dresden-elektronik/deconz-rest-plugin #2725: The easiest way to set power on defaults for lights?

mman, 2020-04-27:
Hi guys,

I am using lot of OSRAM lights with deConz and Conbee II stick and for historical reasons (and because I have not found any zigbee wall switches that can replace regular ABB switches) they all are wired to be physically powered on and off.

OSRAM lights provide (if I understand it correctly) a special OSRAM cluster with "set-poweron-defaults" method that when executed remembers the current state of the light and uses it the next time the light is physically powered on.

What is the best way to invoke this method programatically via API so that I do not need to open the deCONZ GUI and invoke the cluster manually?

thanks,
Martin

ebaauw, 2020-04-27:
> What is the best way to invoke this method programatically via API

Not, as the API doesn't support this.  It's only a one-time setting, so very low prio to add API support for this.

mman, 2020-04-27:
Thanks @ebaauw for the explanation, what would be the component where I would put the code to invoke this cluster automatically after say 10 seconds every time the light parameters change?

What I would like to achieve is when RGB light is adjusted via Siri, and then physically switched off, to get the last specified color when it is powered on again.

Sounds like this must be part of deconz executable that talks to zigbee, right?

ebaauw, 2020-04-27:
I don't think OSRAM lights support a power-on setting to restore the last colour (unlike the Hue lights).  So you'd have to update the power-on setting on each state change.  This is a bad idea, as the light's non-volatile memory (used to store the power-on settings) likely has a limited number of write cycles.

Even if you don't care about the light's life span, I see no easy way to achieve this without API support.  You could issue the command from a script calling the deconz-cli-plugin, but then you'd need to detect that the light state has changed to call the script.  You'll probably find some homebridge plugin to do that, but then you need a HomeKit automation to detect the light change.  Never tried that myself - not sure if it'll work.  If you don't mind to wear down your lights even quicker, you could run the script to store the light setting every _n_ seconds from crontab.  
Even with API support, I'm not sure if a deCONZ rule would be able to detect a colour change, but you could bluntly use `lastupdated`.  I'd use a `ddx` condition to only store the light state after say 10-15 seconds, to wait for the user to finish changes to the light state.

mman, 2020-04-27:
This is what I see @ebaauw for one of the old Zigbee LED strips from OSRAM, ``FC0F`` cluster with ``Store Power-ON Defaults`` method. When executed with color changed, it resumes on next power on with the last color that was set this way. I think this is precisely the method the official OSRAM app uses to store power on defaults.

[image]

And thanks for all the explanation about NVRAM, it all makes sense.

ebaauw, 2020-04-27:
> FC0F cluster with Store Power-ON Defaults method.

Yes, I know.  I was the one who added this cluster and the command to `general.xml` so the deCONZ GUI shows them.

> I think this is precisely the method the official OSRAM app uses to store power on defaults.

Yes it is.  I sniffed the Zigbee messages sent by the Lightify gateway to discover this.

mman, 2020-04-27:
Oh cool, I guess I am stuck at the moment in a situation where I want to control the lights like a human using the non-smart switches, and facing the limitations of the "always on" smart things design :)

ebaauw, 2020-04-27:
Yeah, you need to decide what century you want to be in.  ;-)  Using 20th century wall switches with 21st century smart lights doesn’t work too well.  Also not a good idea to power down Zigbee routers, causing constant reconfigurations of the mesh network.  It’s probably a plot by the vendors to sell you more devices...

mman, 2020-04-28:
I'm staying in 20th until ABB (or somebody else) comes up with a little zigbee 220V powered gadget that will turn existing wall light switch into a smart one without replacing the externals. And I think I'm not alone. We in software engineering use the term https://en.wikipedia.org/wiki/Wife_acceptance_factor for this :)

So for now, I have come up with the following plan:

1/ install couple always-on zigbee ikea power outlets hidden around  the house to make the zigbee network stable when lights come and go off/on mains.

2/ modified ``homebridge-hue`` installation to expose ``groups: true`` and ``lights: false`` which seems to address the racing issues around ``deconz`` and ``homebridge`` startup and also seems to address homekit removing lights from rooms.

3/ Use deconz gui to store power on defaults for RGB lights and simply live with the fact that they will restart to a known color after mains off/on.

I have also found this nice thread of yours: https://github.com/ebaauw/homebridge-hue/issues/41.

Thanks for all the explanations, I'm closing this issue as resolved.

closed 2020-04-28 by mman

ebaauw, 2020-04-28:
> I'm staying in 20th until ABB (or somebody else) comes up with a little zigbee 220V powered gadget that will turn existing wall light switch into a smart one without replacing the externals.

I don't think I've seen main-powered wireless switches.  There's plenty of battery-powered switches, and even ZGP (green power) switches that harness kinetic energy, that fit standard EU wall switch frames.  Unfortunately, not the cheapest.

> to address the racing issues around deconz and homebridge startup and also seems to address homekit removing lights from rooms.

That's no race condition, but a bug in deCONZ, which has been resolved multiple times already, but keeps popping up.  See https://github.com/ebaauw/homebridge-hue/issues/471.  Best workaround is to create a `lightlist` resourcelink to make Homebridge Hue wait for deCONZ to expose all lights.

> I have also found this nice thread of yours

That's a very old one, from before the Hue lights supported power-on state in their firmware.

mrquincle, 2020-06-09:
> I'm staying in 20th until ABB (or somebody else) comes up with a little zigbee 220V powered gadget that will turn existing wall light switch into a smart one without replacing the externals. And I [...]

This is a great idea. I've suggested it for our team. We already upgrade a conventional power outlet with something we call [switchcraft](https://crownstone.rocks/2018/03/13/switchcraft-and-dimming). Basically, you'll adjust your switch so that it's a "push to break" switch. In the firmware of the Crownstone (which is placed near the light where you have both line and neutral) we register a short dip in the power. This detection can than trigger a relay.

Of course, I realize that you don't use Bluetooth (I concur with @ebaauw that most implementations are too slow - although we aren't slow by any means). However, what I'm think of is to decouple our "switch recognition" software from our actuator. That means that we might send out "switch events" towards third parties. Of course, then you'll have a round-trip that does actually incur latency. I mean then it goes: switch -> Crownstone switch detection -> Crownstone mesh -> hub -> Zigbee hub -> Zigbee lamp. 

Summarized, the ability to **recognize a switch event** is a cool feature. :-)