dresden-elektronik/deconz-rest-plugin #2726: Support for Philips (Signify) EasySmart LED tubes

robidog, 2020-04-27:
I was wondering what the chances are for deCONZ to recognize Philips EasySmart LED retrofit tubes. They claim Zigbee 3.0 standard, but I could not find any details beyond that.

https://www.usa.lighting.philips.com/products/product-highlights/instantfit/easysmart 

Has anyone ever attempted to control these with deCONZ?

jensflorian, 2020-04-28:
Ledvance / Osram has identical Zigbee T8 tubes,  they are labeled  Zigbee 3.0 however the seem to pair only with custom sensors. https://m.youtube.com/watch?v=qbqXPgnqTWw

See comment about pairing here 
https://www.ledvance.com/appsinfo/pdc/pdf.do?cid=GPS01_3509494&vid=PP_EUROPE_DE_eCat%2cPP_EUROPE_Europe_retail_eCat&lid=EN

I believe this is very similiar to your Philips lamps.

jensflorian, 2020-05-04:
Update: I just ordered one of these Ledvance / Osram Substitubes. They pair successfully with Deconz. So the manufacturer comments about custom sensors is misleading.

robidog, 2020-05-04:
Great, thanks for the update.

rkjnsn, 2020-05-05:
I just got a set of Philips EasySmart USB tubes. I *was* able to get them to pair with Deconz, but it took a few tries and some futzing. They didn't seem to want to show up when scanning, but I decided to reset the gateway, and they all popped up in deCONZ once it had restarted. After they were connected, I still had to scan for lights a couple of times in Phoscon for them to be identified as lights and show up there.

I also got the IR remote with my bulbs, and I'm honestly not sure if it helped or not, but it does let one reset the lights if something goes wrong, and I was able to use it to turn the lights on and off before I got them paired.

As a side note, the first thing I had to do was replace my ballasts, since I had older rapid-start T8 ballasts, and the EasySmart bulbs only work with the newer instant-start ballasts. (I don't know if they would work with program-start ballasts, but I went with instant start because they're cheaper, officially supported, and the whole point of program-start is to preserve florescent bulb life, which isn't really relevant when using LEDs.)

robidog, 2020-05-05:
Interesting. I wonder if they require a ballast at all if I rewire the sockets. Most T8 retrofit tubes just want to see mains power at the 2 pins on one of the ends.

jensflorian, 2020-05-05:
No, most LED tubes don't require a ballast. The "LED starters" are essentially a dummy with a small fuse. They cost usu. less than 1 US$/€ so replacing the old ballast with these is much easier than to rewire the whole lamp holder.

rkjnsn, 2020-05-05:
I did not try wiring them directly to mains, so I don't know if that would work, but the fact that they don't work with rapid-start ballasts suggests they aren't very forgiving. Instan-start ballasts provide hot to both pins at one end of the bulb, and neutral to both ends at the other, so I doubt the bulb would work with hot and neutral at the same end, and it might even cause a short circuit if the pins are shunted within the bulb.

Mimiix, 2020-06-06:
Without any devices, its hard to add a device to the supported devices list.

Closing for now.

closed 2020-06-06 by Mimiix