dresden-elektronik/deconz-rest-plugin #2727: ConBee II Stick is not recognized in Win10 or Raspbian

RockoRocknase, 2020-04-27:
Hi guys,

just received my ConBee II USB-stick and was happy to test. Neither my raspbian Linux installation nor my Windows 10 installation is recognizing the ConBee stick.

Raspbian:
_lsusb_ does not show any device from dresden-elektronik. No difference if I plug in the stick directly to my  Raspberry or via a USB-cable.

Windows 10:
After plugging in the stick nothing happens. Also the device manager is not showing any difference or new device or new interface.

deConz Software is not showing any device to connect to.

At the moment I think the ConBee II stick is defect on arrival. Do you have any ideas to share with me to check the stick?

Thank you.

Smanar, 2020-04-27:
I m sorry but if you haven't reaction on PC, I think you can contact directly the support.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix