dresden-elektronik/deconz-rest-plugin #2728: Ikea on-off switch E1743

gck303, 2020-04-28:
I am a new owner of a conbee II stick. I am running it in a docker on a HP380e Ubuntu server. 

Can the attached Ikea switches please be supported? I have tried adding them, but cannot. Only two Ikea switches seem to be available to be added (the rotating dimmer and five button control). 

I am happy to supply any additonal information needed. I only have access to the phoscon app, as I am not sure how to open up the X interface on a CLI-only machine. 

Many thanks, George

[image]
[image]

jensflorian, 2020-04-29:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/853#issuecomment-458695666

gck303, 2020-04-29:
Thank you. That works.

closed 2020-04-29 by gck303