dresden-elektronik/deconz-rest-plugin #2729: Can't get API KEY

CarpeDiemRo, 2020-04-28:
**Prerequisites**

1. Conbee II USB stick
2. RPI 4
3. Hassio 0.108.9
4. Postman on macOS
5. RESTClient 2 for Mozilla

**How to reproduce**
`GET https://dresden-light.appspot.com/discover`
--> 200 OK

`POST 172.30.33.1:40850/api` or
`POST https://172.30.33.1:40850/api`

> Could not get any response
There was an error connecting to http://172.30.33.1:40850/api.
Why this might have happened:
The server couldn't send a response:
Ensure that the backend is working properly
Self-signed SSL certificates are being blocked:
Fix this by turning off 'SSL certificate verification' in Settings > General
Proxy configured incorrectly
Ensure that proxy is configured correctly in Settings > Proxy
Request timeout:
Change request timeout in Settings > General

--> in Postman and NO answer in RESTClient 2

The body is 
`{
    "devicetype": "myapplication"
}`

and **raw** button is checked

Smanar, 2020-04-28:
Hello, perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2716

CarpeDiemRo, 2020-04-28:
@Smanar Thank you, I will try.

CarpeDiemRo, 2020-04-29:
Everything is OK. It's worth mentioning that the IP that must be used is the **server IP** not the internal IP of the Deconz Gateway. 
 Therefore: 192.168.x.y. instead of 172.30.33.1

closed 2020-04-29 by CarpeDiemRo