dresden-elektronik/deconz-rest-plugin #2732: New Device Support - RGBgenie ZB-5121

puithove, 2020-04-28:
I'd like to request support of these nice battery powered ZigBee 3.0 remotes.  I previously had these working well under Home Assistant's ZHA but am working on converting over to deconz.  Under ZHA, they create very reliable zha_event events that can be used to trigger automation.  I'd like to be able to continue to use these in deconz the same way, exposing deconz_event to Home Assistant.

https://rgbgenie.com/?product=rgbgenie-mini-remote-with-single-scene-recall-zigbee

[image]

[image]

[image]

[image]

SwoopX, 2020-05-10:
Can you also please provide a screenshot of the node info panel?

puithove, 2020-05-10:
@SwoopX It looks like that last screenshot is the Node Info panel?  Is there something that I'm missing?  Can you give me an example of what to look for?

puithove, 2020-05-10:
Here's another one, just in case
[image]

SwoopX, 2020-05-10:
Sry, my bad. Thanks anyway.

puithove, 2020-05-11:
No worries 👍 let me know if you need anything else.

gmitch64, 2020-05-12:
Would it be possible to add support for the remote in the issue that I just see linked :)

G

SwoopX, 2020-05-22:
So, support should be in and available with version .76. You need to try it out and let me know if it works alright.

puithove, 2020-05-23:
Nice, I will check it out and let you know here once I see my deconz container updated in HA.  Thanks!

puithove, 2020-05-24:
I added it via Switches (like I did my one Phillips Hue Dimmer and the others of these) in Phoscon, but while deconz registered it, Phoscon didn't.

[image]
[image]
[image]

You can see that it was recognized properly by deconz now.  However, it is not generating deconz_event

SwoopX, 2020-05-24:
Hm, have you tried pushing some buttons while it was searching? Maybe also try the sensor seach to see if that has any notable effect.

puithove, 2020-05-25:
I did, but it doesn't seem to change anything.  Same results unfortunately.  Is there a log file I can collect that might help?

SwoopX, 2020-05-25:
Absolutely. You can rund deconz with extensive debug output enabled if you add `--dbg-info=2` to the binary call and pipe the output to a file. When you then press each button short, long and double, that should give more insights. You just need to recall that order, so I know which button was pressed.

I've also provided an update for the remote, as I recognized a probably non working button mapping.

puithove, 2020-05-26:
I'm still on .76

Unfortunately since I'm running deconz with HA's built-in addon, HA is managing the docker container and I don't see if/where it writes out the console output.  It does show the output in their web UI, so I can grab just snippets, it's difficult to send the whole log (which I'd normally do).  Here is a snippet for the short-press "Off" button (there is no long-press action for the On or Off buttons on this remote). 
`07:29:13:667 poll node 00:17:88:01:08:14:90:cc-0b
07:29:13:668 Poll light node Living Room - Hanging Vintage
07:29:13:720 read attributes of 0x00178801081490CC cluster: 0x0006: [ 07:29:13:720 0x0000 07:29:13:720 ]
07:29:13:720 add task 2215 type 19 to 0x00178801081490CC cluster 0x0006 req.id 124
07:29:13:721 Poll APS request 124 to 0x00178801081490CC cluster: 0x0006
07:29:14:242 Idle timer triggered
07:29:14:243 Force read attributes for SensorNode Garage - Door
07:29:16:242 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
07:29:16:341 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0006 cmd: 0x00 pl[0]: 000
07:29:21:183 node 0EC1BBDFFFE9870B3 leave wait state
07:29:21:243 Idle timer triggered
07:29:21:243 Force read attributes for SensorNode Workshop - Motion Sensor
07:29:21:269 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
07:29:21:342 node 0001788010817BBBE leave wait state`. 

The only thing I can see is the `no button map` 

for each button that I press, but I'll grab the surrounding lines in case there's something there that's useful for you.

Here is the short-press "On" button:
`07:33:56:242 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
07:34:00:263 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0006 cmd: 0x01 pl[0]: 000
07:34:01:242 Idle timer triggered
07:34:01:243 binding for attribute reporting SensorNode Foyer - Motion Sensor of cluster 0x0400 seems to be active
07:34:01:243 Force read attributes for SensorNode Foyer - Motion Sensor
07:34:01:271 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)`

Short-press "Dim" button:
`07:34:57:243 Force read attributes for SensorNode Garage - Door
07:35:00:991 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x06 pl[0]: 001
07:35:01:268 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)`

Long-press and release "Dim" button:
`07:35:54:765 Node data 0x286d9700010a7d8d profileId: 0x0104, clusterId: 0xFC02
07:35:54:765 0x286D9700010A7D8D: update ZCL value 0x01/0xFC02/0x0012 after 0 s
07:35:54:765 0x286D9700010A7D8D: update ZCL value 0x01/0xFC02/0x0013 after 0 s
07:35:54:765 0x286D9700010A7D8D: update ZCL value 0x01/0xFC02/0x0014 after 0 s
07:35:54:765 no button map for: multi ep: 0x01 cl: 0xFC02 cmd: 0x0A pl[0]: 012
07:35:54:765 ZCL attribute report 0x286D9700010A7D8D for cluster: 0xFC02, ep: 0x01, frame control: 0x0C, mfcode: 0x1241 
07:35:54:765 	payload: 120029f4ff130029e2ff14002911fc
07:35:54:766 Websocket 172.30.32.1:48632 send message: {"e":"changed","id":"28","r":"sensors","state":{"lastupdated":"2020-05-26T11:35:54.764","orientation":[-12,-30,-1007],"vibration":false},"t":"event","uniqueid":"28:6d:97:00:01:0a:7d:8d-01-fc02"} (ret = 194)
07:35:54:943 node 000178801085172A4 leave wait state
07:35:55:743 poll node 00:17:88:01:03:ab:53:0b-0b
07:35:55:743 Poll light node Upstairs Hall - Globe
07:35:55:792 read attributes of 0x0017880103AB530B cluster: 0x0006: [ 07:35:55:793 0x0000 07:35:55:793 ]
07:35:55:793 add task 4110 type 19 to 0x0017880103AB530B cluster 0x0006 req.id 28
07:35:55:793 Poll APS request 28 to 0x0017880103AB530B cluster: 0x0006
07:35:56:242 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
07:35:56:368 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 001
07:35:57:342 node 0001788010817BBBE leave wait state
07:35:58:589 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
07:36:00:243 Idle timer triggered
07:36:00:243 Force read attributes for SensorNode Office - Motion Sensor`

Short-press scene button (Scene recall):
`07:38:08:971 Poll APS request 111 to 0xEC1BBDFFFE9870B3 cluster: 0x0006
07:38:11:271 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
07:38:11:677 Websocket 172.30.32.1:48632 send message: {"e":"scene-called","gid":"0","r":"scenes","scid":"1","t":"event"} (ret = 66)
07:38:11:678 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0005 cmd: 0x05 pl[0]: 000
07:38:13:242 timeout on poll APS confirm
07:38:13:242 Idle timer triggered
07:38:13:242 Force read attributes for SensorNode Upstairs Bath - Motion Sensor
07:38:14:168 poll node 00:17:88:01:08:14:22:de-0b
07:38:14:168 Poll light node Master Bed - Shower 1`

Long-press scene button (Scene Store):
`07:39:23:242 Idle timer triggered
[... 36 lines cut ...]
Short-press brighten button:
`07:40:58:462 node 00017880104067772 leave wait state
07:41:00:813 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x06 pl[0]: 000
07:41:01:243 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
07:41:01:243 Idle timer triggered
07:41:01:243 Force read attributes for SensorNode Garage - Door
07:41:01:270 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)`

Long-press and release brighten button:
`07:42:06:828 Erase task req-id: 159, type: 19 zcl seqno: 248 send time 0, profileId: 0x0104, clusterId: 0x0006
07:42:07:662 node 000178801085172A4 leave wait state
07:42:07:817 poll node 00:17:88:01:03:ab:53:0b-0b
07:42:07:817 Poll light node Upstairs Hall - Globe
[... cut ...]
07:42:09:105 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 000
07:42:10:795 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
07:42:11:242 Idle timer triggered
07:42:11:244 binding for attribute reporting SensorNode Office - Motion Sensor of cluster 0x0400 seems to be active
07:42:11:244 Force read attributes for SensorNode Office - Motion Sensor
07:42:11:281 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)`

Hopefully that's helpful, but if not let me know what else I can do.  I appreciate you looking at this.

SwoopX, 2020-05-26:
👍 A fix for the button map is already part of my PR. However, that info should help to double chek it. Will have a closer look tonight.

puithove, 2020-05-26:
FWIW - I saw an update was available for deconz - updated to .77, reset and re-paired remote, but same results - doesn't show up in Phoscon, no events in HA (confirmed events are working from the Hue Dimmer shown above).

SwoopX, 2020-05-27:
The PR is still open, so it has not yet been merged.

SwoopX, 2020-05-27:
Hey, thanks for that debug output. Based on that, the button map had to be updated. Can you please double check long press dim up and long press dim down once again? It feel a bit weired that the command upon release is identical for both buttons...

puithove, 2020-05-28:
Sure thing.  Here is dim up long press and release.
`07:25:34:272 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
07:25:35:860 MAC Poll 0x02 0xBFA3
07:25:37:019 add task 647 type 14 to group 0x0002 cluster 0x0006 req.id 49
07:25:37:021 Websocket 172.30.32.1:44194 send message: {"e":"changed","id":"8","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"90:fd:9f:ff:fe:61:93:d1-01"} (ret = 140)
07:25:37:024 Websocket 172.30.32.1:44194 send message: {"e":"changed","id":"2","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 89)
07:25:37:026 discard group state push for 2: state/any_on (already pushed)
07:25:37:084 Erase task req-id: 49, type: 14 zcl seqno: 80 send time 0, profileId: 0x0104, clusterId: 0x0006
07:25:37:131 Node data 0x90fd9ffffe6193d1 profileId: 0x0104, clusterId: 0x0006
07:25:37:131 0x90FD9FFFFE6193D1: update ZCL value 0x01/0x0006/0x0000 after 0 s
07:25:37:131 ZCL attribute report 0x90FD9FFFFE6193D1 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
07:25:37:131 	payload: 00001000
07:25:37:158 Node data 0x0017880106f73465 profileId: 0x0104, clusterId: 0x0406
07:25:37:158 0x0017880106F73465: update ZCL value 0x02/0x0406/0x0000 after 0 s
07:25:37:158 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
07:25:37:158 ZCL attribute report 0x0017880106F73465 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
07:25:37:158 	payload: 00001800
07:25:37:160 Websocket 172.30.32.1:44194 send message: {"e":"changed","id":"30","r":"sensors","state":{"lastupdated":"2020-05-28T11:25:37.158","presence":false},"t":"event","uniqueid":"00:17:88:01:06:f7:34:65-02-0406"} (ret = 163)
07:25:37:161 discard sensor state push for 30: state/lastupdated (already pushed)
07:25:37:286 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 000
07:25:39:243 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
07:25:39:540 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
07:25:39:608 MAC Poll 0x02 0xBFA3
07:25:39:608 APS-DATA.indication from child 0xBFA3
07:25:39:609 Node data 0x001788010646167e profileId: 0x0104, clusterId: 0x0402
07:25:39:609 0x001788010646167E: added ZCL value 0x%02/0x0002/0x0402
07:25:39:609 no button map for: SML002 ep: 0x02 cl: 0x0402 cmd: 0x0A pl[0]: 000
07:25:39:609 ZCL attribute report 0x001788010646167E for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
07:25:39:609 	payload: 0000296009
07:25:39:610 Websocket 172.30.32.1:44194 send message: {"e":"changed","id":"21","r":"sensors","state":{"lastupdated":"2020-05-28T11:25:39.608","temperature":2400},"t":"event","uniqueid":"00:17:88:01:06:46:16:7e-02-0402"} (ret = 165)
07:25:39:612 discard sensor state push for 21: state/lastupdated (already pushed)`

And here is dim down long press and release
`07:30:29:264 Current channel 25
07:30:29:276 Device TTL 1891 s flags: 0x7
07:30:30:480 MAC Poll 0x02 0xBFA3
07:30:30:480 verify 0x001788010646167e is child node after 10 s
07:30:30:486 verify neighbor status: APP_SUCCESS (0x00) 
07:30:31:130 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 001
07:30:31:998 node 0001788010498A343 leave wait state
07:30:32:634 no button map for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
07:30:34:243 Idle timer triggered
07:30:34:243 Force binding of attribute reporting for node Master Bed - Shower 3
07:30:34:243 binding for attribute reporting SensorNode Office - Motion Sensor of cluster 0x0406 seems to be active
07:30:34:243 Force read attributes for SensorNode Office - Motion Sensor
07:30:34:275 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
07:30:34:965 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x8124
07:30:35:243 Close socket port: 51044
07:30:35:511 MAC Poll 0x02 0xBFA3`

FWIW - These remotes can be paired via Touch Link and control lights directly even when joined to an existing Zigbee network.  When doing that, it adds each paired light to its Zigbee group and sends on/off/dim/brighten/scene store/scene recall commands to the group.  When lights send their group membership, deconz/phoscon show the group.  (I'm still learning Zigbee things, but that's what I've figured out for these :) )

I didn't capture all the events when I had it under ZHA, but I did have a couple of automations based off it.  Looking back at a previous version of my automations, I didn't have one tied to the long press on both up and down, but I did have one on the long-press of dim down.  I can see that the short press had just one attribute that was unique between up and down buttons. (this is from a different remote from what I'm using above for testing)

Short-press dim-up
`  - event_data:
      args:
      - 0
      - 50
      command: move_with_on_off
      unique_id: 00:0d:6f:00:10:65:92:d6:1:0x0008
`

Short-press dim-down
`  - event_data:
      args:
      - 1
      - 50
      command: move_with_on_off
      unique_id: 00:0d:6f:00:10:65:92:d6:1:0x0008
`
Long-press dim-down
`  - event_data:
      args:
      - 1
      - 32
      - 0
      command: step_with_on_off
      unique_id: 00:0d:6f:00:10:65:92:d6:1:0x0008
`

There was of course more data in the event that I didn't use.

tchellomello, 2020-06-10:
Hello everyone, I'm also a Home Assistant user and I'm trying to add an `RGBgenie ZB-5028` https://rgbgenie.com/?product=rgbgenie-rgbw-remote-zigbee 

[image]

I'm hitting a similar issue. I tried to join the controller as reported https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2732#issuecomment-633210387 which resulted in the same way. The interface does not map the controller. 

I'm running the controller on a kubernetes cluster and interesting I can see the deconz found the controller, no mapping attribute happens and therefore, no event is generated by `deconz_event`. I can see some stuff when I press buttons like this:

```
# power
[zigbee-bridge-56f97fc78c-z2826] 00:44:01:084 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0006 cmd: 0x01 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:44:01:094 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0006 cmd: 0x01 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:44:01:098 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0006 cmd: 0x01 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:44:01:102 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0006 cmd: 0x01 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:44:02:078 CTRL skip polling while permit join is set 

# select button ##1 and then used the color wheel
[zigbee-bridge-56f97fc78c-z2826] 00:44:32:179 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 002 
[zigbee-bridge-56f97fc78c-z2826] 00:44:32:556 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 076 
[zigbee-bridge-56f97fc78c-z2826] 00:44:33:133 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:44:33:162 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 083 
[zigbee-bridge-56f97fc78c-z2826] 00:44:33:540 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 0F3 
[zigbee-bridge-56f97fc78c-z2826] 00:44:34:136 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:44:34:411 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 068 
[zigbee-bridge-56f97fc78c-z2826] 00:44:34:824 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 0BE 
[zigbee-bridge-56f97fc78c-z2826] 00:44:35:141 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:44:35:361 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 07A 
[zigbee-bridge-56f97fc78c-z2826] 00:44:35:390 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 03F 
[zigbee-bridge-56f97fc78c-z2826] 00:44:35:621 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 028 
[zigbee-bridge-56f97fc78c-z2826] 00:44:36:035 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 0B0 
[zigbee-bridge-56f97fc78c-z2826] 00:44:36:139 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:44:36:323 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0300 cmd: 0x07 pl[0]: 03F 
[zigbee-bridge-56f97fc78c-z2826] 00:44:37:142 CTRL skip polling while permit join is set 

## using bright buttons
[zigbee-bridge-56f97fc78c-z2826] 00:47:19:612 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0008 cmd: 0x06 pl[0]: 001 
[zigbee-bridge-56f97fc78c-z2826] 00:47:19:916 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0008 cmd: 0x06 pl[0]: 001 
[zigbee-bridge-56f97fc78c-z2826] 00:47:20:031 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:47:21:038 CTRL skip polling while permit join is set 
[zigbee-bridge-56f97fc78c-z2826] 00:47:21:048 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0008 cmd: 0x06 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:47:21:277 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0008 cmd: 0x06 pl[0]: 000 
[zigbee-bridge-56f97fc78c-z2826] 00:47:21:517 no button map for: RGBgenie ZB-5028 ep: 0x02 cl: 0x0008 cmd: 0x06 pl[0]: 000 
```

@SwoopX I would happy to test the PR if needed. Tomorrow I'll test using the remote directly to the Philips Hue light strip I have using TouchLink but I would love to have this supported on my controller. 

Thank you!

SwoopX, 2020-06-10:
Please issue a new device support request as per wiki, as this is a totally different device. Thanks!

puithove, 2020-06-23:
@SwoopX - hey, I wanted to let you know - I've upgraded to deconz 2.05.78, and I now see a battery entity in Home Assistant, and also seeing button events for some of the buttons.  I don't see the remote listed under Switches in Phoscon, but I haven't removed and re-paired yet either (I have a feeling I'd need to do that).  Some of the buttons actions still seem to have no mapping.  I'll include them here:

Short-press scene button (scene recall):
```no button handler for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0005 cmd: 0x05 pl[0]: 0x00```

Long-press scene button (scene store):
```no button handler for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0005 cmd: 0x04 pl[0]: 0x00```

Release of dim down button after long press (short press and long-press held down both create events, just the release is missing - release of dim up after long press works fine)
```no button handler for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 0x00```

Looking back above, I see there was some question on the release events.  They're clearly differentiated somehow since I see the "no button handler" in the log for the dim-down release with no event generated, but everything works fine for the dim-up release - event generated and no mention in the log.  FWIW, here is what I see on the event side in HA.

short-press dim down:
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "rgbgenie_zb_5121",
        "unique_id": "00:0d:6f:00:13:e9:d1:7e",
        "event": 4002
    },
    "origin": "LOCAL",
    "time_fired": "2020-06-23T19:11:48.401038+00:00",
    "context": {
        "id": "2434f6e441294c6e926dab99d74fdc13",
        "parent_id": null,
        "user_id": null
    }
}
```
short-press dim up:
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "rgbgenie_zb_5121",
        "unique_id": "00:0d:6f:00:13:e9:d1:7e",
        "event": 3002
    },
    "origin": "LOCAL",
    "time_fired": "2020-06-23T19:12:30.977081+00:00",
    "context": {
        "id": "0a33235db0c8492cbe50f3521a6c98ce",
        "parent_id": null,
        "user_id": null
    }
}
```
long-press dim down
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "rgbgenie_zb_5121",
        "unique_id": "00:0d:6f:00:13:e9:d1:7e",
        "event": 4001
    },
    "origin": "LOCAL",
    "time_fired": "2020-06-23T19:12:59.065967+00:00",
    "context": {
        "id": "eae8f475b1f445b58acff2e107ecaff2",
        "parent_id": null,
        "user_id": null
    }
}
```
long-press dim up
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "rgbgenie_zb_5121",
        "unique_id": "00:0d:6f:00:13:e9:d1:7e",
        "event": 3001
    },
    "origin": "LOCAL",
    "time_fired": "2020-06-23T19:13:31.897217+00:00",
    "context": {
        "id": "1f6ec51ea7a44e8eb511fee68c65247e",
        "parent_id": null,
        "user_id": null
    }
}
```
release after long-press dim up
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "rgbgenie_zb_5121",
        "unique_id": "00:0d:6f:00:13:e9:d1:7e",
        "event": 3003
    },
    "origin": "LOCAL",
    "time_fired": "2020-06-23T19:13:32.779308+00:00",
    "context": {
        "id": "717078d2b0c1473d91720ba5011ec71d",
        "parent_id": null,
        "user_id": null
    }
}
```

But we're almost there!!!!!

puithove, 2020-06-23:
Ok, I decided to remove and re-pair the remote, but it still does not show in Phoscon.

SwoopX, 2020-06-27:
The dim button should be fixed. For the scene button, I need to check back on a few details for creating groups. I'm pretty sure it works if a groups is assigned.

For visibility in Phoscon, you might want to raise an issue at the Phoscon beta repo.

SwoopX, 2020-08-11:
How does it look like with the latest beta release?

gmitch64, 2020-08-11:
Does the latest beta have support for the 5001 as well? I've not seen any mention of it recently, so I haven't upgraded beyond V77 yet. If there is support there, I'm happy to upgrade and test.

Mimiix, 2020-08-11:
@gmitch64 Why not upgrading anyway? Hard to say it does.

puithove, 2020-08-12:
@SwoopX ok, I'm on deconz 2.05.79 now.

Scene button still does not create any events in Home Assistant.  I get this from deconz (short press followed by long press of scene button)

```
17:45:50:214 no button handler for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0005 cmd: 0x05 pl[0]: 0x00
17:45:55:914 no button handler for: RGBgenie ZB-5121 ep: 0x01 cl: 0x0005 cmd: 0x04 pl[0]: 0x00
```

Other than that, the one remaining issue is the long-press release of the dim-down button.  It now appears to have a mapping, but it generates the same event ID in HA that the long-press release of the dim-up button.  Each button I expect to start with the same digit - like dim up, each code starts with 300x, and dim down short press and long press hold start with 400x - but the long press release of dim down returns 3003.

On:  100x
Off: 200x
Dim-up: 300x
Dim-down: 400x
Scene: 500x

Here's what I see in the deconz log for all the dim up button actions:

```
17:52:37:077 button 3002 Step up (with on/off)
17:52:39:367 button 3001 Move up (with on/off)
17:52:40:541 button 3003 Stop_ (with on/off)
```

And here's what I see for all the dim down button actions

```
17:53:30:374 button 4002 Step down (with on/off)
17:53:32:451 button 4001 Move down (with on/off)
17:53:33:666 button 3003 Stop_ (with on/off)
```

closed 2020-08-14 by manup