dresden-elektronik/deconz-rest-plugin #2733: RGBW bulb color temperature ...

Kristian8606, 2020-04-29:
Hello!  I have a problem with the rgbw bulb that is described here. https://github.com/ebaauw/homebridge-hue/issues/682#issuecomment-621470457
Can it be improved?

ebaauw, 2020-05-15:
Looking at the code, the REST API plugin creates `state.ct`, when it receives a value for _Color Temperature_, irrespective of the value of _Color Capabilities_.  
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1caed496f90b01c69be6f03886597004183152c8/de_web_plugin.cpp#L2529-L2547

I guess more devices report the wrong capabilities than colour temperature wrongly.  I'm afraid, for now, we have to whitelist the light to prevent this.

Could you attach a screenshot of the _Basic_ cluster (making sure the attributes have been read).

Kristian8606, 2020-05-15:
Is this enough?

[image]

ebaauw, 2020-05-15:
Yes, thanks.  The model looks like a generated password, I hope I can copy that without making a typo...

Kristian8606, 2020-05-20:
The problem is solved! 
Thanks @ebaauw

closed 2020-05-20 by Kristian8606