dresden-elektronik/deconz-rest-plugin #2734: New Device support - Iris keypad v3 (iL02_1)

bjonness406, 2020-05-01:
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-05-04:
Sorry to say that but I'm afraid that's a biggy. We currently do not have any logic implemented for the cluster 0501 (IAS ACE) which will be even more challenging without having such a device available to sniff the traffic.

bjonness406, 2020-05-11:
I see, thanks for the help anyway.

closed 2020-05-11 by bjonness406