dresden-elektronik/deconz-rest-plugin #2735: Slow reaponse time

Allras82, 2020-05-02:
Hi.

I'm suddenly having problems with the response time of my system... I sometimes have to command a switch 2-3-4 times before I get the response (light turns on/off)

I'm on latest version and I have the system running at a rpi3.
I have 71 nodes. Mix of xiaomi sensors, IKEA bulbs and a couple of HUE lamps.

Any suggestion how to solve this?

Br
Allan

Smanar, 2020-05-03:
Hello,

Have you try to run deconz in debug mode ?

>deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=0

As I have said, I have already see this situation with an user with a bad working ikea plug that slow down all the other devices.

Allras82, 2020-05-13:
Hi,

I think i have found that a philips hue lamp were causing the problem.

I have removed this lamp from the network and added again in the original HUE bridge.
And this setup is working just fine. So strange that this lamp were causing problems in the deconz setup.

Br
Allan

Smanar, 2020-05-13:
I can't explain, but as I have said before, I know an user who had exactly the same problem but with an ikea plug.
All was fine after removing it, and IDK if the device is working again now.

And now If you try to re-include it in deconz after a reset, the problem is back ?

Allras82, 2020-05-13:
I have not yet tested the reconnect...
I will try in the comming weekend.

Br. Allan

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix