dresden-elektronik/deconz-rest-plugin #2736: Zigbee meshing / routing optimization

Schaussi, 2020-05-03:
Hi,

using Conbee II with the recommended, current RPi4 image, I am trying to optimize my Zigbee network's stability, mainly the reliability of motion sensors.

- What works fine: Xiaomi motion sensor with direct connection to the gateway
- What only works in about 8 of 10 cases: Motion sensors which are routed e.g. via Osram smart plugs. Sometimes the sensors' signals seem not to reach the gateway for e.g. 30 seconds, afterwards it might work again for an hour, ... and so on.

Because it often helps to restart some of the smart plugs, I think the mesh/routing functionality does not work correctly on those devices.
Doing firmware updates (#2611), the situation got a little better, but did not solve the issue completely.

Disabling meshing on the routing nodes for testing purposes seems not to be possible.

Would it be useful (i.e. is it a good practice) to use multiple ConBee II sticks?
How do Zigbee devices like the Xiaomi motion sensor choose their preferred connection? (What do they see as "best": do they also check if they can reach the gateway through the router?, or do they take any connection which is available?
Are there any parameters I could change, e.g. in deConz GUI, which affect the routing behaviour of the smart plugs?

Thank you, best regards

Smanar, 2020-05-04:
About Xiaomi and Osram, you can find some answer here https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

And no you can't change anything.

But Xiaomi never change parent, so you can disable unwanted router, make inclusion for Xiaomi device and re-enable all router after some time.

pergolafabio, 2020-05-05:
i ditched my osram smart plugs, i hade xiami sensors connected to them too....
verry unreliable
changed osram with ikea smart plugs, never had issues anymore... also the range distance for ikea plus is far better then osram

Schaussi, 2020-05-11:
Thank you for your help, I have been running my network for a few days now without the Osram smart plugs. Everything is running fine... it's a pity Xiaomi is not really compatible with the signal repeating of Osram.

So I will try some of the recommended alternative smart plugs, or will maybe separate them into 2 networks.

closed 2020-05-11 by Schaussi