dresden-elektronik/deconz-rest-plugin #2737: Are Gledopto "plus" series controllers really compatible with Deconz?

SnaKKo, 2020-05-03:
I'm talking about the Gledopto RGB CCT controller, I have just ordered one of these controllers (to manage a LED strip) and meanwhile I read a lot of issues: I see that Gledopto gain a lot of popularity in these months so I hope that Deconz will be 100% compatible with these controllers.

I find very usefull the possibility to manage the LED strip with both RF remote and zigbee gateway (in this case I hope to use Deconz).

What's your esperience? Suggestions? Alternatives?

Thanks

SnaKKo, 2020-05-14:
anyone?

StephanStS, 2020-05-30:
I have two Gledopto based light strips. I have RGB+CCT and also RGB+WW controllers working fine with Conbee II.
This screenshot shows the WW controller: 
[image]

Mimiix, 2020-06-06:
As this seems to be answered , i close this.

closed 2020-06-06 by Mimiix

SnaKKo, 2020-06-15:
I've received the controller, all works correctly!

niquini, 2020-12-21:
> I have two Gledopto based light strips. I have RGB+CCT and also RGB+WW controllers working fine with Conbee II.
> This screenshot shows the WW controller:
> [image]

how did you pair? here is not for nothing

StephanStS, 2020-12-22:
On the Gledopto devices there is a pushbutton.  
Also in the description there is an option (which did not work so well for me) by switching off and on a couple of times until the light begins to flash (I think it was 5 times off/on).

Do you have a controller? Is there a quick setup manual present?
If yes: Try the RTFM method. :-)

GallusMax, 2021-01-29:
> On the Gledopto devices there is a pushbutton.
are you sure? then we are not talking about the same device.

> Also in the description there is an option (which did not work so well for me) by switching off and on a couple of times until the light begins to flash (I think it was 5 times off/on).
> 
you name it - "not so well" is just saying "never ever" to me. I had _no_ reaction or any flashing, and I tried, um, "more than twice" ;-)

> Do you have a controller? Is there a quick setup manual present?
> If yes: Try the RTFM method. :-)

StephanStS, 2021-01-29:
@GallusMax: Could you supply us with a photo of the Gledopto device posted here?

GallusMax, 2021-01-30:
> @GallusMax: Could you supply us with a photo of the Gledopto device posted here?

of course! This is the cover of my unresponsive device, could as well be an empty case..
(there is no "Gledopto" String to be seen - could this be a fake?)

[
[image]
](url)

StephanStS, 2021-01-31:
The documentation says:

1. Switch on your device
2. Now switch off an on within 2 seconds
3. Repeat off/on four times
4. Reset is done when the device is switched on the fifth time and the light stays on after blinking 4 times

In the photo I see that you mounted a single LED. Did you test this with a "real" LED strip?

GallusMax, 2021-01-31:
> The documentation says:
> 
>     1. Switch on your device
> 
>     2. Now switch off an on within 2 seconds
> 
>     3. Repeat off/on four times
> 
>     4. Reset is done when the device is switched on the fifth time and the light stays on after blinking 4 times
> 
Thanks for your advice. I do not understand the "blinking 4 times". Does it blink _after_ being connected the fifth time, or do the 1+4 disconnections count as "blinking"? 
You see: I am in doubt of handling a brick..

The result is: lights go on when under power, and off again when disconnected. I can do this 3, 4, 5, 6 or seven times, all within 2 seconds, or each toggle within 2 seconds, to _no_ further reaction.

> 
> In the photo I see that you mounted a single LED. Did you test this with a "real" LED strip?
Of course, I connected a 12V WW/CW strip and all I saw the WW at full power while the CW seems off constantly. The connected load should not affect the pairing, doesnt it?

I got the hint to change the system channel from ch25 to (something else?). Thats what I am going to try as last effort. But with little hope.

>"On the Gledopto devices there is a pushbutton." 
Can you identify a button on the box?

GallusMax, 2021-01-31:
>I got the hint to change the system channel from ch25 to (something else?). Thats what I am going to try as last effort. But with little hope.

sometime later: I managed to change the channel to 11, 15, 20, but got not any clue there was another device waiting to pair. sigh.

StephanStS, 2021-01-31:
The 4 times blinking means that when you switch on the fifth time, the controller lets blink the connected light strip 4 times and then stays light. 
Every of the four switch off operations must be within a 2 seconds duration related to the last switch on (i assume the off time duration is not of interest). Example with 1 s on time and 1.5 s off time
- t = 0 s: Switch to `ON #1`
- t = 1 s: Switch to OFF
- t = 2.5 s: Switch to `ON #2`
- t = 3.5 s: Switch to OFF
- t = 5 s: Switch to `ON #3`
- t = 6 s: Switch to OFF
- t = 7.5 s: Switch to `ON #4`
After this the controller lets blink the strip 4 times and ends with strip ON. After this you can scan for the device.

On your box there is no button. I had another different led controller with a button. But the description above is from the manual of my device which is identical to your photo.