dresden-elektronik/deconz-rest-plugin #2738: Last commit is one month old, what's happening?

SnaKKo, 2020-05-03:
I'm wondering if Deconz is still an active project.

andriej, 2020-05-03:
I hope they finish the zigbee2mqtt integration ASAP, because it's getting hilarious sometimes to wait and observe how slow development becomes.

SnaKKo, 2020-05-03:
@andriej Are you refering at this?

https://github.com/Koenkk/zigbee-herdsman/issues/72

andriej, 2020-05-03:
[yes](https://www.reddit.com/r/homeassistant/comments/f7qhxs/zigbee2mqtt_integration_in_deconz/)

osos, 2020-05-03:
I have now moved to zigbee2mqtt. Unfortunately with the CC2531.

I will reattach the Conbee II once the firmware is fixed and zigbee2mqtt fully and stable supports it.

Otherwise I might look towards the Sonoff ZBBridge as alternative hardware.

andriej, 2020-05-03:
Don't. Just order cc2530+cc2590 and it's as powerful (in terms of range) as conbee.

felixstorm, 2020-05-05:
@osos Stay with zigbee2mqtt, it _is_ great, but the CC2531 really s..ks for it's lack of stability when using more than a few devices. If ConBee integration does not move fast enough for you, then get a CC2652 (TI LaunchPad): This seems to be very stable for a lot of people (I only get lockups maybe once every few weeks with around 40+ devices).

I also did like deCONZ with a ConBee stick for it's stability and close adherence to ZigBee standards, but was frustrated with how difficult it is to support new devices. Then I move to Z2M which by now also supports a lot of ZigBee features like OTA, binding, reporting etc. directly. And adding support for new devices is as easy as adding a few lines of code to a JavaScript file...

stigvi, 2020-05-05:
> Then I move to Z2M which by now also supports a lot of ZigBee features like OTA, binding, reporting etc. directly. And adding support for new devices is as easy as adding a few lines of code to a JavaScript file...

I suppose I will move to Z2M, too, when I have time for it. I have to IKEA blinds waiting to get included.....

I suppose there isn't a fast and easy route from deconz to Z2M. I have 150 nodes and it will take me days to reset them and include them in Z2M

andriej, 2020-05-05:
Unfortunately. I've tried setting up same MAC, network key, pan ID but doesn't seem to work. I have more than 60 nodes too...

felixstorm, 2020-05-05:
No, unfortunately I do not know of a fast route either, especially when switching coordinator hardware. It might be possible when keeping the ConBee stick, but that is definitely only a personal guess (and not even an educated one).

Mimiix, 2020-05-05:
As @manup  mentioned here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2658#pullrequestreview-405670616

He was away for a few weeks:) I know for a fact there will be a release this or next week.

anatolyk82, 2020-05-13:
Two months with no update! The last release is terrible. All blinds are gone from the UI, they constantly lose binding on current position. Three times for the last two months all devices in the entire house lost connections to the gateway which with unexplainable reasons cound't recover for 10-20 minutes. Only constant restarts could suddenly bring the device to live. The only hope on an update but there has no been any for 2 months. What's happening ?!

SnaKKo, 2020-05-19:
There is only one dev from dresden that manage this repo, right? I see just few pull request from external collaborators (thanks a lot).

Mimiix, 2020-05-19:
It seems like it. Unfortunately, their twitter is not really helpful either. Nor do i have had any luck in contacting @manup to join the deCONZ discord(or any other DE Rep).

Ofcourse it's corona time, but it should be nice to have some transparency.

SnaKKo, 2020-05-19:
Yes, it's sad because I consider this project really promising; with a little refactoring this project could rocks.

The alternatives runs faster...

Mimiix, 2020-05-19:
Not only that, but also because theres people contributing that don't even have a relationship with DE. I feel they should do something about that too.

Mimiix, 2020-06-09:
Closing as it was answered :)

closed 2020-06-09 by Mimiix