dresden-elektronik/deconz-rest-plugin #2739: [Request Device Integration] Profalux/Eveno ZigBee roller shutters

deliwyn, 2020-05-03:
Hello everyone,

Would it be possible to integrate roller shutters from Profalux ? It is a well-known roller shutter brand in France. They work with Zigbee (see https://zigbeealliance.org/member/profalux/) but I cannot add them in Deconz.
When I try to discover them, they are created as lights.
See below some screenshot of the cluster/node info. As you can see, there is not much nformation.
Basically, in the Phoscon app, there is no Vendor/Model or Version information.

Please tell me if I need to provide more screenshots or perform further testing.
Thank you very much! I have 11 of them in my new house and I'd like to control them through my Home Assistant :)

[image]

[image]

[image]

Smanar, 2020-05-03:
Hello,
But theses device aren't already working ?

I have see some comment about that !

Have you tried using the API ? the on/off fonction aren't working ?

deliwyn, 2020-05-03:
Here are the information I can see when I press the second circle on the node title.

[image]

I'm sorry, I don't know how to test using the API since me Conbee II is plugin on my RPi running HaasIO. I was able to access to the GUI using VNC but I don't know how to launch the API... sorry (I have to confess that I am a french newbie).

Smanar, 2020-05-03:
I can make some improvement.
- Add vendor
- Add model ID
- Correct some problem you can have with min and max ( need to check)
- Invertion if needed
- Stop fonction

I will not change
- Device id, because this device don't work like a shutter (no shutter cluster) but like a dimmable light, and deconz use special procedure for window covering

I need too you make some tries:
This device work like a dimmable light, If you select the cluster 0006, you can tell me if "on" and "off" command are working
And on Cluster 0008, you can select the shutter position for sure.

deliwyn, 2020-05-03:
Here is the result of the Attributes of Cluster 0008.

[image]

[image]

But in fact, as I said I think there is a issue now with the calibration of my shutter. Because 254 = closed and 1 is a very little opened.

Smanar, 2020-05-03:
But it works or not ?
You can move it using cluster 0006 or 0008 or both ?

deliwyn, 2020-05-03:
When I click on exec for On or Off for cluster 0006, the shutter goes up & down very quickly.
In fact, it seems that he receives a command but the shutter doesn't go completly to the top or to the down.

Smanar, 2020-05-03:
Perfect will be easy. I m on code ATM
Can you compile it on your side to make test ? From my memory it's not possible if you are using Hassio.

deliwyn, 2020-05-03:
I'm very sorry, I would like to make test for you but I don't know what and how :( I'm very new in this world of automation since I've installed Hassio on a RPi with a Conbee II USB key two weeks ago. I discover more and more each day but it is not easy :)

If you tell me exactly what I should do for you, I can make a try.

BTW, I try to add a simple button in Home Assistant and I can the light (my shutter in fact). If I click on the button, it will open or close completly the shutter! It is a first victory for me! Thanks to you!

Smanar, 2020-05-03:
I realy don't remember the problem on Hassio, I think it's because they are using "special package".

But you can try, the procedure is here https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"

Just at step 1 use this command instead
git clone https://github.com/Smanar/deconz-rest-plugin.git

But look at step 4, if you can find this file, if not, you can forget.

Ok so I have modified (If I m right)
- model Id and manufacture
- "stop" command usable with API, you are using bri:125 for set it a 50% and bri:stop to stop it

This device never reach 0 and 255, always 1 > 254, but IDK if it can change something.

I haven touched at bind and report, tell me if you need it.

deliwyn, 2020-05-03:
Hi @Smanar and thank you again for your help!

I was able to add 5 of my 11 Profalux shutters in Deconz and I can use them in Hassio using basic light commands for the moment (Turn On or Turn Off).

I'm also able to set a default position of one of my shutter. In the screenshot below, you can see that I've specified 50% as the brightness and it seems to be equal to 1/3 of the full travel of the shutter. 

[image]

For now, the shutters are "unknown" in Phoscon but it does work !! :)

[image]
[image]

Tomorrow, I will see how can I send the bri:stop command in Home Assistant.
But I don't think I'm skilled enough to do the test with the new version of the Deconz plugin. I'm afraid that it won't work anymore afterwards in Home Assistant :) I hope you'll understand.
However, thanks again for your reactivity and for having had the manufacturer and the ID model !

Smanar, 2020-05-04:
There is no danger in compiling the code, at final you replace only 1 file, and you can make a backup of it, it don't change anything in your config, in worst situation it will just broke the deconz working mode.

For the 50 % = 1/ 3, idk, if 98% is almost closed, I think it's normal.

BTW I m reading that on Profalux corner https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Profalux-corner.md
> It is adviced to use only the LevelControl even for a full Open or Close.

I m trying to get more information.

Smanar, 2020-05-04:
Ok so I have some return from pipiche.

He have a contact at D&A, and it's the support that have preconised use the level control isntead of the on/off
You can see a working difference when you use a method or the other ? Perhaps slow-down at start or end of cycle ?

deliwyn, 2020-05-04:
No I don't see any "visible" difference in the cycle between on/off and the level control.

In HA, to open it (brightness = 100) :
[image]

And I also use the turn_on command but with brightness = 0 to close it:
[image]

[image]

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

bckthomas, 2020-06-25:
Hi @deliwyn 
I'm in France too. For the moment i'm using a zigate but it has some misfires. Do you use the conbee II ? Does it work perfectly for covers in home assistant ?
Thanks for your feedback

vanackej, 2020-07-15:
Hi @Smanar 
I'm using Conbee 2 with Profalux roller shutters.
I successfully integrated them and I am able to control them as expected.
But I face an issue : All shutters (I have 4 of them) are doing some very quick up/down cycles, approximately every 30 minutes, when the gateway is up.
Do you have any idea how I could debug this issue ? 
Am I the only one facing this issue ?

Any help appreciated

Smanar, 2020-07-15:
Hello, nope, it s the first time I m listen that.
Wich one deconz version have you ?

But something you can try the constructor say it s better using the setlevel cluster instead of the on/off cluster.

So idk wich one application you are using with deconz but instead of using open/close command, try to open or close  it at 95% (without using the stop command ofc, only the setlevel one) and wait 30m, t osee if you still have the problem on it.

Are you sure you haven't a script/automation on them ?

vanackej, 2020-07-20:
Hello @Smanar 
Yes I'm sure I don't have any automation/script on lights.
If I shut down my domotic solution (Jeedom) and just start the gateway, the problem occurs. When I stop the gateway, the problem disappear. So I really think the problem is related to the gateway.
I've started the gateway in debug and have logs around the moment it happens, but I'm new to Zigbee and there is really a lot of debug info, so I was unable to understand where the problem is. Maybe you could ?
I know the problem occurred somewhere between 17:53 and 17:54
Is there any kind of periodic task performed by the gateway that could explain the issue ?
[deconzServer.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4946362/deconzServer.txt)

vanackej, 2020-07-20:
@Smanar Another issue with the Profalux shutters support. The shutters are doing a double movement when I send any command to them through the gateway. 
As the `bri` parameter is not accepted alone for setting the level, I have to include the `on` parameter too. And the shutter seems like to react as if it process first the `on` command, then the `bri` command, thus doing 2 motor start.
Can you confirm me how the gateway is processing this ` { "bri": xxx, "on": true } ` command ? Is there another way to call the gateway and to avoid the double movement ?

Smanar, 2020-07-20:
Yep, you are right.
I have forget this device is reconized as light on not window covering. And yes if you use both, you will have 2 commands.

BTW I don't see anything about "bri" on your previous link ? But yes there is a removed part.
Have you tried directly with API ? I think in one position (open or closed) you will have this message "parameter, bri, is not modifiable. Device is set to off."

In last solution, I can make a code that disable the "on" if there is, a "bri" just for this device, not a problem ( 1/2 line code to add), but on your side you can just use "bri" on not "on", if it works ofc (cf my previous phrase).

And yes there is periodic action on device, attribute reporting , but its 300s max not 30mn.
You have the mac adress of the shutter ? to search in your log ?

Your shutters are at 0 % or 100 % or other value ?

vanackej, 2020-07-20:
In the commit I linked, there is conditional test about, if I understood well :

- If there is a `bri` parameter (enclosing `if`)
- If there is no `on` parameter and the device is not a `PLFX Shutter` (which is precisely my device)
- Then raise an error (the one you are speaking about in your previous message)

So, in this version, the `bri`parameter was allowed to be without the `on` parameter for my device. But all this logic was removed, as I can't see it in latest code.

I think your proposal should fix the issue, so it's worth a try.

I tried with the API directly, and I always get the error message when the `on` parameter is omitted, whatever the shutter position is.

About my other issue (the periodic movement), I will have a look at my logs with the one device mac, but as I have only 4 roller shutters associated to gateway, and nothing else, I think all logs should be relevant.

ebaauw, 2020-07-20:
Why aren’t you using `open` and `lift` for the shutter?  `on` and `bri` have been deprecated for window covering devices and will be removed.

For lights, `bri` is allowed when the light is on, or when `"on": true` has also been specified.  Same for setting the colour.

vanackej, 2020-07-21:
Hi @ebaauw, 
Well, I don't know really why, but the Profalux Roller Shutter are currently managed as lights by the Deconz gateway. I think it's related to the device Zigbee implementation. I would love to manage them as proper window covering devices, but I have no choice currently.
When you say that managing roller shutters as lights has been deprecated, what does it mean ? I specifically bought the Deconz usb key in order to control my roller shutters as support was in progress since May 2020, what is the future for this ? Are you continuing support for them as light ? Do you plan to make them available as windows covering ? Do you plan to drop support ?

Anyway, current support (as lights) is almost ok, excepted this double command/movement, and the fix proposed by @Smanar seems ok to me, but I'd be interested to know about the long term plans, if any.

ebaauw, 2020-07-21:
> When you say that managing roller shutters as lights has been deprecated

No, I don't say that.  Window covering devices are exposed as such (see the `type` attribute), but using a `/lights` resource.  It will be APIv2 before we would use `/devices` instead of `/lights` and `/sensors`.  What's deprecated is the use of `state.on` and `state.bri` for window covering devices, in favour of `state.open` and `state.lift`.

vanackej, 2020-07-21:
Are you saying that I should be currently able to interact with my shutters using the open and lift parameters ?
I tried with the API, with no luck, it seems the parameters are unknown.

Here are the node infos: 
```json
{
  "etag": "80eac512d4b89cb2d0230e6c1e1d9225",
  "hascolor": false,
  "lastseen": "2020-07-20T07:53:43.595",
  "manufacturername": "Profalux",
  "modelid": "PFLX Shutter",
  "name": "VR Chambre amis",
  "state": {
    "alert": "none",
    "bri": 254,
    "on": true,
    "reachable": true
  },
  "swversion": null,
  "type": "Color light",
  "uniqueid": "20:91:8a:00:00:0a:fb:a1-01"
}
```

The device type is Color light, and there is currently no open or lift state.

@Smanar can you confirm my understanding of the current situation, and that for now the easiest fix is to ignore the `on` parameter when the `bri` parameter is provided for this device ?

Smanar, 2020-07-21:
Don't worry, your device is not a shutter for deconz, so it don't work like other shutter.
It s a dimmable light, but unfortuntely for you light don't work like this device, for exemple a light in "off " state can receive command, your device can.

In all solution you need a hack, but easier to make a 4 lines hack in bulb gestion , than a big code in covering device juste for 1 device.

I can make a fast code today or tommorow for that, but can you compile it to test ? It s possible on jeedom if you have a raspberry (not arm64 machine)

vanackej, 2020-07-21:
My setup run on a core i7 server with proxmox, and the virtual machine use kvm64 processor. I honestly don't know if it would be ok to compile it, as I'm not really familiar with processors architecture and last time I compiled C++ was 15 years ago ! But I'm ok to make a try with some guidance. Should I just follow instructions here : https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only ?

Smanar, 2020-07-21:
Yep, but there i a problem, I don't think there is dev package for your machine :(

You can try if this command is working, if not, you can give up
`sudo apt install deconz-dev`

ebaauw, 2020-07-21:
> Are you saying that I should be currently able to interact with my shutters using the open and lift parameters ?
> Don't worry, your device is not a shutter for deconz

No, I’m asking why not.  I’m not convinced by the “device exposes itself as a light” argument.  If we’d follow that, I should have exposed the Xiaomi door sensors as on/off lights.

Smanar, 2020-07-21:
Yep, but the "patch" is not so ugly, and don't interact with the rest of the code

> https://github.com/Smanar/deconz-rest-plugin/commit/8ccd9a8d134d689ec84847dc705a9b154e6d1814

So for me it's the better solution.

vanackej, 2020-07-21:
The patch looks good to me.
I'm not sure, but do we agree with this patch the 'on' parameter won't ever be sent to the device, right ? Just the brightness setting. So no more double movement for the shutter ?

Smanar, 2020-07-21:
In fact I m not so sure ... I  m looking the 

```
    if (taskRef.lightNode->modelId().startsWith(QLatin1String("lumi.curtain.hagl04")))
    {
        cluster = ANALOG_OUTPUT_CLUSTER_ID;
    }
```

It s not so hard in reality. It s possible with just 10 lines maximum.

vanackej, 2020-07-21:
I'll try to compile it, but as I'm getting married this weekend, I'm not sure I will be able to confirm before next week 😅
Anyway, thanks for your help and your reactivity !

Smanar, 2020-07-21:
lool, you prefered married than use deconz ?

Gz :)

But you have time, the next version will be prepared in 20 days, ATM we just need to be sure it s working, and take a look on your strange bug (and perhaps move your device in covering device, but I think not)

I have ask pipiche too for you profalux problem, some way to explore

- Network problem.
- Identification request, but I don't see it in deconz.
- You haven't this problem every time you set phoscon in permit join ?

And you are the first one with this problem ...

vanackej, 2020-07-21:
I'll try to compile and use the dev version next week and will explore the different causes you listed for the other issue. 
I'll let you know ! thx

deliwyn, 2020-07-29:
> Hi @deliwyn
> I'm in France too. For the moment i'm using a zigate but it has some misfires. Do you use the conbee II ? Does it work perfectly for covers in home assistant ?
> Thanks for your feedback

Hi @bckthomas,
Sorry for my late answer! Yes I use a Conbee II with Home Assistant (Haasio on a RPi) and it does work well now thanks to the help of @Smanar :) I can open or close all of my covers or set them to a certain position (using brightness).
The only limitation I see is that I cannot send a stop command to the roller shutter while it's opening or closing, but I do not think this is related to the Conbe II nor to Home Assistant...

Smanar, 2020-07-29:
> The only limitation I see is that I cannot send a stop command to the roller shutter while it's opening or closing, but I do not think this is related to the Conbe II nor to Home Assistant...

Yes it is ^^
I haven't implemented it, but profalux give a way to do that, I will try to implement it in the profalux branch, but like usual will be fine if someone can test it.

Smanar, 2020-07-29:
Ok , so I have added a feature, trying to don't broke the ebaauw cleaning work.

Re added the stop part, but only activated by profalux device.
To stop the device just use "bri_inc" with the parameter you want, the code just check for the parameter is present.

But I haven't tested the code on my side, I can try to compile it, but it s useless because I can't try it. So I realy need someone test it on your side.

deliwyn, 2020-07-30:
Hi @Smanar and thank you again for your quick reply as always :)

I do not know how or when can I use "bri_inc" in Home Assistant for my Profalux device.
The only option I see in the UI is "brightness_pct". I use that option to open my roller by 75% as an example (in YAML below).

[image]
[image]

If you can tell me how I can do the test, I will be glad to help you.

vanackej, 2020-07-30:
@Smanar Hi
I'm trying to build your profalux branch, but I'm struggling a bit to set up the dev environnement. I'm not familiar with QT programming, and I still have missing QT modules.
Is there any documentation about how to set up the build env ? I didn't find any (wiki, readme, ...)
Thanks

Smanar, 2020-07-30:
@deliwyn Sorry I don't know HA at all, but you have not a way to use custom JSON ? Because there is so much param possible, lot of more than in your screenshoot.
But If you have a better idea ? I can still change the code, I have thinked a command without "on" and "bri" (an empty command) but not possible without breaking Ebaauw code.

@vanackej You have all the procedure on the github home page.
Just change the step 1 by
`git clone --branch profalux https://github.com/Smanar/deconz-rest-plugin.git`

and can ignore " git checkout -b mybranch HEAD"

Smanar, 2020-07-30:
If you are not using the same machine, you will have a new zigbee network, so an emtpy one, so you can use the phoscon feature in gateway to make a backup / restaure.

Or keep the empty network and just re-include a profalux device, but I know they are boring to include, so if you don't have too much device, the backup/restore feature is usefull.

In this situation just need to replace the lib file (its the last step), you can make a backup of the file (IN ANOTHER FOLDER), and just restart deconz, no need to delete and re-include device.

And for you, you can configure jeedom to use "distant deconz install" (don't remember the exact name), so all will work like before.

vanackej, 2020-07-30:
I already performed a backup/restore on the new virtual machine running the compiled gateway, my devices are here and API access is ok.
I'm performing my tests without Jeedom, directly using the deconz API, in order to be sure where my problems are coming from. 
Ok for the .so file replacement, I think thats what was missing (because for now I still have the double engine movement issue)

vanackej, 2020-07-30:
Great, shutter positionning now works perfectly ! No more double movement. Thanks !
In the branch I checked out, there is also something related to the stop command. So I tried and it's ok using `{ "bri_inc": any_value_here }` command, that's great too ! 
But I don't really get why the original design you planned `{ "bri": "stop" }` or an alternative one using `{ "stop": true }` was changed for using this strange `bri_inc` parameter.
Anyway, I can confirm the part related to shutter open/close and positionning is working as expected, and also that the close command works, even if the command syntax may still have to evolve.

vanackej, 2020-07-31:
I just opened a separate issue for the periodic movement, as this issue is getting a little messy

Smanar, 2020-07-31:
it s because someone have clean the code and make a part for covering device and one part for light.
In reality there is a "stop" command for light too, but not used by other gateway so he have removed it.

For the moment your device is in light section because it's easier (your device realy work like a light) but I think one day it will move in covering device and use command like "open" "lift" "stop" and ect ...

And yet, good idea the new issue, and you still have the problem even with the new code ?

Have you this problem too when your shutter are at 50 %.

Can you try to cut the power on 3 of them ? if the device memorise the config ofc, I know those device are hard to configure.

Smanar, 2020-07-31:
Another you can try, is starting deconz but without the API lib, but you need to do that on your second machine, because you need to use an OS with GUI to disable the API in deconz option.

Like that we can know if the but is from the zigbee network or a bug in the API code.

I have try to read again your log, but there is realy too much line to read ...

vanackej, 2020-07-31:
Yes, the problem is still there and I pretty sure it's related to the gateway, as I removed every other possible causes (as described in the new issue).
I'm leaving for 2 weeks (holidays), I'll try to reproduce with one single shutter and get logs of the exact moment it happens when I come back.
In the meantime, I think all the other improvements can be integrated as I have been able to test and validate them.

vanackej, 2020-07-31:
Ok, I'll try that as soon as I'm back

closed 2020-08-12 by manup

GLLM, 2022-05-30:
Hello @deliwyn, @vanackej, all on the thread,

Sorry for pinging ppl directly, but I'm getting absolutely bored about my issue.

I've recently installed Profalux windows equiped with Eveno rolling shutter (i.e. with Zigbee). Seemingly the same as that of @deliwyn.

I'm working with 
- Phoscon 2.16.01 (with docker) / firmware deCONZ_ConBeeII_0x26780700.bin.GCF
- Homeassistant 2022.5.5 (with docker)
- on a intel NUC

I've tried litteraly dozens & dozens of time to add one of my (9) shutters to deconz and HA ... to no avail.

Can anyone advise if they can spot what I'm doing wrong:
- reset the remote with 5xR + up then when the moves end STOP
- open my zigbee network from phoscon in the light section
- on the remote 1xR + UP then when the moves end STOP

in phoscon webapp : I see nothing
in VNC (after stopping deconz service), I see a unique device with a short alphanumeric name, when opening its properties, it has 2 sections
[image]

From that point, the native remote control does not control my shutter anymore.
But : I dont see anything in Phoscon (web)app nor in Home Assistant.

So I cannot do anything anymore.
 Next thing I must do is stop the electricity for my shutter on my central electric board for 5+ seconds, reset the physical remote so that I (successfully get back to square one).
Note: in my rolling shutter block, my black electrical cable is not plugged (neither to the blue nor to the brown cable).

Can anyone spot what I'm doing wrong ?
If you have any hint, please help me :-)
Thanks a lot !!!!!

GLLM

Mimiix, 2022-05-30:
@GLLM can you click the basic cluster and read that?

Smanar, 2022-05-30:
@GLLM But what are you doing ?
You want to include the covering or the remote ?
The only thing you have included in deconz is the remote

Perhaps this can help you https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Corner_Profalux-corner.md

GLLM, 2022-05-31:
> @GLLM can you click the basic cluster and read that?
Sure, here it is:
[image]

GLLM, 2022-05-31:
> @GLLM But what are you doing ? You want to include the covering or the remote ? The only thing you have included in deconz is the remote
> 
> Perhaps this can help you https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Corner_Profalux-corner.md

Hi @Smanar
I'm trying to include whatever I can into my Homeassistant instance, via Phoscon.
I understand that I'm not the first one to have 2 things included at once.

I have included whatever I could given all the different methods I found on the internet.

Included the link you provided, which I knew about.
As a matter of fact, I followed exactly their step by step method (as described above).
[image]

I'm lost.
What I just want to do is : be able to control my shutters from homeassistant.
As I've read multiple times it's not possible to have the HA instance + the physical remotes work at the same time.

Thanks in advance for any hint !

GLLM

Mimiix, 2022-05-31:
Hi,

The screenshot is correct but you need to click the read button. Then the values get filled in.

GLLM, 2022-05-31:
I'll redo it right away.
... incoming ...

Smanar, 2022-05-31:
>I'm not the first one to have 2 things included at once (remote+shutter), as visible in my first post's screenshot

But you have only 1 remote included. No covering at all

GLLM, 2022-05-31:
> > I'm not the first one to have 2 things included at once (remote+shutter), as visible in my first post's screenshot
> 
> But you have only 1 remote included. No shutter at all

Then I have no clue how I should do it  ... :-/

Smanar, 2022-05-31:
>NB: I've not been provided with a general/global Zoe remote. Only one remote per shutter/windwo.

Can be the problem, are you french ? https://zigate.fr/2018/02/03/association-des-volets-profalux-a-la-zigate/
On this procedure you need both

https://i0.wp.com/zigate.fr/wp-content/uploads/2018/02/association-volet-%C3%A0-la-zigate.jpg

GLLM, 2022-06-03:
> > NB: I've not been provided with a general/global Zoe remote. Only one remote per shutter/windwo.
> 
> Can be the problem, are you french ? https://zigate.fr/2018/02/03/association-des-volets-profalux-a-la-zigate/ On this procedure you need both
> 
> [image]
> 
> You need the master remote to open the network.
> 
> Here another procedure using the wire https://zigate.fr/forum/topic/probleme-dassociation-de-volets-profalux-avec-la-zigate/
> 
> Some lucky user just make
> 
>     * Set the coordinator in permit join
> 
>     * R + UP
> 
>     * Stop the permit join

Hi @Smanar
this message just to once again thank you very much for the wired method to pair the cover.
I now have 5 paired covers+native_remotes in my HomeAssistant instance. All work great, native remote still work as well. 4 more to go, but it's easy now :)

For posterity & archival, I have the model=MOT-C1Z06C of Profalux shutters.

Bye,
GLLM