dresden-elektronik/deconz-rest-plugin #2741: [Request Device Integration] Iris Contact Sensor v2 (CentraLite 3320-L)

poblabs, 2020-05-04:
I know the Iris Contact Sensor **v3** is supported but it doesn't look like the v2 is. 

Looks like it's really a CentraLite 3320-L based on the information below.

Looks like they can pair and be discovered but nothing shows in Phoscon or in Home Assistant 

Let me know if I can provide more info. Excited to get these working since I have a bunch of them!

[image]

[image]

[image]

SwoopX, 2020-05-04:
Duplicate of #2632.

It is already supported by this commit https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2572/commits/1896ee207424ba99285d4db0bce866c6ed457596 and should be available with the next release.

poblabs, 2020-05-04:
@SwoopX Awesome! Is there a way to integrate it sooner than the new release in the hass.io package? Maybe a custom dockerfile or something?

SwoopX, 2020-05-04:
You can always compile the latest version available here but since I don't use docker, HA or HassIO, I cannot tell you how it's done.

Folks ate deconz Discord channel may help you out with that https://discord.gg/rgpT96

SwoopX, 2020-05-10:
@poblabs Mind closing this issue if your questions have answered?

poblabs, 2020-05-10:
Sure will. Do you know when the release will come out? Looking forward to
start using these devices.

SwoopX, 2020-05-10:
Folks are saying it should be next week ;)

poblabs, 2020-05-10:
Looking forward to it!

closed 2020-05-10 by poblabs